intel/compute-runtime #428: Open CL on Comet Lake UHD Graphics

MAMorgado, 2021-07-01:
Hello,
Does your program work for Comet Lake UHD Graphics? I noticed that it's not on the supported platforms... 
I need OpenCL version 3.0 if possible.
I tried to install (sudo apt-get install intel-opencl-icd) but when I run clinfo I only get 
Number of platforms                               0
Can you provide some help?

MAMorgado, 2021-07-01:
Update: If I follow the instructions in Installation procedure on Ubuntu 18.04/20.04 ( i'm using ubuntu 20.04) during the last step I have the following error:
(Reading database ... 323160 files and directories currently installed.)
Preparing to unpack intel-gmmlib_21.1.3_amd64.deb ...
Unpacking intel-gmmlib (21.1.3) over (21.1.3) ...
dpkg: warning: downgrading intel-igc-core from 1.0.7712 to 1.0.7683
Preparing to unpack intel-igc-core_1.0.7683_amd64.deb ...
Unpacking intel-igc-core (1.0.7683) over (1.0.7712) ...
dpkg: warning: downgrading intel-igc-opencl from 1.0.7712 to 1.0.7683
Preparing to unpack intel-igc-opencl_1.0.7683_amd64.deb ...
Unpacking intel-igc-opencl (1.0.7683) over (1.0.7712) ...
dpkg: warning: downgrading intel-level-zero-gpu from 1.1.20114 to 1.1.20098
Preparing to unpack intel-level-zero-gpu_1.1.20098_amd64.deb ...
Unpacking intel-level-zero-gpu (1.1.20098) over (1.1.20114) ...
dpkg: warning: downgrading intel-ocloc from 21.25.20114 to 21.24.20098
Preparing to unpack intel-ocloc_21.24.20098_amd64.deb ...
Unpacking intel-ocloc (21.24.20098) over (21.25.20114) ...
Selecting previously unselected package intel-opencl.
Preparing to unpack intel-opencl_21.24.20098_amd64.deb ...
Unpacking intel-opencl (21.24.20098) ...
Replaced by files in installed package intel-opencl-icd (21.23.20043+i571~u20.04) ...
Setting up intel-gmmlib (21.1.3) ...
Setting up intel-igc-core (1.0.7683) ...
Setting up intel-igc-opencl (1.0.7683) ...
Setting up intel-level-zero-gpu (1.1.20098) ...
Setting up intel-ocloc (21.24.20098) ...
dpkg: dependency problems prevent configuration of intel-opencl:
 intel-opencl-icd (21.23.20043+i571~u20.04) breaks intel-opencl and is installed.

dpkg: error processing package intel-opencl (--install):
 dependency problems - leaving unconfigured
Processing triggers for libc-bin (2.31-0ubuntu9.2) ...
Errors were encountered while processing:
 intel-opencl

JacekDanecki, 2021-07-19:
Comet Lake is  supported, see: https://github.com/intel/compute-runtime/blob/master/shared/source/dll/devices/devices_base.inl#L169.
You can't mix packages from different sources. If you want to use packages from github release, you have to remove previously installed packages.
Can you upgrade ocl-icd package to fix problems with NULL platform?

MAMorgado, 2021-07-20:
Hi Jacek. Thank you for your reply. 
I did what you asked and it seems to have solved the problem with the null no platform detection. Thank you very much. This is what I get back now:

NULL platform behavior
  clGetPlatformInfo(NULL, CL_PLATFORM_NAME, ...)  Intel(R) OpenCL HD Graphics
  clGetDeviceIDs(NULL, CL_DEVICE_TYPE_ALL, ...)   Success [INTEL]
  clCreateContext(NULL, ...) [default]            Success [INTEL]
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_DEFAULT)  Success (1)
    Platform Name                                 Intel(R) OpenCL HD Graphics
    Device Name                                   Intel(R) UHD Graphics [0x9bca]
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CPU)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_GPU)  Success (1)
    Platform Name                                 Intel(R) OpenCL HD Graphics
    Device Name                                   Intel(R) UHD Graphics [0x9bca]
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ACCELERATOR)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CUSTOM)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ALL)  Success (1)
    Platform Name                                 Intel(R) OpenCL HD Graphics
    Device Name                                   Intel(R) UHD Graphics [0x9bca]

ICD loader properties
  ICD loader Name                                 OpenCL ICD Loader
  ICD loader Vendor                               OCL Icd free software
  ICD loader Version                              2.2.11
  ICD loader Profile                              OpenCL 2.1
	NOTE:	your OpenCL library only supports OpenCL 2.1,
		but some installed platforms support OpenCL 3.0.
		Programs using 3.0 features may crash
		or behave unexpectedly
The only thing that might be wrong is the 3.0 vs 2.1 note in the end, but i don't know if it's normal...

JacekDanecki, 2021-07-20:
You need to install ocl-icd 2.2.13 with OpenCL 3.0 support, see: https://github.com/OCL-dev/ocl-icd/commit/3b7ded60ebb7e1afddcbae6f82ac8645b276e358#diff-7ee66c4f1536ac84dc5bbff1b8312e2eef24b974b3e48a5c5c2bcfdf2eb8f3ceR6

Kerilk, 2021-07-22:
I create a ppa for this purpose:
https://launchpad.net/~ocl-icd/+archive/ubuntu/ppa
It should allow for installing the latest ocl-icd on recent  (and not so recent) ubuntus.

Please also note that the clinfo here is outdated as well so you may get a slightly different warning once the loader is updated (see https://github.com/Oblomov/clinfo/pull/55).

JacekDanecki, 2021-08-12:
Closing issue, please create new issue when you find any problems.

closed 2021-08-12 by JacekDanecki