intel/compute-runtime #431: Latest release (21.26.20194) build fails to LLVM cli options usage error

eero-t, 2021-07-07:
compute-runtime "21.26.20194" release build fails to incorrect LLVM (v10) CLI options usage:
```
RUN git clone --branch ${TAG_COMPUTE} --depth 1 https://github.com/intel/compute-runtime.git  &&     cd compute-runtime  &&  mkdir build  &&  cd build  &&      cmake -LH -Wno-dev -G Ninja       -DCMAKE_INSTALL_PREFIX=${INSTALL_DIR} -DCMAKE_BUILD_TYPE=Release       -DSUPPORT_GEN8=0 -DSUPPORT_GEN9=1 -DSUPPORT_GEN11=1       -DSUPPORT_GEN12LP=1 -DSUPPORT_DG1=1       -DDO_NOT_RUN_AUB_TESTS=1 -DDONT_CARE_OF_VIRTUALS=1       ../  &&     ninja
...
[291/2286] Generating ../../../../bin/built_ins/x64/gen12lp/bindful_copy_buffer_rect_Gen12LPlp.spv
FAILED: bin/built_ins/x64/gen12lp/bindful_copy_buffer_rect_Gen12LPlp.spv 
cd /home/nobody/source/compute-runtime/shared/source/built_ins/kernels && LD_LIBRARY_PATH=/home/nobody/source/compute-runtime/build/bin /home/nobody/source/compute-runtime/build/bin/ocloc -q -file copy_buffer_rect.builtin_kernel -spv_only -device tgllp -64 -output bindful_copy_buffer_rect -out_dir /home/nobody/source/compute-runtime/build/bin/built_ins/x64/gen12lp -options -cl-kernel-arg-info
: CommandLine Error: Option 'mc-relax-all' registered more than once!
LLVM ERROR: inconsistency in registered CommandLine options
```
In case it matters, this build happens in Ubuntu 20.10 container with latest IGC release "igc-1.0.7862" built using LLVM v10 just before it (which requires workaround for https://github.com/intel/intel-graphics-compiler/issues/186  bug).

Here's more of the output before that error:
```
...
[0m-- The C compiler identification is GNU 10.3.0
-- The CXX compiler identification is GNU 10.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done

-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- branch dir list: /
-- WDK Directory: 
-- WDK Version is 
-- Driver model : drm
-- Release build configuration
-- Computed OpenCL version major is: 21
-- Computed OpenCL version minor is: 26
-- GTest repeat count set to 1
-- GTest shuffle set to --gtest_shuffle;--gtest_random_seed=0
-- Source Level Debugger headers dir: /home/nobody/source/compute-runtime/third_party/source_level_debugger
-- Aub Stream Headers dir: /home/nobody/source/compute-runtime/third_party/aub_stream/headers
-- Metrics Library dir: /home/nobody/source/compute-runtime/third_party/metrics_library
-- Metrics Discovery dir: /home/nobody/source/compute-runtime/third_party/metrics_discovery
-- i915 includes dir: /home/nobody/source/compute-runtime/third_party/uapi/dg1
-- Khronos OpenCL headers dir: /home/nobody/source/compute-runtime/third_party/opencl_headers
-- Khronos OpenGL headers dir: /home/nobody/source/compute-runtime/third_party/opengl_headers
-- Third party dir: /home/nobody/source/third_party
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'igc-opencl'
--   Found igc-opencl, version 1.0.1
-- IGC include dirs: /usr/local/include/igc;/usr/local/include/igc/cif;/usr/local/include/igc/ocl_igc_shared/executable_format;/usr/local/include/igc/ocl_igc_shared/device_enqueue
-- VISA Dir: /usr/local/include/visa
-- IGA Includes dir: /usr/local/include/iga
-- Checking for module 'igdgmm'
--   Found igdgmm, version 11.3.0
-- GmmLib include dirs: /usr/local/include/igdgmm;/usr/local/include/igdgmm/GmmLib;/usr/local/include/igdgmm/GmmLib/inc;/usr/local/include/igdgmm/inc;/usr/local/include/igdgmm/inc/common;/usr/local/include/igdgmm/util
-- Checking for module 'libva>=1.0.0'
--   Found libva, version 1.12.0
-- Looking for vaGetLibFunc in va
-- Looking for vaGetLibFunc in va - found
-- Using libva 
-- LibVA include dirs: /usr/local/include
-- AUB_STREAM_DIR = 
-- Engine node dir: /home/nobody/source/compute-runtime/third_party/aub_stream/headers
-- All supported platforms:  TGLLP DG1 RKL ADLS ICLLP LKF EHL SKL KBL GLK CFL BXT
-- All tested platforms:  TGLLP RKL ADLS ICLLP LKF EHL SKL KBL GLK CFL BXT
-- Default supported platform: SKL
-- Default tested platform: SKL
-- All supported core families: GEN9;GEN11;GEN12LP
-- All tested core families: GEN9;GEN11;GEN12LP
[... 188 lines cut ...]
// Use OpenCL program binary cache
USE_CL_CACHE:BOOL=ON

// Build unit tests.
level-zero-gpu_BUILD_TESTS:BOOL=ON

// Path to a file.
libigsc_INCLUDE_DIR:PATH=libigsc_INCLUDE_DIR-NOTFOUND

// Path to a library.
libigsc_LIBRARIES:FILEPATH=libigsc_LIBRARIES-NOTFOUND

[1/2286] Building CXX object offline_compiler/source/CMakeFiles/ocloc.dir/main.cpp.o
...
[290/2286] Generating ../../../../bin/built_ins/x64/gen12lp/bindful_aux_translation_Gen12LPlp.spv
FAILED: bin/built_ins/x64/gen12lp/bindful_aux_translation_Gen12LPlp.spv 
cd /home/nobody/source/compute-runtime/shared/source/built_ins/kernels && LD_LIBRARY_PATH=/home/nobody/source/compute-runtime/build/bin /home/nobody/source/compute-runtime/build/bin/ocloc -q -file aux_translation.builtin_kernel -spv_only -device tgllp -64 -output bindful_aux_translation -out_dir /home/nobody/source/compute-runtime/build/bin/built_ins/x64/gen12lp -options -cl-kernel-arg-info
: CommandLine Error: Option 'mc-relax-all' registered more than once!
LLVM ERROR: inconsistency in registered CommandLine options
[291/2286] Generating ../../../../bin/built_ins/x64/gen12lp/bindful_copy_buffer_rect_Gen12LPlp.spv
FAILED: bin/built_ins/x64/gen12lp/bindful_copy_buffer_rect_Gen12LPlp.spv 
cd /home/nobody/source/compute-runtime/shared/source/built_ins/kernels && LD_LIBRARY_PATH=/home/nobody/source/compute-runtime/build/bin /home/nobody/source/compute-runtime/build/bin/ocloc -q -file copy_buffer_rect.builtin_kernel -spv_only -device tgllp -64 -output bindful_copy_buffer_rect -out_dir /home/nobody/source/compute-runtime/build/bin/built_ins/x64/gen12lp -options -cl-kernel-arg-info
: CommandLine Error: Option 'mc-relax-all' registered more than once!
LLVM ERROR: inconsistency in registered CommandLine options
```

eero-t, 2021-07-12:
Same problem with the new "21.27.20266" release.

jbeich, 2021-08-17:
Ditto on FreeBSD with 21.32.20609 (igc-1.0.8279) using [llvm10](https://github.com/intel/compute-runtime/files/7002200/intel-compute-runtime-llvm10-21.32.20609.log) but [llvm11](https://github.com/intel/compute-runtime/files/7002207/intel-compute-runtime-llvm11-21.32.20609.log) is fine.

jbeich, 2021-08-17:
FWIW, llvm90 and older are even more buggy (at least with vanilla/unpatched llvm sources) and llvm12 isn't supported yet.

JacekDanecki, 2021-08-30:
Have you build IGC with system llvm, or with llvm sources?

eero-t, 2021-08-30:
It's system LLVM from Ubuntu.

JacekDanecki, 2021-08-31:
What about opencl-clang and SPIRV-LLVM-Translator? Have you built them or used from the system?
If you built them yourself what method did you use? (In-tree build or Out-of-tree build)

eero-t, 2021-09-01:
Those are also from the system.

eero-t, 2021-09-01:
Same problem also with latest IGC (igc-1.0.8517) and compute-runtime (21.34.20767).

Compute-runtime (still) does not build with LLVM v10 (in Ubuntu 20.10) any more:
`apt install  clang-10 libopencl-clang-dev opencl-headers llvm-10-dev liblld-10-dev llvm-spirv libllvmspirvlib-dev`

but it does build with LLVM v11 (in Ubuntu 21.04):
`apt install clang-11 libopencl-clang-dev opencl-headers llvm-11-dev liblld-11-dev llvm-spirv libllvmspirvlib-dev`

(Using `-DIGC_OPTION__LLVM_PREFERRED_VERSION=11` instead of `-DIGC_OPTION__LLVM_PREFERRED_VERSION=10`, with `-DINSTALL_SPIRVDLL=0`option naturally still remaining with IGC. )

JacekDanecki, 2021-09-06:
This is not Neo problem, but how [opencl-clang](https://github.com/intel/opencl-clang) library was compiled on Ubuntu, and I suppose the same is on FreeBSD. To workaround this issue you need to recompile opencl-clang and llvm-spriv-translator with llvm sources using In-tree method. You don't have to recompile IGC after opencl-clang/llvm-spirv-translator build.
This is not specific to llvm10, but was observed in previous llvm versions. Similar issues were observed in another projects using llvm static libraries on Ubuntu.

eero-t, 2021-09-06:
> This is not specific to llvm10, but was observed in previous llvm versions.   Similar issues were observed in another projects using llvm static libraries on Ubuntu.

Any idea why it worked earlier with the same Ubuntu (20.10) version of LLVM v10, and still works with Ubuntu (21.04) LLVM v11?

JacekDanecki, 2021-09-06:
As the issue is on compiler side, transferring it to IGC project.