intel/compute-runtime #435: OpenCL program that Builds fine but does not Compile+Link

Kerilk, 2021-07-15:
Hello,

I was advised to raise this issue here as well: https://github.com/intel/intel-graphics-compiler/issues/185

The fact that we are using separate compilation is due to the use of the `-cl-take-global-address` at link time to be able to use the clGetDeviceGlobalVariablePointerINTEL extension function. Is there a way to specify this flag a build time directly? If this requires a longer discussion I can open another issue.

Thanks,

Brice

Kerilk, 2021-07-29:
Sorry if I didn't word my question correctly, let me try to reformulate: `clBuildProgram` rejects the `-cl-take-global-address` option while `clLinkProgram` accepts it. Is there a way to pass the `-cl-take-global-address` option to `clBuildProgram`?

JablonskiMateusz, 2021-07-30:
Based on what you said, I'm assuming that you try to build program created from source and you are hitting frond-end compiler failure as it doesn't support `-cl-take-global-address` option. 
If you are building program from source in two steps, then `clCompileProgram` calls only front-end compiler and `clLinkProgram` calls only back-end compiler. You use two different functions, so you can specify separate build options for each compiler. In case of `clBuildProgram` it takes build options and calls both front-end and back-end compilers using the same build options. There is no way to separate them on our side as we don't parse the options.

If you want to call `clBuildProgram` with `-cl-take-global-address` then try to create program using SPIR-V input instead of source code.

Kerilk, 2021-07-30:
Thanks for the follow up. I think I remember trying this and getting the same issue. I will try to replicate, though I am currently on vacations so it may take a few days.

Kerilk, 2021-08-18:
> If you want to call `clBuildProgram` with `-cl-take-global-address` then try to create program using SPIR-V input instead of source code.

I can confirm that this works, thank you very much.

closed 2021-08-18 by Kerilk