intel/compute-runtime #436: error: "‘GMM_FLATCCS_FORMAT’ has not been declared"

semenovad, 2021-07-16:
I followed the instructions given in BUILD.md, and ran into trouble at step 6. I run make -`j`nproc`` and **get an error**:

In member function ‘void NEO::Gmm::applyAuxFlagsForImage(NEO::ImageInfo&)’:
`/home/noname/ComputeRuntime/workspace/neo/shared/source/gmm_helper/gmm.cpp:165:59: error: ‘GMM_FLATCCS_FORMAT’ has not been declared
  165 |         compressionFormatSupported = compressionFormat != GMM_FLATCCS_FORMAT::GMM_FLATCCS_FORMAT_INVALID;
      |                                                           ^~~~~~~~~~~~~~~~~~`

I tried to get repos that I needed from the ppa:intel-opencl/intel-opencl repository.
Then I decided that the problem was in gmmlib version (in previous step I got gmmlib-21.1.3) and installed the last one - gmmlib-21.2.1. 
But again and again got the same error.

OS: Ubuntu 20.04 (64-bit)

I'm sure that the problem isn't in the code or hardware, but really don't know what to do with that.

`cmake -DCMAKE_BUILD_TYPE=Release -DSKIP_UNIT_TESTS=1 ../neo
-- branch dir list: /
-- Driver model : drm_or_wddm
-- Release build configuration
-- Computed OpenCL version major is: 21
-- Computed OpenCL version minor is: 29
-- Source Level Debugger headers dir: /home/noname/ComputeRuntime/workspace/neo/third_party/source_level_debugger
-- Aub Stream Headers dir: /home/noname/ComputeRuntime/workspace/neo/third_party/aub_stream/headers
-- Metrics Library dir: /home/noname/ComputeRuntime/workspace/neo/third_party/metrics_library
-- Metrics Discovery dir: /home/noname/ComputeRuntime/workspace/neo/third_party/metrics_discovery
-- i915 includes dir: /home//noname/ComputeRuntime/workspace/neo/third_party/uapi
-- Khronos OpenCL headers dir: /home/noname/ComputeRuntime/workspace/neo/third_party/opencl_headers
-- Khronos OpenGL headers dir: /home/noname/ComputeRuntime/workspace/neo/third_party/opengl_headers
-- Third party dir: /home/noname//ComputeRuntime/workspace/third_party
-- WDK include paths: /home/noname/ComputeRuntime/workspace/neo/third_party/DirectX-Headers/include/wsl;/home/noname/ComputeRuntime/workspace/neo/third_party/DirectX-Headers/include/wsl/stubs;/home/noname/ComputeRuntime/workspace/neo/third_party/DirectX-Headers/include/directx;/home/noname/ComputeRuntime/workspace/neo/third_party/DirectX-Headers/include/dxguids;/home/noname/ComputeRuntime/workspace/neo/third_party/libdxg/include/
-- IGC include dirs: /usr/local/include/igc;/usr/local/include/igc/cif;/usr/local/include/igc/ocl_igc_shared/executable_format;/usr/local/include/igc/ocl_igc_shared/device_enqueue
-- VISA Dir: /usr/local/include/visa
-- IGA Includes dir: /usr/local/include/iga
-- GmmLib include dirs: /usr/include/igdgmm;/usr/include/igdgmm/GmmLib;/usr/include/igdgmm/GmmLib/inc;/usr/include/igdgmm/inc;/usr/include/igdgmm/inc/common;/usr/include/igdgmm/util
-- Using libva 
-- AUB_STREAM_DIR = 
-- Engine node dir: /home/noname/ComputeRuntime/workspace/neo/third_party/aub_stream/headers
-- All supported platforms:  TGLLP RKL ADLS ICLLP LKF EHL SKL KBL GLK CFL BXT BDW
-- All tested platforms: 
-- Default supported platform: SKL
-- Default tested platform: 
-- All supported core families: GEN8;GEN9;GEN11;GEN12LP
-- All tested core families: 
-- Default tested family name: 
CMake Warning at CMakeLists.txt:817 (message):
  Spectre mitigation -mindirect-branch=thunk flag is not supported by the
  compiler

CMake Warning at CMakeLists.txt:823 (message):
  Spectre mitigation -mfunction-return=thunk flag is not supported by the
  compiler

-- Level Zero driver version: 1.1.0
-- LevelZero_INCLUDE_DIRS: /usr/local/include
-- LibXml2 Library headers directory: /usr/include/libxml2
-- LibGenl headers not available. Building without
-- Could NOT find libigsc (missing: libigsc_LIBRARIES libigsc_INCLUDE_DIR) 
-- libigsc Library headers not available. Building without
-- L0::Gmm Dir: /home/noname/ComputeRuntime/workspace/gmmlib
Prebuilt kernels are linked to Level Zero.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/noname/ComputeRuntime/workspace/build
`

BartoszDunajski, 2021-07-16:
Please update GMM.
GMM_FLATCCS_FORMAT was added in intel-gmmlib-21.1.2

semenovad, 2021-07-16:
> Please update GMM.
> GMM_FLATCCS_FORMAT was added in intel-gmmlib-21.1.2

I checked the [ppa:intel-opencl/intel-opencl ](url) repository, there is an intel-gmmlib - 21.1.3-1~ppa1~focal1 version for 20.04u, so I just install it like that

`sudo add-apt-repository ppa:intel-opencl/intel-opencl
sudo apt-get update
sudo apt install libigdgmm-dev` 

But the error is still with me

JacekDanecki, 2021-07-16:
ppa:intel-opencl/intel-opencl will be updated today for Ubuntu 20.04, so you can use testing repository: https://launchpad.net/~jdanecki/+archive/ubuntu/intel-opencl, where there is gmmlib 21.2.1 already prepared: https://launchpad.net/~jdanecki/+archive/ubuntu/intel-opencl/+packages?field.name_filter=&field.status_filter=published&field.series_filter=focal

semenovad, 2021-07-16:
> ppa:intel-opencl/intel-opencl will be updated today for Ubuntu 20.04, so you can use testing repository: https://launchpad.net/~jdanecki/+archive/ubuntu/intel-opencl, where there is gmmlib 21.2.1 [...]

Thank You, now everything is correct!

closed 2021-07-16 by semenovad