intel/compute-runtime #440: OpenCL compiler issues for Intel 11th gen CPU?

vm3tr1c1, 2021-07-28:
- opencv 3.4.15
- centos 7.9, kernel 5.13.5-1 with i915.force_probe=4c8a
- gcc 7.3.1
- CPU: i5-11600
- intel-opencl-21.28.20343-1.el7.x86_64

At the runtime of an opencv application which use the ocl4dnn module, a lot of 'simd size' errors are generated, like these ones:

OpenCV(ocl4dnn): The OpenCL compiler chose a simd size (16) that does not equal the size (8) kernel source required.
                 Skip this kernel U_GEMM_LIKE_CONV_k3x3_cn3_g1_s2x2_d1x1_b1_in256x256_p1x1_num1_M32_activ1_eltwise0_FP32_5_1_8_32_SIMD8
OpenCV(ocl4dnn): The OpenCL compiler chose a simd size (32) that does not equal the size (16) kernel source required.
                 Skip this kernel IDLF_k3x3_cn3_g1_s2x2_d1x1_b1_in256x256_p1x1_num1_M32_activ1_eltwise0_FP32_2_7_3_1_SIMD16

If the CPU is replaced with an i5-10600, without changing anything else, the errors disappear and the GPU code runs without any issue at all.

alalek, 2021-07-28:
BTW,
OpenCL kernel attributes for the second case are [here](https://github.com/opencv/opencv/blame/3.4.15/modules/dnn/src/opencl/conv_layer_spatial.cl#L222).
Host-side check is based on `clGetKernelWorkGroupInfo()` call with `CL_KERNEL_PREFERRED_WORK_GROUP_SIZE_MULTIPLE` argument.

@vm3tr1c1 Do you use `OPENCV_OCL4DNN_FORCE_AUTO_TUNING=1` environment variable?

vm3tr1c1, 2021-07-29:
> @vm3tr1c1 Do you use `OPENCV_OCL4DNN_FORCE_AUTO_TUNING=1` environment variable?

I don't, but it doesn't seem to make any difference if I use it.

vm3tr1c1, 2021-07-29:
I'm not an expert in opencl, but please let me know how can I be of help (testing, debugging, whatever).