intel/compute-runtime #441: [opencl] how are the in order queue and out of order queue implemented from hardware perspective

guoyejun, 2021-07-29:
I understand in order queue and out of queue from OpenCL API perspective. For example, there are 10 independent kernels, the host code may look like:

clEnqueueNDRangeKernel(q, kernel0);
clEnqueueNDRangeKernel(q, kernel1);
...
clEnqueueNDRangeKernel(q, kernel9);

If q is in order queue, kernel i will be executed after kernel i-1 is finished. In other words, even if the output of kernel i-1 is the input of kernel i, the above host code is still correct, without any explicit dependency at host code.

 If q is out of order queue, kernel0 will be first executed on EU, and kernel1 (here, it is impossible for kernel 2, ahead of kernel 1) will be executed whenever there is free EU available, and then kernel2, kernel3 ... 

Just curious how the hardware implement such mechanism, is it about command fetch part? is it controlled by one register, or something else? 

thanks.

alalek, 2021-07-29:
BTW, previous similar question is here: #357

HoppeMateusz, 2021-08-11:
Hello,

For out of order execution driver does not inject any stalling commands in the pipeline between kernels. HW fetches and consumes issued commands sequentially and dispatches threads (when possible) for multiple kernels, allowing for parallel execution. Stalling hw commands, when processed by HW, cause the pipeline to wait for all previous commands to complete.

closed 2021-08-12 by guoyejun

guoyejun, 2021-08-12:
thanks @HoppeMateusz 

For OOO queue, if there's an event dependency command, how does the HW wait for the dependency to be satisfied? For example.
- The GPU HW fetches and parses the command, and finds that it is a wait not satisfied, so the HW fetching&parsing part is blocked (waiting for signal). In other words, HW will not try to fetch the next command even if there are some.
- After the hardware threads are spawned, these threads with dependency are blocked (waiting for signal) and so not dispatched. And, the HW continues the work as normal, for example, HW continues to fetch and parse next commands (suppose their dependency are satisfied already), spawns threads and dispatch these threads.