intel/compute-runtime #442: clinfo:<getWGsizes:675: build program : error -6>

pobin6, 2021-07-31:
I run openvino benchmark_app demo, it show the erro :  "Program build failed:" and no more message,
then I run clinfo, it show :  <getWGsizes:675: build program : error -6>
```
root@Storage:/home/Openvino/bin/intel64/Release# ./benchmark_app -m S3D-model/nnx-test.xml -d GPU
[Step 1/11] Parsing and validating input arguments
[ INFO ] Parsing input parameters
[Step 2/11] Loading Inference Engine
[ INFO ] InferenceEngine:
        API version ............ 2.1
        Build .................. custom_2021.1-offline_8104247c99b58491b3a7bf1da8a8b487c1c4461c
        Description ....... API
[ INFO ] Device info:
        GPU
        clDNNPlugin version ......... 2.1
        Build ........... custom_2021.1-offline_8104247c99b58491b3a7bf1da8a8b487c1c4461c

[Step 3/11] Setting device configuration
[ WARNING ] -nstreams default value is determined automatically for GPU device. Although the automatic selection usually provides a reasonable performance,but it still may be non-optimal for some cases, for more information look at README.
[Step 4/11] Reading network files
[ INFO ] Loading network files
[ INFO ] Read network took 95.45 ms
[Step 5/11] Resizing network to match image sizes and given batch
[ INFO ] Network batch size: 1
[Step 6/11] Configuring input of the model
[Step 7/11] Loading the model to the device
**[ ERROR ] Program build failed:**

root@Storage:/home/Openvino/bin# clinfo

  **Preferred work group size multiple              <getWGsizes:675: build program : error -6>**

```

pwilma, 2021-08-05:
If you get build program error even in clinfo it may indicate some problems with your setup like for example incompatible IGC component. -6 is generic CL_OUT_OF_HOST_MEMORY error so it doesn't say much. Could you please provide more details about your setup and installed software components? Did you use binary release of compute runtime or you built it from source? Strace log from clinfo run may also be useful.

pobin6, 2021-09-03:
I have solve the problem, it is due to the system, using ubuntu 18.04 is ok, in the custom linux system failed.

JacekDanecki, 2021-09-03:
Closing issue, as it's configuration problem.

closed 2021-09-03 by JacekDanecki