intel/compute-runtime #443: v21.23.20043 through v21.30.20482 fail to compile with LLVM 12's experimental pass manager

serebit, 2021-08-01:
If the LLVM used to compile compute-runtime is compiled with `-DENABLE_EXPERIMENTAL_NEW_PASS_MANAGER=ON`, building compute-runtime will fail around object 250-300 with the following error *spammed* in the logs:
```
Error: Required pass not found! Possible causes:
		- Pass misconfiguration (e.g.: missing macros)
		- Corruption of the global PassRegistry
```
This happens with versions v21.23.20043, v21.29.20389, and v21.30.20482. It can be assumed that all versions from v21.23.20043 exhibit this error, though they have not been tested. v21.22.19967 does not exhibit this error.

serebit, 2021-08-01:
The offending commit is ab52c70. Reversing this commit allows v21.30.20482 to build cleanly.

JacekDanecki, 2021-08-30:
I can't reproduce this on Debian bookworm/sid. I've compiled llvm/clang 12, then compiled gmmlib using default compiler (gcc 10.3.0).
IGC packages I've used from github release: igc-1.0.8365
I've compiled Neo 21.30.20482 and HEAD of Neo repository using compiled clang 12 without problems.
```
git log --oneline -1
13bea6987 (HEAD -> master, origin/master, origin/HEAD) Add missing pragma once
```

serebit, 2021-08-30:
@JacekDanecki And you enabled the new pass manager when compiling LLVM/Clang 12?

JacekDanecki, 2021-08-30:
Yes, I've used below command to configure llvm/clang
```
cmake -G Ninja .. -DENABLE_EXPERIMENTAL_NEW_PASS_MANAGER=ON -DCLANG_DEFAULT_LINKER="lld" -DCLANG_DEFAULT_OBJCOPY="llvm-objcopy"  -DLLVM_INSTALL_UTILS=ON -DLLVM_TARGETS_TO_BUILD="host;AMDGPU;BPF" -DLLVM_BUILD_LLVM_DYLIB=ON  -DLLVM_LINK_LLVM_DYLIB=ON -DCLANG_LINK_CLANG_DYLIB=ON -DLLVM_ENABLE_PROJECTS="clang;compiler-rt;lld;llvm"
```
and for Neo I've used
```
cmake -G Ninja -DSKIP_UNIT_TESTS=ON .. -DCMAKE_C_COMPILER=clang -DCMAKE_CXX_COMPILER=clang++
```

JacekDanecki, 2021-08-31:
As I've built llvm on Debian, I've not applied any patches for llvm/clang from Solus.

AdamCetnerowski, 2021-09-30:
Closing as inactive. Please re-open or open a new issue if relevant

closed 2021-09-30 by AdamCetnerowski