intel/compute-runtime #444: unable to install binary on skylake with ubuntu18.04

guoyejun, 2021-08-07:
my hardware is skylake:
00:02.0 VGA compatible controller [0300]: Intel Corporation HD Graphics 530 [8086:1912] (rev 06)

my system is: 
Ubuntu 18.04.5 LTS (GNU/Linux 4.15.0-117-generic x86_64)

I got the six .deb files from https://github.com/intel/compute-runtime/releases/tag/21.31.20514, and I have checked the sha256sum are correct.

but i'm unable to install them. How to install it? thanks.
```
neo$ sudo dpkg -i *.deb
(Reading database ... 239340 files and directories currently installed.)
Preparing to unpack intel-gmmlib_21.2.1_amd64.deb ...
Unpacking intel-gmmlib (21.2.1) over (21.2.1) ...
Preparing to unpack intel-igc-core_1.0.8173_amd64.deb ...
Unpacking intel-igc-core (1.0.8173) over (1.0.8173) ...
Preparing to unpack intel-igc-opencl_1.0.8173_amd64.deb ...
Unpacking intel-igc-opencl (1.0.8173) over (1.0.8173) ...
Preparing to unpack intel-level-zero-gpu_1.1.20514_amd64.deb ...
Unpacking intel-level-zero-gpu (1.1.20514) over (1.1.20514) ...
Preparing to unpack intel-ocloc_21.31.20514_amd64.deb ...
Unpacking intel-ocloc (21.31.20514) over (21.31.20514) ...
Preparing to unpack intel-opencl_21.31.20514_amd64.deb ...
Unpacking intel-opencl (21.31.20514) over (21.31.20514) ...
Setting up intel-gmmlib (21.2.1) ...
dpkg: dependency problems prevent configuration of intel-igc-core:
 libigc (1.0.4882-1~ppa1~bionic1) breaks intel-igc-core and is installed.

dpkg: error processing package intel-igc-core (--install):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of intel-igc-opencl:
 intel-igc-opencl depends on intel-igc-core (= 1.0.8173); however:
  Package intel-igc-core is not configured yet.
 libigc (1.0.4882-1~ppa1~bionic1) breaks intel-igc-opencl and is installed.

dpkg: error processing package intel-igc-opencl (--install):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of intel-level-zero-gpu:
 intel-level-zero-gpu depends on intel-igc-opencl (= 1.0.8173); however:
  Package intel-igc-opencl is not configured yet.

dpkg: error processing package intel-level-zero-gpu (--install):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of intel-ocloc:
 intel-ocloc depends on intel-igc-opencl (= 1.0.8173); however:
  Package intel-igc-opencl is not configured yet.

dpkg: error processing package intel-ocloc (--install):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of intel-opencl:
 intel-opencl depends on intel-igc-opencl (= 1.0.8173); however:
  Package intel-igc-opencl is not configured yet.

dpkg: error processing package intel-opencl (--install):
 dependency problems - leaving unconfigured
Processing triggers for libc-bin (2.27-3ubuntu1.2) ...
Errors were encountered while processing:
 intel-igc-core
 intel-igc-opencl
 intel-level-zero-gpu
 intel-ocloc
 intel-opencl
```

JacekDanecki, 2021-08-09:
Before installing packages from github releases, please remove installed packages from launchpad.

guoyejun, 2021-08-09:
thanks, I remembered that I once met the similar issue. Just curious why we provide different package names at different places for the same functionality.

This time, I run the following commands, but clinfo does not show GPU platform.
```
$ clinfo -l
Platform #0: Intel(R) FPGA Emulation Platform for OpenCL(TM)
 `-- Device #0: Intel(R) FPGA Emulation Device
Platform #1: Intel(R) OpenCL
 `-- Device #0: Intel(R) Core(TM) i5-6500 CPU @ 3.20GHz

$ sudo apt-get remove intel-gmmlib intel-igc-core intel-igc-opencl intel-level-zero-gpu intel-ocloc intel-opencl libigc-dev libigc
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages will be REMOVED:
  intel-gmmlib intel-igc-core intel-igc-opencl intel-level-zero-gpu intel-ocloc intel-opencl libigc libigc-dev
0 upgraded, 0 newly installed, 8 to remove and 378 not upgraded.
5 not fully installed or removed.
After this operation, 638 MB disk space will be freed.
Do you want to continue? [Y/n] y
(Reading database ... 239339 files and directories currently installed.)
Removing intel-opencl (21.31.20514) ...
Removing intel-ocloc (21.31.20514) ...
Removing intel-level-zero-gpu (1.1.20514) ...
Removing intel-gmmlib (21.2.1) ...
Removing libigc-dev (1.0.4882-1~ppa1~bionic1) ...
Removing libigc (1.0.4882-1~ppa1~bionic1) ...
Removing intel-igc-opencl (1.0.8173) ...
Removing intel-igc-core (1.0.8173) ...
dpkg: warning: while removing intel-igc-core, directory '/usr/local/lib' not empty so not removed
Processing triggers for libc-bin (2.27-3ubuntu1.2) ...

$ sudo dpkg -i *.deb
Selecting previously unselected package intel-gmmlib.
(Reading database ... 239180 files and directories currently installed.)
Preparing to unpack intel-gmmlib_21.2.1_amd64.deb ...
Unpacking intel-gmmlib (21.2.1) ...
Selecting previously unselected package intel-igc-core.
Preparing to unpack intel-igc-core_1.0.8173_amd64.deb ...
Unpacking intel-igc-core (1.0.8173) ...
Selecting previously unselected package intel-igc-opencl.
Preparing to unpack intel-igc-opencl_1.0.8173_amd64.deb ...
Unpacking intel-igc-opencl (1.0.8173) ...
Selecting previously unselected package intel-level-zero-gpu.
Preparing to unpack intel-level-zero-gpu_1.1.20514_amd64.deb ...
Unpacking intel-level-zero-gpu (1.1.20514) ...
Selecting previously unselected package intel-ocloc.
Preparing to unpack intel-ocloc_21.31.20514_amd64.deb ...
Unpacking intel-ocloc (21.31.20514) ...
Selecting previously unselected package intel-opencl.
Preparing to unpack intel-opencl_21.31.20514_amd64.deb ...
Unpacking intel-opencl (21.31.20514) ...
Setting up intel-gmmlib (21.2.1) ...
Setting up intel-igc-core (1.0.8173) ...
Setting up intel-igc-opencl (1.0.8173) ...
Setting up intel-level-zero-gpu (1.1.20514) ...
Setting up intel-ocloc (21.31.20514) ...
Setting up intel-opencl (21.31.20514) ...
Processing triggers for libc-bin (2.27-3ubuntu1.2) ...

$ clinfo -l
Platform #0: Intel(R) FPGA Emulation Platform for OpenCL(TM)
 `-- Device #0: Intel(R) FPGA Emulation Device
Platform #1: Intel(R) OpenCL
 `-- Device #0: Intel(R) Core(TM) i5-6500 CPU @ 3.20GHz

```

btw, I have installed the oneapi package (l_BaseKit_p_2021.3.0.3219_offline.sh) in the system, not sure if it matters.

JacekDanecki, 2021-08-09:
Can you share file log.txt created by command:
```
strace -f -s 255 -o log.txt clinfo -l
```

guoyejun, 2021-08-09:
[log.txt](https://github.com/intel/compute-runtime/files/6955124/log.txt)

attached, thanks.

JacekDanecki, 2021-08-09:
Here is issue:
```
7920  openat(AT_FDCWD, "/dev/dri/by-path/pci-0000:00:02.0-render", O_RDWR) = -1 EACCES (Permission denied)
```
User you are using has not access to files in /dev/dri directory. See: https://github.com/intel/compute-runtime#installation-options

guoyejun, 2021-08-10:
thanks, it works by adding my name into video group.

closed 2021-08-10 by guoyejun