intel/compute-runtime #445: `clEnqueueNDRangeKernel` does not return `CL_INVALID_WORK_DIMENSION` when its `work_dim` argument is greater than `CL_DEVICE_MAX_WORK_ITEM_DIMENSIONS`

gspr, 2021-08-07:
Querying `CL_DEVICE_MAX_WORK_ITEM_DIMENSIONS` on my device (Intel(R) Graphics Gen9 [0x3e9b]) gives me `3`. Yet, if I call `clEnqueueNDRangeKernel` (with a work queue for that device) with the third argument (`cl_uint work_dim`) equal to `4`, the call succeeds with `CL_SUCCESS`. The relevant kernel produces nonsensical results.

From the documentation, I would expect this call to fail with `CL_INVALID_WORK_DIMENSION`. Indeed, this is the case when I use a device from another OpenCL platform (e.g. POCL).

AdamCetnerowski, 2021-08-12:
Thank you for reporting. We will correct this in the near future.

ddabek-i, 2021-12-31:
Hi @gspr ,
Unfortunately the issue didn't reproduce for me.

If it's still affecting you, could you please provide some more details?

- operating system (and which version/distro)
- graphics driver version

They will help in tracking down this bug.

Thanks!

gspr, 2021-12-31:
I received an email notification that someone (simply named "Dominik") requested that I help reproduce the issue. However, when I go to this issue page, I don't see the comment. Am I missing something?

ddabek-i, 2021-12-31:
> I received an email notification that someone (simply named "Dominik") requested that I help reproduce the issue. However, when I go to this issue page, I don't see the comment. Am I missing something?

Sorry, I mistakenly thought that this did not reproduce.

ddabek-i, 2022-01-03:
Pull request resolving this issue https://github.com/intel/compute-runtime/commit/6c40c712d855fb6070ce2a69a0af49421001fcd3

AdamCetnerowski, 2022-01-13:
Closing as resolved

closed 2022-01-13 by AdamCetnerowski