intel/compute-runtime #446: clinfo unable to run on ADL-S with device id 0x4690

VUNNYSODHI, 2021-08-16:
**Environment:**
OS: Yocto 
ADL-S device: 0x4690
Kernel version: linux-intel-ese-lts-5.10
GuC: tgl_guc_51.0.0.bin

**Steps to reproduce issue**
./clinfo

**Observation:** 
root@intel-corei7-64:~# ./clinfo
Number of platforms                               0

When we change platform to ADL-S with device id 4680 it able to display opencl properties properly.

VUNNYSODHI, 2021-08-16:
0x4690 device id is exposed by Linux Kernel i915 graphics driver https://cgit.freedesktop.org/drm-tip/tree/include/drm/i915_pciids.h#n645 

Suspect it could be missing device id from compute-runtime.

Please let me know if any more inputs needed.

AdamCetnerowski, 2021-08-19:
We have a fix in-flight.

VUNNYSODHI, 2021-08-19:
Thanks @AdamCetnerowski for the update

JacekDanecki, 2021-08-20:
Device ID 0x4690 was added with commit: https://github.com/intel/compute-runtime/commit/4461fcdc2a0d28cab8c80b6cfc3a11baef707acc
Closing an issue, please create new one if you find any problems.

closed 2021-08-20 by JacekDanecki