intel/compute-runtime #449: cpu too high

pobin6, 2021-09-03:
in th file: ./shared/source/command_stream/command_stream_receiver.cpp
```C++
    while (*getTagAddress() < taskCountToWait && timeDiff <= timeoutMicroseconds) {
        std::this_thread::yield();
        CpuIntrinsics::pause();

        if (enableTimeout) {
            time2 = std::chrono::high_resolution_clock::now();
            timeDiff = std::chrono::duration_cast<std::chrono::microseconds>(time2 - time1).count();
        }
    }
```
it waste a lot of cpu time in waiting result, can not use callback or interrupt?

BartoszDunajski, 2021-09-07:
This function should fallback into callback/interrupt mode after timeout (depending on platform).
To enable that mode by default, you can create your command queue with CL_QUEUE_THROTTLE_LOW_KHR or use PowerSavingMode=1 debug variable.

JacekDanecki, 2021-09-07:
Closing issue, as it's “expected behavior.

closed 2021-09-07 by JacekDanecki