intel/compute-runtime #450: After installed compute-runtime, DPC++ cannot find device

xiaoyun2, 2021-09-05:
I'm using oneAPI Explicit SIMD Extension.

I'm using [dpcpp_compiler](https://github.com/intel/llvm/commit/27c632e38dba2820b326fdd7531a2788ddbe306f) to compiler and run the offical example [vadd_usm.cpp](https://github.com/intel/llvm-test-suite/blob/intel/SYCL/ESIMD/vadd_usm.cpp).

compiling:
`clang++ -fsycl vadd_usm.cpp`
running:
`SYCL_DEVICE_FILTER=opencl:gpu ./a.out`

And I get the following runtime error message:
`terminate called after throwing an instance of 'cl::sycl::runtime_error' what(): No device of requested type available. -1 (CL_DEVICE_NOT_FOUND) Aborted (core dumped)`

jandres742, 2021-09-05:
@xiaoyun2 : that means that there's something missing in the environment, like a missing package or no GPU at all. Please confirm first that clinfo displays the available GPUs.

xiaoyun2, 2021-09-06:
@jandres742 Thanks. I'm sure I have one GPU. My machine is i7-7567U with Intel® Iris® Plus Graphics 650. But `clinfo` cannot list available GPU device.

jandres742, 2021-09-06:
is this Windows or Linux? Where are you installing your packages from?

xiaoyun2, 2021-09-06:
The OS is Ubuntu 18.04. I installed the releases 21.34.20767(1638554).

JacekDanecki, 2021-09-06:
Could you run below command, and attach created file log.txt 
```
strace -s 255 -f -o log.txt clinfo
```

xiaoyun2, 2021-09-06:
@JacekDanecki Thanks, I try and it shows no platforms.
```
:~$ strace -s 255 -f -o log.txt clinfo
Number of platforms                               0
```

alalek, 2021-09-06:
> and attach created file log.txt

@xiaoyun2 `log.txt` file is necessary for further investigation.

xiaoyun2, 2021-09-06:
Thanks, Here is `log.txt`. [log.txt](https://github.com/intel/compute-runtime/files/7118035/log.txt)

alalek, 2021-09-10:
Thanks!
I believe this problem comes from here:

```
13506 openat(AT_FDCWD, "/opt/intel/mediasdk/lib64/libigdgmm.so.11", O_RDONLY|O_CLOEXEC) = 4
```

> mediasdk

The right path should be from `/usr/lib/x86_64-linux-gnu/intel-opencl`.

There are several workaronds how to force correct library loading:
- use `LD_LIBRARY_PATH=/usr/lib/x86_64-linux-gnu/intel-opencl:$LD_LIBRARY_PATH clinfo`
- or remove `mediasdk` from system or rename its file in `/etc/ld.so.conf.d/` and run `ldconfig` (under root, be careful, use backups). To use mediasdk you would need to use `LD_LIBRARY_PATH`

---

It is unlikely to run modern Intel OpenCL runtime and outdated MediaSDK in one application (due to misaligned gmm versions in their packages). Consider migration MediaSDK -> oneVPL.

xiaoyun2, 2021-09-15:
@alalek Thanks very much for checking! I think the reason you said is right. And my `libigdgmm.so.11` is under `/usr/local/lib`. When change to use the library under this path, it work!

closed 2021-09-16 by JacekDanecki