intel/compute-runtime #452: [Q&A]On SLM usage with multi work groups

GHGmc2, 2021-09-09:
Take SIMD8 as an example, the max number of work groups on a DualSubSlice(DSS) is: (16EU * 8 Thread * SIMD8 / wg_size 512) = 2.
Say a DSS has 64KB SLM, and a workgroup needs 40KB SLM. Does the runtime know the max number of workgroups can be scheduled on an DSS is 1 (because 40 *2 > 64), or the developer must make sure not to use SLM size larger than 32KB for one workgroup on the application side?

MichalMrozek, 2021-09-16:
Each of your workgroup takes 40KB of SLM.
So we cannot run more then 1 of those in one DSS.
Hardware knows this and will not schedule work until it has free slots.
So until this workgroup that takes 40KB leaves DSS, nothing else will be scheduled there.

So as developer you may not care about SLM usage in the workgroup as long as you do not cross 64KB per workgroup.

But that creates a performance problem, as with high SLM usage the number of workgroup that may run concurrently is limited by the SLM size. And this is what you need to take into consideration in your application.

GHGmc2, 2021-09-16:
@MichalMrozek thanks for the explanation!

JacekDanecki, 2021-09-16:
Closing issue as answer was provided.

closed 2021-09-16 by JacekDanecki