intel/compute-runtime #453: DG1 not recognized/supported on upstream kernel that does support DG1

eero-t, 2021-09-10:
I'm building latest Git version of public "drm-tip" kernel which does include DG1 support.

However, `clinfo` and `zello_sysman` do not work with DG1 when using latest "compute-runtime" release (or master branch HEAD):
```
$ clinfo
Number of platforms
$ zello_sysman -f
ZE_RESULT_ERROR_UNINITIALIZED returned by zeInit(ZE_INIT_FLAG_GPU_ONLY): getDeviceHandles: 110
```
Although:
*  Mesa Git version works fine on DG1 with the same public "drm-tip" kernel
* same `clinfo` / `zello_sysman` / "compute-runtime" stack works fine when used on top of internal kernel (instead of public "drm-tip" one)

 From latter, it seems that either:
* public kernel (i915 driver) is still missing some basic DG1 support needed by compute-runtime, or
* "compute-runtime" is using wrong kernel API(s)

Please clarify what is needed for DG1 to work with public kernel.

Note: according to `dmesg`, "drm-tip" kernel does not support GuC on DG1 yet, but AFAIK that should affect only GPU scheduling and loading of media FW.

PS. In case it matters, I'm building "compute-runtime" and underlying components with following options:
* gmmlib: `-DGMM_GFX_GEN=120`
* VC: -
* L0: -
* IGC: `-DIGC_OPTION__LLVM_PREFERRED_VERSION=11 -DINSTALL_SPIRVDLL=0`
* compute-runtime: `-DSUPPORT_GEN8=0 -DSUPPORT_GEN9=1 -DSUPPORT_GEN11=1 
 -DSUPPORT_GEN12LP=1 -DSUPPORT_DG1=1`

JacekDanecki, 2021-09-13:
Could you provide files created by below commands executed with drm-tip kernel:
```
strace -f -e ioctl -s 255 -o ioctl.log clinfo
strace -f -s 255 -o clinfo.log clinfo
```
and dmesg log after enabling drm debug using commands as root:
```
dmesg -C
echo 15 > /sys/module/drm/parameters/debug 
clinfo
dmesg
echo 0 > /sys/module/drm/parameters/debug 
```

eero-t, 2021-09-17:
Strace of IOCTL calls with yesterday drm-tip kernel HEAD:
```
strace -f -e openat,ioctl -s 255 clinfo | grep -v ENOENT
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libOpenCL.so.1", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libdl.so.2", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/etc/OpenCL/vendors", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 3
openat(AT_FDCWD, "/etc/OpenCL/vendors/intel.icd", O_RDONLY) = 4
openat(AT_FDCWD, "/usr/local/lib/intel-opencl/libigdrcl.so", O_RDONLY|O_CLOEXEC) = 4
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 4
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libpthread.so.0", O_RDONLY|O_CLOEXEC) = 4
openat(AT_FDCWD, "/usr/local/lib/libigdgmm.so.11", O_RDONLY|O_CLOEXEC) = 4
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libstdc++.so.6", O_RDONLY|O_CLOEXEC) = 4
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libm.so.6", O_RDONLY|O_CLOEXEC) = 4
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libgcc_s.so.1", O_RDONLY|O_CLOEXEC) = 4
openat(AT_FDCWD, "/dev/dri/renderD128", O_RDWR) = 4
ioctl(4, DRM_IOCTL_VERSION, 0x7ffee528c680) = 0
openat(AT_FDCWD, "/dev/dri/renderD129", O_RDWR) = 5
ioctl(5, DRM_IOCTL_VERSION, 0x7ffee528c680) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7ffee528c7c0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7ffee528c7c0) = 0
ioctl(4, DRM_IOCTL_I915_QUERY, 0x7ffee528c6d0) = 0
ioctl(4, DRM_IOCTL_I915_QUERY, 0x7ffee528c6d0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7ffee528c7c0) = 0
ioctl(4, DRM_IOCTL_I915_GEM_CONTEXT_SETPARAM, 0x7ffee528c7f0) = -1 EINVAL (Invalid argument)
ioctl(4, DRM_IOCTL_I915_QUERY, 0x7ffee528c750) = 0
ioctl(4, DRM_IOCTL_I915_QUERY, 0x7ffee528c750) = 0
ioctl(4, DRM_IOCTL_I915_QUERY, 0x7ffee528c750) = 0
ioctl(4, DRM_IOCTL_I915_QUERY, 0x7ffee528c750) = 0
ioctl(4, DRM_IOCTL_I915_GEM_VM_CREATE, 0x7ffee528c7b0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7ffee528c2c0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7ffee528c2c0) = 0
ioctl(4, DRM_IOCTL_I915_QUERY, 0x7ffee528c230) = 0
ioctl(4, DRM_IOCTL_I915_QUERY, 0x7ffee528c230) = 0
ioctl(4, DRM_IOCTL_I915_GEM_CONTEXT_GETPARAM, 0x7ffee528c2f0) = 0
openat(AT_FDCWD, "/sys/bus/pci/devices/0000:03:00.0/drm", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 6
openat(AT_FDCWD, "/sys/bus/pci/devices/0000:03:00.0/drm/card0/gt_max_freq_mhz", O_RDONLY) = 6
ioctl(4, DRM_IOCTL_I915_GEM_CONTEXT_GETPARAM, 0x7ffee528c2e0) = 0
ioctl(4, DRM_IOCTL_I915_GEM_CONTEXT_GETPARAM, 0x7ffee528c2f0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7ffee528c2a0) = 0
ioctl(5, DRM_IOCTL_I915_GETPARAM, 0x7ffee528c7c0) = 0
ioctl(5, DRM_IOCTL_I915_GETPARAM, 0x7ffee528c7c0) = 0
ioctl(5, DRM_IOCTL_I915_QUERY, 0x7ffee528c6d0) = 0
ioctl(5, DRM_IOCTL_I915_QUERY, 0x7ffee528c6d0) = 0
ioctl(5, DRM_IOCTL_I915_GETPARAM, 0x7ffee528c7c0) = 0
ioctl(5, DRM_IOCTL_I915_GEM_CONTEXT_SETPARAM, 0x7ffee528c7f0) = -1 EINVAL (Invalid argument)
ioctl(5, DRM_IOCTL_I915_QUERY, 0x7ffee528c750) = 0
ioctl(5, DRM_IOCTL_I915_QUERY, 0x7ffee528c750) = 0
ioctl(5, DRM_IOCTL_I915_QUERY, 0x7ffee528c750) = 0
ioctl(5, DRM_IOCTL_I915_QUERY, 0x7ffee528c750) = 0
ioctl(5, DRM_IOCTL_I915_GEM_VM_CREATE, 0x7ffee528c7b0) = 0
ioctl(5, DRM_IOCTL_I915_GETPARAM, 0x7ffee528c2c0) = 0
ioctl(5, DRM_IOCTL_I915_GETPARAM, 0x7ffee528c2c0) = 0
ioctl(5, DRM_IOCTL_I915_QUERY, 0x7ffee528c230) = 0
ioctl(5, DRM_IOCTL_I915_QUERY, 0x7ffee528c230) = 0
ioctl(5, DRM_IOCTL_I915_GEM_CONTEXT_GETPARAM, 0x7ffee528c2f0) = 0
openat(AT_FDCWD, "/sys/bus/pci/devices/0000:0a:00.0/drm", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 6
openat(AT_FDCWD, "/sys/bus/pci/devices/0000:0a:00.0/drm/card1/gt_max_freq_mhz", O_RDONLY) = 6
ioctl(5, DRM_IOCTL_I915_GEM_CONTEXT_GETPARAM, 0x7ffee528c2e0) = 0
ioctl(5, DRM_IOCTL_I915_GEM_CONTEXT_GETPARAM, 0x7ffee528c2f0) = 0
ioctl(5, DRM_IOCTL_I915_GETPARAM, 0x7ffee528c2a0) = 0
strace: Process 10 attached
[pid     9] ioctl(4, DRM_IOCTL_I915_GEM_USERPTR, 0x7ffee528c7d0) = 0
[pid     9] ioctl(5, DRM_IOCTL_I915_GEM_USERPTR, 0x7ffee528c7d0) = 0
[pid     9] openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 6
[pid     9] ioctl(4, DRM_IOCTL_I915_GEM_CONTEXT_CREATE_EXT, 0x7ffee528c560) = 0
[pid     9] ioctl(4, DRM_IOCTL_I915_GEM_CONTEXT_SETPARAM, 0x7ffee528c570) = 0
[pid     9] ioctl(4, DRM_IOCTL_I915_GEM_CONTEXT_SETPARAM, 0x7ffee528c580) = 0
[pid     9] ioctl(4, _IOC(_IOC_READ|_IOC_WRITE, 0x64, 0x5b, 0x18), 0x7ffee528c350) = 0
[... 96 lines cut ...]
[ 2327.645615] [drm:drm_mode_object_get] OBJ ID: 350 (3)
[ 2327.645617] [drm:drm_mode_object_get] OBJ ID: 350 (4)
[ 2327.645619] [drm:drm_mode_object_get] OBJ ID: 349 (1)
[ 2327.645630] [drm:drm_mode_object_get] OBJ ID: 350 (5)
[ 2327.645631] [drm:drm_mode_object_put.part.0] OBJ ID: 350 (6)
[ 2327.645634] [drm:drm_mode_object_get] OBJ ID: 320 (4)
[ 2327.645635] [drm:drm_mode_object_get] OBJ ID: 320 (5)
[ 2327.645637] [drm:drm_mode_object_put.part.0] OBJ ID: 320 (5)
[ 2327.645638] [drm:drm_mode_object_get] OBJ ID: 320 (4)
[ 2327.645648] [drm:drm_mode_object_put.part.0] OBJ ID: 350 (5)
[ 2327.645650] [drm:drm_mode_object_get] OBJ ID: 350 (4)
[ 2327.645689] i915 0000:03:00.0: [drm:drm_vblank_enable] enabling vblank on crtc 0, ret: 0
[ 2327.654411] i915 0000:03:00.0: [drm:drm_handle_vblank] vblank event on 139341, current 139341
[ 2327.654426] [drm:drm_mode_object_put.part.0] OBJ ID: 320 (5)
[ 2327.654429] [drm:drm_mode_object_put.part.0] OBJ ID: 320 (4)
[ 2327.654430] [drm:drm_mode_object_put.part.0] OBJ ID: 349 (2)
[... cut ...]
[ 2327.654433] [drm:drm_mode_object_put.part.0] OBJ ID: 350 (4)
[ 2327.654436] [drm:drm_lastclose] driver lastclose completed
[ 2327.654455] [drm:drm_ioctl] comm="clinfo" pid=47807, dev=0xe281, auth=0, I915_GEM_VM_DESTROY
[ 2327.654474] [drm:drm_file_free.part.0] comm="clinfo", pid=47807, dev=0xe281, open_count=1
[ 2327.654479] [drm:drm_lastclose] 
[ 2327.654480] [drm:drm_lastclose] driver lastclose completed
[ 2327.671072] i915 0000:03:00.0: [drm:vblank_disable_fn] disabling vblank on crtc 0
```

eero-t, 2021-09-17:
Full strace output is attached: [clinfo-strace.txt.gz](https://github.com/intel/compute-runtime/files/7187540/clinfo-strace.txt.gz)

"clinfo" in same container works fine on TGL with the same kernel and container options:
`docker run -it --cap-drop ALL --network none --user nobody:109 --device /dev/dri:/dev/dri:rw git-driver-image:latest  clinfo`

(109 is the group ID allowing access to render device files on my machines.)

smilczar, 2021-09-22:
From the dmesg log provided above this failure is relevant,
because it disallows retrieving a valid offset to mmap object specified by handle:
_[ 2327.645141] [drm:drm_ioctl] comm="clinfo" pid=47807, dev=0xe280, auth=0, I915_GEM_MMAP_OFFSET
[ 2327.645143] [drm:drm_ioctl] comm="clinfo", pid=47807, ret=-19_

Looking at the changes in uAPI between upstream and internal kernel,
this interface introduced a new flag `I915_MMAP_OFFSET_FIXED for discrete platforms:

struct drm_i915_gem_mmap_offset {
	/** @handle: Handle for the object being mapped. */
	__u32 handle;
	/** @pad: Must be zero */
	__u32 pad;
	/**
	* @offset: The fake offset to use for subsequent mmap call
	*
	* This is a fixed-size type for 32/64 compatibility.
	*/
	__u64 offset;
 
	/**
	* @flags: Flags for extended behaviour.
	*
	* It is mandatory that one of the `MMAP_OFFSET` types
	* should be included:
	*
	* - `I915_MMAP_OFFSET_GTT`: Use mmap with the object bound to GTT. (Write-Combined)
	* - `I915_MMAP_OFFSET_WC`: Use Write-Combined caching.
	* - `I915_MMAP_OFFSET_WB`: Use Write-Back caching.
	*** - `I915_MMAP_OFFSET_FIXED`: Use object placement to determine caching.
	*
	* On devices with local memory `I915_MMAP_OFFSET_FIXED` is the only valid
	* type. On devices without local memory, this caching mode is invalid.
	***
	* As caching mode when specifying `I915_MMAP_OFFSET_FIXED`, WC or WB will
	* be used, depending on the object placement on creation. WB will be used
	* when the object can only exist in system memory, WC otherwise.
	*/
	__u64 flags;

From the description of the flags for the I915_GEM_MMAP_OFFSET interface it follows that
we should use MMAP_OFFSET_FIXED on discrete platforms on upstream and MMAP_OFFSET_WC/WB on integrated.

eero-t, 2021-09-22:
Or rather, use MMAP_OFFSET_FIXED, and fallback to other only if former fails, checked e.g. on first call?

(Logic should be based on available features, not kernel versions as features can be backported etc.)

smilczar, 2021-09-22:
Yes, the option is to try to use MMAP_OFFSET_FIXED and, if that fails, fall back to MMAP_OFFSET_WB.
This is what we're usually do to address KMD changes on uAPI and give a support for both kernels.

eero-t, 2021-09-29:
Happens also with yesterday drm-tip kernel (which can now load DG1 GuC + HuC).

This is when using few releases old "compute-runtime".  I cannot test latest "compute-runtime" releases due to https://github.com/intel/compute-runtime/issues/455.

smilczar, 2021-10-04:
We have identified more differences in uAPI between DG1 published and upstream headers:
1.	No definition of GEM_MMAP_OFFSET_FIXED flag in DG1 headers
2.	Different extensions used for I915_GEM_CREATE_EXT (I915_GEM_CREATE_EXT_SETPARAM  vs. I915_GEM_CREATE_EXT_MEMORY_REGIONS)
3.	The same DRM_I915_QUERY_MEMORY_REGIONS but different layout of the structure drm_i915_memory_region_info
4.	Different definitions of DRM_IOCTL_I915_GEM_CREATE_EXT & __I915_EXEC_UNKNOWN_FLAGS

The runtime support for the above differences has been introduced in ci-neo-21184.

eero-t, 2021-10-04:
Thanks!  I'll probably test it only when the next compute-runtime version is tagged/released though.

FYI: there's related work being done also in media-driver, see: https://github.com/intel/media-driver/pull/1269

PS. Some things in drm-tip *may* still change before they end up in Linus' tree.  I'm testing only drm-tip, not official upstream kernel tree. You may want to verify from i915 maintainer(s) whether they expect all the uPI changes for dGPUs to be stable now (media-driver developers may have already checked that).

smilczar, 2021-10-05:
Yes, currently we managed to create a single OCL driver working on both kernels.
Media is going to have 2 separate driver build paths – one for upstream, another for “production” kernel. 

We will see how the things go further….