intel/compute-runtime #455: Builds are broken for last two releases (ones after 21.35.20826)

eero-t, 2021-09-17:
Building works for following versions:
 - GMMlib: intel-gmmlib-21.3.1
 - VC: HEAD
 - IGC: igc-1.0.8517
 - level-zero: v1.5.0
 - compute-runtime: 21.35.20826

i.e. latest releases except for compute-runtime.

But fails with two latest compute-runtime releases:
- 21.36.20889
- 21.37.20939

To following issue:
```
[2436/2507] Linking CXX executable bin/igdrcl_aub_tests
FAILED: bin/igdrcl_aub_tests 
: && /usr/bin/c++ -mindirect-branch-register -ftemplate-depth=1024 -Wall -Wempty-body -Wignored-qualifiers -Wtype-limits -Wuninitialized -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -Winvalid-pch -Wno-unknown-pragmas -Wno-unused-local-typedefs -Wno-unused-but-set-variable -Wclobbered -Wimplicit-fallthrough=4 -Wno-noexcept-type -Werror -fstack-protector-strong   -Wformat -Wformat-security -O3 -DNDEBUG -O0 -fuse-ld=gold   -rdynamic @CMakeFiles/igdrcl_aub_tests.rsp -o bin/igdrcl_aub_tests  && :
/usr/bin/ld.gold: error: cannot find -ligdgmm
opencl/test/unit_test/aub_tests/CMakeFiles/igdrcl_aub_tests.dir/__/__/__/__/shared/source/os_interface/linux/gmm_interface_linux.cpp.o:gmm_interface_linux.cpp:function NEO::GmmInterface::initialize(_GMM_INIT_IN_ARGS_*, _GMM_INIT_OUT_ARGS_*): error: undefined reference to 'InitializeGmm'
opencl/test/unit_test/aub_tests/CMakeFiles/igdrcl_aub_tests.dir/__/__/__/__/shared/source/os_interface/linux/gmm_interface_linux.cpp.o:gmm_interface_linux.cpp:function NEO::GmmInterface::destroy(_GMM_INIT_OUT_ARGS_*): error: undefined reference to 'GmmAdapterDestroy'
collect2: error: ld returned 1 exit status
[2437/2507] Building CXX object level_zero/tools/test/unit_tests/CMakeFiles/ze_intel_gpu_tools_tests.dir/sources/debug/debug_session_tests.cpp.o
[2438/2507] Linking CXX executable bin/ze_intel_gpu_aub_tests
FAILED: bin/ze_intel_gpu_aub_tests 
: && /usr/bin/c++ -mindirect-branch-register -ftemplate-depth=1024 -Wall -Wempty-body -Wignored-qualifiers -Wtype-limits -Wuninitialized -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -Winvalid-pch -Wno-unknown-pragmas -Wno-unused-local-typedefs -Wno-unused-but-set-variable -Wclobbered -Wimplicit-fallthrough=4 -Wno-noexcept-type -Werror -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -fpermissive -fPIC -O3 -DNDEBUG -fuse-ld=gold   -rdynamic @CMakeFiles/ze_intel_gpu_aub_tests.rsp -o bin/ze_intel_gpu_aub_tests  && :
/usr/bin/ld.gold: error: cannot find -ligdgmm
level_zero/core/test/aub_tests/CMakeFiles/ze_intel_gpu_aub_tests.dir/__/__/__/__/shared/source/os_interface/linux/gmm_interface_linux.cpp.o:gmm_interface_linux.cpp:function NEO::GmmInterface::initialize(_GMM_INIT_IN_ARGS_*, _GMM_INIT_OUT_ARGS_*): error: undefined reference to 'InitializeGmm'
level_zero/core/test/aub_tests/CMakeFiles/ze_intel_gpu_aub_tests.dir/__/__/__/__/shared/source/os_interface/linux/gmm_interface_linux.cpp.o:gmm_interface_linux.cpp:function NEO::GmmInterface::destroy(_GMM_INIT_OUT_ARGS_*): error: undefined reference to 'GmmAdapterDestroy'
collect2: error: ld returned 1 exit status
```

Ps. This is with following options (which have been working fine for long time):
```
-DCMAKE_BUILD_TYPE=Release
-DSUPPORT_GEN8=0 -DSUPPORT_GEN9=1 -DSUPPORT_GEN11=1
-DSUPPORT_GEN12LP=1 -DSUPPORT_DG1=1
-DDO_NOT_RUN_AUB_TESTS=1
-DDONT_CARE_OF_VIRTUALS=1
```
(I have no idea / remember any more what the last option means.)

JacekDanecki, 2021-09-20:
Is the gmmlib visible for linker? What is output from
```
ldconfig -p | grep igdgmm
```

eero-t, 2021-09-27:
> Is the gmmlib visible for linker?

media-driver is built before compute-runtime, and links "igdgmm" fine.  Compute-runtime itself claims to have found it:
```
$ cmake -LH -Wno-dev -G Ninja       -DCMAKE_INSTALL_PREFIX=${INSTALL_DIR} -DCMAKE_BUILD_TYPE=Release       -DSUPPORT_GEN8=0 -DSUPPORT_GEN9=1 -DSUPPORT_GEN11=1       -DSUPPORT_GEN12LP=1 -DSUPPORT_DG1=1       -DDO_NOT_RUN_AUB_TESTS=1 -DDONT_CARE_OF_VIRTUALS=1 && ninja
...
-- Driver model : drm_or_wddm
-- Using linker: ld.gold
-- Release build configuration
-- Computed OpenCL version major is: 21
-- Computed OpenCL version minor is: 37
...
-- Checking for module 'igc-opencl'
--   Found igc-opencl, version 1.0.1
-- IGC include dirs: /usr/local/include/igc;/usr/local/include/igc/cif;/usr/local/include/igc/ocl_igc_shared/executable_format;/usr/local/include/igc/ocl_igc_shared/device_enqueue
-- VISA Dir: /usr/local/include/visa
-- IGA Includes dir: /usr/local/include/iga
-- Checking for module 'igdgmm'
--   Found igdgmm, version 11.3.0
```

So problem with linking seems internal to compute-runtime (= issue in CMake files?).

NOTE: I do not get the same issue when testing latest releases (as of today) of the stack.

Latest "compute-runtime" (21.38.21026) fails to:
```
[280/2510] Generating ../../../bin/Gen11lp/0/test_files/x64/stateful_copy_buffer_Gen11lp.spv, ../../../bin/Gen11lp/0/test_files/x64/stateful_copy_buffer_Gen11lp.bin, ../../../bin/Gen11lp/0/test_files/x64/stateful_copy_buffer_Gen11lp.gen
FAILED: bin/Gen11lp/0/test_files/x64/stateful_copy_buffer_Gen11lp.spv bin/Gen11lp/0/test_files/x64/stateful_copy_buffer_Gen11lp.bin bin/Gen11lp/0/test_files/x64/stateful_copy_buffer_Gen11lp.gen 
cd /home/nobody/source/compute-runtime/opencl/test/unit_test/test_files && LD_LIBRARY_PATH=/home/nobody/source/compute-runtime/build/bin /home/nobody/source/compute-runtime/build/bin/ocloc -q -file stateful_copy_buffer.cl -device icllp -64 -revision_id 0 -out_dir /home/nobody/source/compute-runtime/build/bin/Gen11lp/0/test_files/x64/
```

JacekDanecki, 2021-10-04:
For gmmlib issue, can you add -DNEO_DISABLE_LD_GOLD=1 to cmake parameters.

In case of ULT failure, can you run ocloc manually without "-q" parameter, to see what is an issue
```
cd /home/nobody/source/compute-runtime/opencl/test/unit_test/test_files
LD_LIBRARY_PATH=/home/nobody/source/compute-runtime/build/bin /home/nobody/source/compute-runtime/build/bin/ocloc -file stateful_copy_buffer.cl -device icllp -64 -revision_id 0 -out_dir /home/nobody/source/compute-runtime/build/bin/Gen11lp/0/test_files/x64/
```
Adding LD_DEBUG=libs or LD_DEBUG=all can also help to debug this issue.

eero-t, 2021-10-04:
After running `ocloc` manually, I could see that issue to be bug on my part (I never remember that unlike ldconfig "-v" option, "-p" prevents ldconfig from doing its actual work, and its man page does not state it).

I can understand why debug messages are disabled in builds, and can see (rarely) a reason to disable warnings in builds, but why the heck you are disabling `ocloc` build error messages? And by *default*???

---

Thanks for the `-DNEO_DISABLE_LD_GOLD=1` option, that fixed compute-runtime build issue!

(Feel free to close this if you think that enough.)

Without that option, builds fail also with the new (21.39.x) release:
```
FAILED: bin/ze_intel_gpu_aub_tests 
: && /usr/bin/c++ -mindirect-branch-register -ftemplate-depth=1024 -Wall -Wempty-body -Wignored-qualifiers -Wtype-limits -Wuninitialized -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -Winvalid-pch -Wno-unknown-pragmas -Wno-unused-local-typedefs -Wno-unused-but-set-variable -Wclobbered -Wimplicit-fallthrough=4 -Wno-noexcept-type -Werror -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -fpermissive -fPIC -O3 -DNDEBUG -fuse-ld=gold   -rdynamic @CMakeFiles/ze_intel_gpu_aub_tests.rsp -o bin/ze_intel_gpu_aub_tests  && :
/usr/bin/ld.gold: error: cannot find -ligdgmm
level_zero/core/test/aub_tests/CMakeFiles/ze_intel_gpu_aub_tests.dir/__/__/__/__/shared/source/os_interface/linux/gmm_interface_linux.cpp.o:gmm_interface_linux.cpp:function NEO::GmmInterface::initialize(_GMM_INIT_IN_ARGS_*, _GMM_INIT_OUT_ARGS_*): error: undefined reference to 'InitializeGmm'
level_zero/core/test/aub_tests/CMakeFiles/ze_intel_gpu_aub_tests.dir/__/__/__/__/shared/source/os_interface/linux/gmm_interface_linux.cpp.o:gmm_interface_linux.cpp:function NEO::GmmInterface::destroy(_GMM_INIT_OUT_ARGS_*): error: undefined reference to 'GmmAdapterDestroy'
collect2: error: ld returned 1 exit status
[2429/2499] Linking CXX executable bin/igdrcl_aub_tests
FAILED: bin/igdrcl_aub_tests 
: && /usr/bin/c++ -mindirect-branch-register -ftemplate-depth=1024 -Wall -Wempty-body -Wignored-qualifiers -Wtype-limits -Wuninitialized -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -Winvalid-pch -Wno-unknown-pragmas -Wno-unused-local-typedefs -Wno-unused-but-set-variable -Wclobbered -Wimplicit-fallthrough=4 -Wno-noexcept-type -Werror -fstack-protector-strong   -Wformat -Wformat-security -O3 -DNDEBUG -O0 -fuse-ld=gold   -rdynamic @CMakeFiles/igdrcl_aub_tests.rsp -o bin/igdrcl_aub_tests  && :
/usr/bin/ld.gold: error: cannot find -ligdgmm
opencl/test/unit_test/aub_tests/CMakeFiles/igdrcl_aub_tests.dir/__/__/__/__/shared/source/os_interface/linux/gmm_interface_linux.cpp.o:gmm_interface_linux.cpp:function NEO::GmmInterface::initialize(_GMM_INIT_IN_ARGS_*, _GMM_INIT_OUT_ARGS_*): error: undefined reference to 'InitializeGmm'
opencl/test/unit_test/aub_tests/CMakeFiles/igdrcl_aub_tests.dir/__/__/__/__/shared/source/os_interface/linux/gmm_interface_linux.cpp.o:gmm_interface_linux.cpp:function NEO::GmmInterface::destroy(_GMM_INIT_OUT_ARGS_*): error: undefined reference to 'GmmAdapterDestroy'
collect2: error: ld returned 1 exit status
```
Note that above happens although I give CMake the `-DDO_NOT_RUN_AUB_TESTS=1` option.

JacekDanecki, 2021-10-05:
ocloc should print build error messages even "-q" parameter is provided. It looks like regression in ocloc.
At this moment you can't disable aub test building, only aub execution can be disabled by -DDO_NOT_RUN_AUB_TESTS=1

Closing an issue, as workaround was provided for gmmlib problem.

closed 2021-10-05 by JacekDanecki