intel/compute-runtime #460: zeModuleGetFunctionPointer sometimes cannot find function

AidanBeltonS, 2021-09-29:
**Issue:**
In some cases `zeModuleGetFunctionPointer` does not find function pointers. 
This comes from a PR for oneAPI's implementation of sycl: https://github.com/intel/llvm/pull/4599

It is possible to get a valid list of kernel names, however when querying for a function pointer which exists ZE_RESULT_ERROR_INVALID_ARGUMENT is returned. I have checked that all inputs are not nullptrs.

```
All Kernel Names:_ZTS5Kern1
Kernel Name we want to find:_ZTS5Kern1
reinterpret cast FunctionPointerRet: 0x7ffddfd48a48
ZE_RESULT_ERROR_INVALID_ARUGMENT is returned

All Kernel Names:_ZTS5Kern1
Kernel Name we want to find:_ZTS5Kern2
reinterpret cast FunctionPointerRet: 0x7ffddfd48a48
ZE_RESULT_ERROR_INVALID_ARUGMENT is returned

```

**Reproduction:**
Unfortunately I do not have a test case in straight level_zero. If you apply the below patch to https://github.com/AidanBeltonS/llvm/tree/feature-piextGetDeviceFunctionPointer. You can reproduce the error.
  
  ```
diff --git a/sycl/plugins/level_zero/pi_level_zero.cpp b/sycl/plugins/level_zero/pi_level_zero.cpp
index 2134b26651e7..7177337b581a 100644
--- a/sycl/plugins/level_zero/pi_level_zero.cpp
+++ b/sycl/plugins/level_zero/pi_level_zero.cpp
@@ -6248,6 +6248,29 @@ pi_result piextGetDeviceFunctionPointer(pi_device Device, pi_program Program,
     return PI_INVALID_PROGRAM_EXECUTABLE;
   }
 
+  {
+    // List all available kernels 
+    std::string PINames{""};
+    bool First = true;
+    _pi_program::ModuleIterator ModIt(Program);
+    while (!ModIt.Done()) {
+      uint32_t Count = 0;
+      ZE_CALL(zeModuleGetKernelNames, (*ModIt, &Count, nullptr));
+      std::unique_ptr<const char *[]> PNames(new const char *[Count]);
+      ZE_CALL(zeModuleGetKernelNames, (*ModIt, &Count, PNames.get()));
+      for (uint32_t I = 0; I < Count; ++I) {
+        PINames += (!First ? ";" : "");
+        PINames += PNames[I];
+        First = false;
+      }
+        ModIt++;
+    }
+    std::cout << "All Kernel Names:" << PINames << std::endl;
+    std::cout << "Kernel Name we want to find:" << FunctionName << std::endl;
+  }
+
+  std::cout << "reinterpret cast FunctionPointerRet: " << reinterpret_cast<void **>(FunctionPointerRet) << std::endl;
+
   // Search for the function name in each module.
   ze_result_t ZeResult = ZE_RESULT_ERROR_INVALID_FUNCTION_NAME;
   _pi_program::ModuleIterator ModIt(Program);
@@ -6255,6 +6278,8 @@ pi_result piextGetDeviceFunctionPointer(pi_device Device, pi_program Program,
     ZeResult = ZE_CALL_NOCHECK(
         zeModuleGetFunctionPointer,
         (*ModIt, FunctionName, reinterpret_cast<void **>(FunctionPointerRet)));
+    if (ZeResult == ZE_RESULT_ERROR_INVALID_ARGUMENT)
+      std::cout << "ZE_RESULT_ERROR_INVALID_ARUGMENT is returned\n";
     if (ZeResult != ZE_RESULT_ERROR_INVALID_FUNCTION_NAME)
       break;
     ModIt++;
  ```

Simplified test case from split-per-kernel.cpp simply creates one kernels then checks if it has one which is should posses 
and one that is shouldn't posses. `zeModuleGetFunctionPointer` is called for each kernel and the list of kernel names is also printed to show that the kernel can be seen.

  ```
 #include <CL/sycl.hpp>

class Kern1;
class Kern2;

int main() {
  cl::sycl::queue Q;
  int Data = 0;
  {
    cl::sycl::buffer<int, 1> Buf(&Data, cl::sycl::range<1>(1));
    cl::sycl::program Prg(Q.get_context());
    Prg.build_with_kernel_type<Kern1>();
    cl::sycl::kernel Krn = Prg.get_kernel<Kern1>();

    Prg.has_kernel<Kern1>();
    Prg.has_kernel<Kern2>();

    Q.submit([&](cl::sycl::handler &Cgh) {
      auto Acc = Buf.get_access<cl::sycl::access::mode::read_write>(Cgh);
      Cgh.single_task<Kern1>(Krn, [=]() { Acc[0] = 1; });
    });
  }
  assert(Data == 1);

  {
    cl::sycl::buffer<int, 1> Buf(&Data, cl::sycl::range<1>(1));
    cl::sycl::program Prg(Q.get_context());
    Prg.build_with_kernel_type<Kern2>();
    cl::sycl::kernel Krn = Prg.get_kernel<Kern2>();

    Q.submit([&](cl::sycl::handler &Cgh) {
      auto Acc = Buf.get_access<cl::sycl::access::mode::read_write>(Cgh);
      Cgh.single_task<Kern2>(Krn, [=]() { Acc[0] = 2; });
    });
  }
  assert(Data == 2);

  return 0;
}
  ```

The first kernel does exist inside the module however `zeModuleGetFunctionPointer` returns an invalid argument error.
The second kernel also does not exist inside the module and the same error is returned.
From the docs https://spec.oneapi.io/versions/0.5.0/oneL0/api/function_ze__module_8h_1a46e5f2e11a0e9a19e8f974e05e436dff.html
this could be because any of the inputs are nullptrs. The most likely case for a bad input would be `FunctionPointerRet` after being cast to void** however from the output this is not returning a nullptr.

Thanks in advance for looking at this issue.

jandres742, 2021-10-07:
Thanks @AidanBeltonS 

We will take a closer look internally and report back.

HoppeMateusz, 2021-10-08:
Hello,

are you trying to retrieve "kernel" address with zeModuleGetFunctionPointer() by passing kernel name ?
Or is it a function name ?

If this is a kernel name - function pointer will not be returned as kernel is not a normal function in the module and INVALID ARGUMENT error is correct in this case.

Pointers to functions can only be returned by zeModuleGetFunctionPointer().

Regards, 
Mateusz

HoppeMateusz, 2021-10-08:
Is there any reason to retrieve kernel pointer?  How is it going to be used?

you cannot call it from another kernel, as this is not a normal function.

jandres742, 2022-01-21:
@AidanBeltonS do the reply above answer your question?

AidanBeltonS, 2022-01-24:
Hi yes that does answer my question. 
Sorry for the late reply this fell off my radar. 

I think this was being used for two purposes, one was to pass functions as arguments, the other was to determine if a kernel existed. It is clear that this function is really only supposed to be used for the first application and not the second. 
I think this can be closed, many thanks.

closed 2022-01-24 by AidanBeltonS