intel/compute-runtime #461: Is OCL diver for TigerLake compatible with Nv driver?

FionaZZ92, 2021-10-08:
Customer meet a problem that install the OCL runtime on tigerlake, and then install nvidia driver. The intel OCL is not work. 

System: Ubuntu 20.04
The process is:
1. Install OpenCL 20.35.17767 on TGL(i7-1185G7E), add user to video/render group, ocl work
2. Insert Nv card, install 11.1 cuda driver, clinfo only show nv, no igpu, intel ocl not work
3. Break nvdia smi, cuda still remain, intel igpu appear in clinfo, and can work with intel ocl.

I would like to check if the intel OCL on TGL not compatible with nvidia gpu driver, is them any potential conflict on TGL?

JacekDanecki, 2021-10-08:
Can you provide log.txt file created by below command from system where Neo and Nvidia drivers are installed
```
strace -s 255 -f -o log.txt clinfo
```
On my system I can use both
```
clinfo -l
Platform #0: Intel(R) OpenCL HD Graphics
 `-- Device #0: Intel(R) HD Graphics 630 [0x591b]
Platform #1: NVIDIA CUDA
 `-- Device #0: Quadro M1200
```

FionaZZ92, 2021-10-11:
Hello Jacek, please check with attachments.

[lspci.txt](https://github.com/intel/compute-runtime/files/7319353/lspci.txt)
[dmesg.txt](https://github.com/intel/compute-runtime/files/7319354/dmesg.txt)
[lsmod.txt](https://github.com/intel/compute-runtime/files/7319356/lsmod.txt)

FionaZZ92, 2021-10-11:
I found root cause, i915 driver not run, can be solved by fixing BIOS. Thanks, can close the issue.