intel/compute-runtime #465: Packages have unmet deps

ionutboscuintel, 2021-10-18:
Installing the latest version for intel-opencl intel-level-zero-gpu  level-zero  level-zero-devel, leads to the following error (tested on intel/oneapi-basekit:devel-ubuntu18.04 container):
The following packages have unmet dependencies:
 intel-level-zero-gpu : Depends: intel-igc-opencl (= 1.0.8517) but 1.0.1629709536 is to be installed
 intel-opencl : Depends: intel-igc-opencl (= 1.0.8517) but 1.0.1629709536 is to be installed

JacekDanecki, 2021-11-02:
IGC in release 1.0.8517 has updated packages, see release notes: https://github.com/intel/intel-graphics-compiler/releases/tag/igc-1.0.8517

eero-t, 2021-11-05:
As to the packaging files in the repo...

Binary package deps for OpenCL seem to have been fixed:
https://github.com/intel/compute-runtime/blob/master/scripts/packaging/opencl/ubuntu_20.04/control

But level-zero one still depends on "intel-igc-opencl":
https://github.com/intel/compute-runtime/blob/master/scripts/packaging/l0_gpu_driver/ubuntu_20.04/control

And source package deps in above files still list "intel-igc-opencl-devel" as dependency.

(Repo seems to have no packaging files for 18.04 Ubuntu version, only for 20.04, and few other distros.)

JacekDanecki, 2021-11-08:
Since https://github.com/intel/compute-runtime/releases/tag/21.39.21127 we don't provide packages for Ubuntu 18.04 in Neo releases.

closed 2022-02-01 by JacekDanecki