intel/compute-runtime #471: Building failure in tests for 21.43.21438

ArchangeGabriel, 2021-10-30:
With exactly the same setup, previous release builds fine, but now:
```
/build/intel-compute-runtime/src/compute-runtime-21.43.21438/opencl/test/unit_test/os_interface/linux/local_memory_helper_tests_dg1.cpp:22:38: error: ‘givenDg1WhenCreateGemExtThenReturnCorrectValue’ has not been declared
   22 | DG1TEST_F(LocalMemoryHelperTestsDg1, givenDg1WhenCreateGemExtThenReturnCorrectValue) {
      |                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/build/intel-compute-runtime/src/compute-runtime-21.43.21438/opencl/test/unit_test/os_interface/linux/local_memory_helper_tests_dg1.cpp:22:1: error: ISO C++ forbids declaration of ‘DG1TEST_F’ with no type [-fpermissive]
   22 | DG1TEST_F(LocalMemoryHelperTestsDg1, givenDg1WhenCreateGemExtThenReturnCorrectValue) {
      | ^~~~~~~~~
/build/intel-compute-runtime/src/compute-runtime-21.43.21438/opencl/test/unit_test/os_interface/linux/local_memory_helper_tests_dg1.cpp:22:84: error: cannot declare parameter to be of abstract type ‘testing::Test’
   22 | DG1TEST_F(LocalMemoryHelperTestsDg1, givenDg1WhenCreateGemExtThenReturnCorrectValue) {
      |                                                                                    ^
In file included from /build/intel-compute-runtime/src/compute-runtime-21.43.21438/opencl/test/unit_test/os_interface/linux/drm_mock_exp.h:13,
                 from /build/intel-compute-runtime/src/compute-runtime-21.43.21438/opencl/test/unit_test/os_interface/linux/local_memory_helper_tests_dg1.cpp:14:
/build/intel-compute-runtime/src/compute-runtime-21.43.21438/third_party/gtest/gtest/gtest.h:14983:18: note:   because the following virtual functions are pure within ‘testing::Test’:
14983 | class GTEST_API_ Test {
      |                  ^~~~
/build/intel-compute-runtime/src/compute-runtime-21.43.21438/third_party/gtest/gtest/gtest.h:15062:16: note:     ‘virtual void testing::Test::TestBody()’
15062 |   virtual void TestBody() = 0;
      |                ^~~~~~~~
/build/intel-compute-runtime/src/compute-runtime-21.43.21438/opencl/test/unit_test/os_interface/linux/local_memory_helper_tests_dg1.cpp: In function ‘int DG1TEST_F(LocalMemoryHelperTestsDg1, int)’:
/build/intel-compute-runtime/src/compute-runtime-21.43.21438/opencl/test/unit_test/os_interface/linux/local_memory_helper_tests_dg1.cpp:42:1: error: no return statement in function returning non-void [-Werror=return-type]
   42 | }
      | ^
/build/intel-compute-runtime/src/compute-runtime-21.43.21438/opencl/test/unit_test/os_interface/linux/local_memory_helper_tests_dg1.cpp: At global scope:
/build/intel-compute-runtime/src/compute-runtime-21.43.21438/opencl/test/unit_test/os_interface/linux/local_memory_helper_tests_dg1.cpp:44:38: error: ‘givenDg1WithDrmTipWhenCreateGemExtWithDebugFlagThenPrintDebugInfo’ has not been declared
   44 | DG1TEST_F(LocalMemoryHelperTestsDg1, givenDg1WithDrmTipWhenCreateGemExtWithDebugFlagThenPrintDebugInfo) {
      |                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/build/intel-compute-runtime/src/compute-runtime-21.43.21438/opencl/test/unit_test/os_interface/linux/local_memory_helper_tests_dg1.cpp:44:1: error: ISO C++ forbids declaration of ‘DG1TEST_F’ with no type [-fpermissive]
   44 | DG1TEST_F(LocalMemoryHelperTestsDg1, givenDg1WithDrmTipWhenCreateGemExtWithDebugFlagThenPrintDebugInfo) {
      | ^~~~~~~~~
/build/intel-compute-runtime/src/compute-runtime-21.43.21438/opencl/test/unit_test/os_interface/linux/local_memory_helper_tests_dg1.cpp:44:103: error: cannot declare parameter to be of abstract type ‘testing::Test’
   44 | DG1TEST_F(LocalMemoryHelperTestsDg1, givenDg1WithDrmTipWhenCreateGemExtWithDebugFlagThenPrintDebugInfo) {
      |                                                                                                       ^
/build/intel-compute-runtime/src/compute-runtime-21.43.21438/opencl/test/unit_test/os_interface/linux/local_memory_helper_tests_dg1.cpp:44:1: error: redefinition of ‘int DG1TEST_F(LocalMemoryHelperTestsDg1, int)’
   44 | DG1TEST_F(LocalMemoryHelperTestsDg1, givenDg1WithDrmTipWhenCreateGemExtWithDebugFlagThenPrintDebugInfo) {
      | ^~~~~~~~~
/build/intel-compute-runtime/src/compute-runtime-21.43.21438/opencl/test/unit_test/os_interface/linux/local_memory_helper_tests_dg1.cpp:22:1: note: ‘int DG1TEST_F(LocalMemoryHelperTestsDg1, int)’ previously defined here
   22 | DG1TEST_F(LocalMemoryHelperTestsDg1, givenDg1WhenCreateGemExtThenReturnCorrectValue) {
      | ^~~~~~~~~
/build/intel-compute-runtime/src/compute-runtime-21.43.21438/opencl/test/unit_test/os_interface/linux/local_memory_helper_tests_dg1.cpp: In function ‘int DG1TEST_F(LocalMemoryHelperTestsDg1, int)’:
/build/intel-compute-runtime/src/compute-runtime-21.43.21438/opencl/test/unit_test/os_interface/linux/local_memory_helper_tests_dg1.cpp:66:1: error: no return statement in function returning non-void [-Werror=return-type]
   66 | }
      | ^
/build/intel-compute-runtime/src/compute-runtime-21.43.21438/opencl/test/unit_test/os_interface/linux/local_memory_helper_tests_dg1.cpp: At global scope:
/build/intel-compute-runtime/src/compute-runtime-21.43.21438/opencl/test/unit_test/os_interface/linux/local_memory_helper_tests_dg1.cpp:68:38: error: ‘givenDg1WhenCreateGemExtWithDebugFlagThenPrintDebugInfo’ has not been declared
   68 | DG1TEST_F(LocalMemoryHelperTestsDg1, givenDg1WhenCreateGemExtWithDebugFlagThenPrintDebugInfo) {
      |                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/build/intel-compute-runtime/src/compute-runtime-21.43.21438/opencl/test/unit_test/os_interface/linux/local_memory_helper_tests_dg1.cpp:68:1: error: ISO C++ forbids declaration of ‘DG1TEST_F’ with no type [-fpermissive]
   68 | DG1TEST_F(LocalMemoryHelperTestsDg1, givenDg1WhenCreateGemExtWithDebugFlagThenPrintDebugInfo) {
      | ^~~~~~~~~
/build/intel-compute-runtime/src/compute-runtime-21.43.21438/opencl/test/unit_test/os_interface/linux/local_memory_helper_tests_dg1.cpp:68:93: error: cannot declare parameter to be of abstract type ‘testing::Test’
   68 | DG1TEST_F(LocalMemoryHelperTestsDg1, givenDg1WhenCreateGemExtWithDebugFlagThenPrintDebugInfo) {
      |                                                                                             ^
/build/intel-compute-runtime/src/compute-runtime-21.43.21438/opencl/test/unit_test/os_interface/linux/local_memory_helper_tests_dg1.cpp:68:1: error: redefinition of ‘int DG1TEST_F(LocalMemoryHelperTestsDg1, int)’
   68 | DG1TEST_F(LocalMemoryHelperTestsDg1, givenDg1WhenCreateGemExtWithDebugFlagThenPrintDebugInfo) {
      | ^~~~~~~~~
/build/intel-compute-runtime/src/compute-runtime-21.43.21438/opencl/test/unit_test/os_interface/linux/local_memory_helper_tests_dg1.cpp:22:1: note: ‘int DG1TEST_F(LocalMemoryHelperTestsDg1, int)’ previously defined here
   22 | DG1TEST_F(LocalMemoryHelperTestsDg1, givenDg1WhenCreateGemExtThenReturnCorrectValue) {
      | ^~~~~~~~~
/build/intel-compute-runtime/src/compute-runtime-21.43.21438/opencl/test/unit_test/os_interface/linux/local_memory_helper_tests_dg1.cpp: In function ‘int DG1TEST_F(LocalMemoryHelperTestsDg1, int)’:
/build/intel-compute-runtime/src/compute-runtime-21.43.21438/opencl/test/unit_test/os_interface/linux/local_memory_helper_tests_dg1.cpp:90:1: error: no return statement in function returning non-void [-Werror=return-type]
   90 | }
      | ^
/build/intel-compute-runtime/src/compute-runtime-21.43.21438/opencl/test/unit_test/os_interface/linux/local_memory_helper_tests_dg1.cpp: At global scope:
/build/intel-compute-runtime/src/compute-runtime-21.43.21438/opencl/test/unit_test/os_interface/linux/local_memory_helper_tests_dg1.cpp:92:38: error: ‘givenDg1AndMemoryRegionQuerySupportedWhenQueryingMemoryInfoThenMemoryInfoIsCreatedWithRegions’ has not been declared
   92 | DG1TEST_F(LocalMemoryHelperTestsDg1, givenDg1AndMemoryRegionQuerySupportedWhenQueryingMemoryInfoThenMemoryInfoIsCreatedWithRegions) {
      |                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/build/intel-compute-runtime/src/compute-runtime-21.43.21438/opencl/test/unit_test/os_interface/linux/local_memory_helper_tests_dg1.cpp:92:1: error: ISO C++ forbids declaration of ‘DG1TEST_F’ with no type [-fpermissive]
   92 | DG1TEST_F(LocalMemoryHelperTestsDg1, givenDg1AndMemoryRegionQuerySupportedWhenQueryingMemoryInfoThenMemoryInfoIsCreatedWithRegions) {
[... cut ...]
/build/intel-compute-runtime/src/compute-runtime-21.43.21438/third_party/gtest/gtest/gtest.h:16650:32: note: in expansion of macro ‘GTEST_ASSERT_NE’
16650 | # define ASSERT_NE(val1, val2) GTEST_ASSERT_NE(val1, val2)
      |                                ^~~~~~~~~~~~~~~
/build/intel-compute-runtime/src/compute-runtime-21.43.21438/opencl/test/unit_test/os_interface/linux/local_memory_helper_tests_dg1.cpp:104:5: note: in expansion of macro ‘ASSERT_NE’
  104 |     ASSERT_NE(nullptr, memoryInfo);
      |     ^~~~~~~~~
```

JablonskiMateusz, 2021-11-02:
Hi @ArchangeGabriel, could you retry with additional cmake flag `-DTESTS_DG1=1`? AFAIK this issue is already fixed since this change: https://github.com/intel/compute-runtime/commit/32592d953bb4b7c13f2a3c35e5eb9db8acda7982#diff-752a561bea67933290e4db1fafb66db65d6dc8a2294af2716c2fd2d40a0ea5a5R60

ArchangeGabriel, 2021-11-02:
That got it working, thanks!

closed 2021-11-02 by ArchangeGabriel