intel/compute-runtime #472: Avoid using glibc-specific execinfo.h functionality on non-glibc

rcombs, 2021-11-04:
Currently, ICR uses execinfo.h for crash backtraces. This is (as I understand it) nonessential functionality, so as it's implemented in a glibc-specific way, it should be disabled when not building against glibc. For instance, in safety_guard_linux.h:
```
diff --git a/shared/offline_compiler/source/utilities/linux/safety_guard_linux.h b/shared/offline_compiler/source/utilities/linux/safety_guard_linux.h
index f3c02b6d4..d7dfb8a35 100644
--- a/shared/offline_compiler/source/utilities/linux/safety_guard_linux.h
+++ b/shared/offline_compiler/source/utilities/linux/safety_guard_linux.h
@@ -10,7 +10,9 @@
 
 #include <cstdio>
 #include <cstdlib>
+#ifdef __GLIBC__
 #include <execinfo.h>
+#endif
 #include <setjmp.h>
 #include <signal.h>
 
@@ -37,6 +39,7 @@ class SafetyGuardLinux {
     }
 
     static void sigAction(int sigNum, siginfo_t *info, void *ucontext) {
+#ifdef __GLIBC__
         const int callstackDepth = 30;
         void *addresses[callstackDepth];
         char **callstack;
@@ -50,6 +53,9 @@ class SafetyGuardLinux {
         }
 
         free(callstack);
+#else
+        printf("GOT SIGNAL: %i\n", sigNum);
+#endif
         longjmp(jmpbuf, 1);
     }
 
```

jbeich, 2021-11-05:
`<execinfo.h>` is supported via libexecinfo on musl and BSDs. Duplicate of https://github.com/intel/compute-runtime/issues/331 but proposes a different fix.

JablonskiMateusz, 2023-01-13:
closing as a duplicate: #331

closed 2023-01-13 by JablonskiMateusz