intel/compute-runtime #473: Running 2D kernels with Level Zero

jjfumero, 2021-11-09:
_This Issue was moved from https://github.com/oneapi-src/level-zero/issues/71_ 

I have a question about running 2D kernels with level zero and the number of threads as block size for each dimension.

In level zero, I am using these set of calls to setup the number of threads:

```cpp
uint32_t groupSizeX = 256u;
uint32_t groupSizeY = 256u;
uint32_t groupSizeZ = 1u;
zeKernelSuggestGroupSize(kernel, items, items, 1U, &groupSizeX, &groupSizeY, &groupSizeZ);

// Set number of threads after suggestion
zeKernelSetGroupSize(kernel, groupSizeX, groupSizeY, groupSizeZ);

// set kernel args
...

// run kernel
ze_group_count_t dispatch;
dispatch.groupCountX = items / groupSizeX;
dispatch.groupCountY = items / groupSizeY;
dispatch.groupCountZ = 1;
zeCommandListAppendLaunchKernel(cmdList, kernel, &dispatch, kernelTsEvent, 0, nullptr);
```

I noticed that, after the LevelZero suggestion (`zeKernelSuggestGroupSize`), the groupSize for the Y dimension (`groupSizeY` variable) is set to 1. 

On my GPU, I see that I can actually run 256x256:

```bash
stype : DEVICE_COMPUTE_PROPERTIES
pNext : 0x0
maxTotalGroupSize : 256
maxGroupSizeX : 256
maxGroupSizeY : 256
maxGroupSizeZ : 256
maxGroupCountX : 4294967295
maxGroupCountY : 4294967295
maxGroupCountZ : 4294967295
maxSharedLocalMemory : 65536
numSubGroupSizes : 3
subGroupSizes : [ 8, 16, 32, 0, 0, 0, 0, 0 ]
```

If I try to force it, I get a `ZE_RESULT_ERROR_INVALID_GROUP_SIZE_DIMENSION`. Is there any way to run with block size with more than 1 thread? Am I missing something? If needed, I can prepare a test-case. 

Note that, when using OpenCL I can run on the same Intel HD Graphics with a block of 256x256 as a local work-group size. 

Any pointers will be appreciated. 

Hardware & Drivers:
* Driver: 21.38.21026 
* Linux 5.14.11-200.fc34.x86_64 (Fedora) 
* Runnig in Intel i9-10885H HD Graphics 
* Level-Zero: [7afa8a6](https://github.com/oneapi-src/level-zero/commit/7afa8a6dda56d4baef950c0a9d5ef8d8b0e14c4e)

jandres742, 2021-11-09:
that should be possible. If a repro is provided, we could take  look.

JablonskiMateusz, 2021-11-10:
@jjfumero 
As per L0 spec:
> uint32_t maxTotalGroupSize
[out] Maximum items per compute group. (groupSizeX * groupSizeY * groupSizeZ) <= maxTotalGroupSize

It means that your total group size (groupSizeX * groupSizeY * groupSizeZ) can't exceed maxTotalGroupSize.

Based on what you mentioned 
> maxTotalGroupSize : 256
maxGroupSizeX : 256
maxGroupSizeY : 256
maxGroupSizeZ : 256

your max total group size is 256 so you can't set group size (256,256,1) as 256*256 > 256

jjfumero, 2021-11-10:
Thank you @JablonskiMateusz for the clarification. I misunderstood this. That solves my issue actually. 

After double-checking what we do for OpenCL in our codebase, we had an extra check to guarantee the correct number of threads, but it was not printed in our debug mode. So in OpenCL, I was getting the wrong number of threads to be deployed and that confused me with Level Zero. 

Any reason why the LevelZero driver does not set any other thread-block combination? e.g., 128x2, 16x16, etc?

JablonskiMateusz, 2021-11-30:
AFAIK it depends on info from compiler. 
E.g. in OpenCL C you can specify attribute to force work group size:
> The work-group size to be used for kernel can also be specified in the program source using the `__attribute__((reqd_work_group_size(X, Y, Z)))` qualifier. In this case the size of work group specified by local_work_size must match the value specified by the reqd_work_group_size `__attribute__` qualifier.

https://www.khronos.org/registry/OpenCL/specs/3.0-unified/html/OpenCL_Ext.html 

If there is no such info, then I think it is linear due to some performance reasons.

jjfumero, 2021-12-07:
Ok, thanks. In our case, if we tune the thread-block size, we increase performance up to 2x (kernel time). So I was thinking that `zeKernelSuggestGroupSize` could also receive a policy for different thread-block selectors and get different options. 

BTW, my question was answered. I think this issue can be closed. 

Thank you!

closed 2021-12-07 by JablonskiMateusz