intel/compute-runtime #475: Mismatch between OpenCL Library and OpenCL platform

awsomecod, 2021-11-15:
My programs crash, mostly because of the note I see at the end of ``clinfo`` command: 

```
your OpenCL library only supports OpenCL 2.1,
		but some installed platforms support OpenCL 3.0.
		Programs using 3.0 features may crash
```

How can I resolve the issue? Below are the very first lines from output of ``clinfo`` command: 

```
Number of platforms                               1
  Platform Name                                   Intel(R) OpenCL HD Graphics
  Platform Vendor                                 Intel(R) Corporation
  Platform Version                                OpenCL 3.0 
  Platform Profile                                FULL_PROFILE
  .
  .
  .
  Platform Name                                   Intel(R) OpenCL HD Graphics
  Number of devices                               1
  Device Name                                     Intel(R) UHD Graphics 750 [0x4c8a]
  Device Vendor                                   Intel(R) Corporation
  Device Vendor ID                                0x8086
  Device Version                                  OpenCL 3.0 NEO 
  Driver Version                                  21.44.21506
  Device OpenCL C Version                         OpenCL C 1.2 
  Device Type                                     GPU
  Device Profile                                  FULL_PROFILE

```

My OS is Ubuntu 20.04, the kernel is ``5.15.2-051502-generic``. I installed Intel compute engine version [21.44.21506](https://github.com/intel/compute-runtime/releases/tag/21.44.21506)

The content of ``/etc/OpenCL/vendors/intel.icd`` file is:

``/usr/lib/x86_64-linux-gnu/intel-opencl/libigdrcl.so``

alalek, 2021-11-15:
Do you really have that problem? Are your programs use OpenCL 3.0 features?

> Programs using 3.0 features may crash

... because installed libOpenCL.so loader is outdated.
The library is not a part of this repository, so this is not the right place for resolving that problem.
OpenCL loader is usually installed from Ubuntu packages.

> Ubuntu 20.04

OpenCL 3.0 loader is shipped with Ubuntu 21.04+:

[Changelog notes](http://changelogs.ubuntu.com/changelogs/pool/universe/o/ocl-icd/ocl-icd_2.2.14-2/changelog):

```
ocl-icd (2.2.13-1) unstable; urgency=medium

  * Team upload.
  * watch: Updated.
  * New upstream release.
    - add support for OpenCL 3.0 (Closes: #960450)
  * patches: gcc-10.patch dropped (upstream), typos.patch refreshed.
  * symbols: Updated.

 -- Timo Aaltonen <tjaalton@debian.org>  Sat, 14 Nov 2020 10:16:16 +0200
```

Package versions:

https://packages.ubuntu.com/search?keywords=ocl-icd-libopencl1&searchon=names

---

Also you can build and install the loader from source code, but you need to know that are you doing (compatibility issues may occur) - code is available on GitHub (not in this repo).

awsomecod, 2021-11-15:
I am try to run a demo program from open model zoo of openvino. My guess is that openvino uses OpenCl 3.0 features. I am note sure if I can build OpenCL loader, so will try using Ubuntu 21.04.

alalek, 2021-11-15:
This is strange.
Ubuntu 20.04 is validation platform for OpenVINO.

I believe it makes sense to ask for help in OpenVINO repositories (with providing error messages and used hardware configuration):
- OMZ: https://github.com/openvinotoolkit/open_model_zoo/issues
- OpenVINO runtime: https://github.com/openvinotoolkit/openvino/issues

JablonskiMateusz, 2021-12-07:
@awsomecod is it still a valid issue?

JablonskiMateusz, 2022-01-18:
closing as inactive

closed 2022-01-18 by JablonskiMateusz