intel/compute-runtime #477: No 'levelzero:gpu' in SYCL devices list

fwinter, 2021-11-16:
I hope this is the right place for this issue.

The system is Ubuntu 20.04 with Intel(R) Xeon(R) W-10885M CPU. I installed the Intel GPU driver following:
https://dgpu-docs.intel.com/installation-guides/ubuntu/ubuntu-focal.html (which went fine). I also installed the oneAPI DPC++ compiler.
However, it seems I can't get the LevelZero GPU to work.

$ sycl-ls 
[opencl:0] ACC : Intel(R) FPGA Emulation Platform for OpenCL(TM) 1.2 [2021.12.9.0.24_005321]
[opencl:0] CPU : Intel(R) OpenCL 3.0 [2021.12.9.0.24_005321]
[host:0] HOST: SYCL host platform 1.2 [1.2]

It's missing the 'levelzero:gpu' entry.

$ id
uid=1000(fwinter) gid=1000(fwinter) groups=1000(fwinter),4(adm),24(cdrom),27(sudo),30(dip),44(video),46(plugdev),109(render),120(lpadmin),132(lxd),133(sambashare)

$ dpkg -l |grep icd
ii  intel-opencl-icd                                                        21.37.20939+i620~u20.04               amd64        Intel graphics compute runtime for OpenCL
ii  ocl-icd-libopencl1:amd64                                                2.2.11-1ubuntu1                       amd64        Generic OpenCL ICD Loader

Is the presence of the generic ICD loader okay?

What else could be missing?

jandres742, 2021-11-16:
@fwinter  most likely you have missing some Level-zero packages. Please take a look at our Release pages. You see you need to have installed the intel-level-zero-gpu library (this is the GPU driver) and you need to have the level-zero library (this is the loader, similar to OpenCL's ICD).

You can also use `strace` and SYCL_PI_TRACE=2 to see more info which might be useful to understand the problem, something like:

```sh
SYCL_PI_TRACE=2 strace -o output.log sycl-ls
```

fwinter, 2021-11-16:
@jandres742 The packages are installed, I believe in their latest version. I attach the output of the command `dpkg -l|grep intel`. This should cover the relevant packages and shows their version.

[dpkg.txt](https://github.com/intel/compute-runtime/files/7548295/dpkg.txt)

The output of the strace command is here.

[output.log](https://github.com/intel/compute-runtime/files/7548299/output.log)

For the unwary eye it is hard to see what's going wrong.

JablonskiMateusz, 2021-11-17:
GPU device discovery fails at very beginning.
For OpenCL: 
> openat(AT_FDCWD, "/dev/dri/by-path/pci-0000:00:02.0-render", O_RDWR) = 10
ioctl(10, DRM_IOCTL_VERSION, 0x7ffde9a160e0) = 0
openat(AT_FDCWD, "/dev/dri/by-path/pci-0000:01:00.0-render", O_RDWR) = 11
ioctl(11, DRM_IOCTL_VERSION, 0x7ffde9a160e0) = 0
close(11)                               = 0
ioctl(10, DRM_IOCTL_I915_GETPARAM, 0x7ffde9a16200) = 0
ioctl(10, DRM_IOCTL_I915_GETPARAM, 0x7ffde9a16200) = 0
close(10)                               = 0

For Level Zero:
>  openat(AT_FDCWD, "/dev/dri/by-path/pci-0000:00:02.0-render", O_RDWR) = 3
ioctl(3, DRM_IOCTL_VERSION, 0x7ffde9a16170) = 0
openat(AT_FDCWD, "/dev/dri/by-path/pci-0000:01:00.0-render", O_RDWR) = 4
ioctl(4, DRM_IOCTL_VERSION, 0x7ffde9a16170) = 0
close(4)                                = 0
ioctl(3, DRM_IOCTL_I915_GETPARAM, 0x7ffde9a16290) = 0
ioctl(3, DRM_IOCTL_I915_GETPARAM, 0x7ffde9a16290) = 0
close(3)                                = 0

It looks like there is an issue with i915.
@fwinter what linux kernel is installed on your system?

fwinter, 2021-11-17:
Hi @JablonskiMateusz 

$ uname -a
Linux fwinter-lpt 5.11.0-40-generic #44~20.04.2-Ubuntu SMP Tue Oct 26 18:07:44 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

$ sudo lshw -c video
  *-display                 
       description: VGA compatible controller
       product: TU104GLM [Quadro RTX 5000 Mobile / Max-Q]
       vendor: NVIDIA Corporation
       physical id: 0
       bus info: pci@0000:01:00.0
       version: a1
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress vga_controller bus_master cap_list rom
       configuration: driver=nvidia latency=0
       resources: irq:187 memory:b3000000-b3ffffff memory:70000000-7fffffff memory:80000000-81ffffff ioport:3000(size=128) memory:b4000000-b407ffff
  *-display
       description: VGA compatible controller
       product: Intel Corporation
       vendor: Intel Corporation
       physical id: 2
       bus info: pci@0000:00:02.0
       logical name: /dev/fb0
       version: 05
       width: 64 bits
       clock: 33MHz
       capabilities: pciexpress msi pm vga_controller bus_master cap_list rom fb
       configuration: depth=32 driver=i915 latency=0 mode=1920x1080 visual=truecolor xres=1920 yres=1080
       resources: iomemory:600-5ff iomemory:400-3ff irq:186 memory:604a000000-604affffff memory:4000000000-400fffffff ioport:4000(size=64) memory:c0000-dffff

$ modinfo i915

[modinfo_i915.txt](https://github.com/intel/compute-runtime/files/7555371/modinfo_i915.txt)

Do you think the NVIDIA driver interferes?

eero-t, 2021-11-17:
What these show?
`ls -l /dev/dri/`
`head /sys/class/drm/{card[0-9],renderD*}/device/uevent`

fwinter, 2021-11-17:
$ ls -l /dev/dri/
total 0
drwxr-xr-x  2 root root        120 Nov 17 09:25 by-path
crw-rw----+ 1 root video  226,   0 Nov 17 09:25 card0
crw-rw----+ 1 root video  226,   1 Nov 17 09:25 card1
crw-rw----+ 1 root render 226, 128 Nov 17 09:25 renderD128
crw-rw----+ 1 root render 226, 129 Nov 17 09:25 renderD129

$ head /sys/class/drm/{card[0-9],renderD*}/device/uevent
==> /sys/class/drm/card0/device/uevent <==
DRIVER=i915
PCI_CLASS=30000
PCI_ID=8086:9BF6
PCI_SUBSYS_ID=1028:09C3
PCI_SLOT_NAME=0000:00:02.0
MODALIAS=pci:v00008086d00009BF6sv00001028sd000009C3bc03sc00i00

==> /sys/class/drm/card1/device/uevent <==
DRIVER=nvidia
PCI_CLASS=30000
PCI_ID=10DE:1EB5
PCI_SUBSYS_ID=1028:09C3
PCI_SLOT_NAME=0000:01:00.0
MODALIAS=pci:v000010DEd00001EB5sv00001028sd000009C3bc03sc00i00

==> /sys/class/drm/renderD128/device/uevent <==
DRIVER=i915
PCI_CLASS=30000
PCI_ID=8086:9BF6
PCI_SUBSYS_ID=1028:09C3
PCI_SLOT_NAME=0000:00:02.0
MODALIAS=pci:v00008086d00009BF6sv00001028sd000009C3bc03sc00i00

==> /sys/class/drm/renderD129/device/uevent <==
DRIVER=nvidia
PCI_CLASS=30000
PCI_ID=10DE:1EB5
PCI_SUBSYS_ID=1028:09C3
PCI_SLOT_NAME=0000:01:00.0
MODALIAS=pci:v000010DEd00001EB5sv00001028sd000009C3bc03sc00i00

JablonskiMateusz, 2021-11-17:
> PCI_ID=8086:9BF6

This device id was added in https://github.com/intel/compute-runtime/commit/4b2f301e52ffc522ea09c0123ebf215ef0e6799d I hope it will fix this issue.

eero-t, 2021-11-17:
> This device id was added in 4b2f301 I hope it will fix this issue.

That commit went just in, so I guess it's not in any release / package yet, but should be in next release (and packages updated after that)?

fwinter, 2021-11-17:
Ok, great. I'll make sure to install the latest package when it's released. From what I understand this should happen within a week, right?

I have these dpkg repos in the apt list:

>/etc/apt/sources.list:deb [arch=amd64] https://repositories.intel.com/graphics/ubuntu focal main
/etc/apt/sources.list:deb https://apt.repos.intel.com/oneapi all main

Will the new compute-runtime be in there or will I have to download the deb packages manually?

eero-t, 2021-11-19:
"21.46.21636" release was tagged already on 12th:
```
$ git show --format=fuller 21.46.21636 | grep ^CommitDate
CommitDate: Fri Nov 12 17:23:35 2021 +0100
```
Whereas the fix commit went in on 17th, so it cannot be in that release yet.

(There's a period of testing after tagging, before the tagged version is released.)

fwinter, 2021-11-29:
Can confirm, the latest release fixes this issue. Thanks!

closed 2021-11-29 by fwinter