intel/compute-runtime #478: clinfo only finds GPU when running as root

awsomecod, 2021-11-16:
I have a UHD 750 GPU. I installed [21.42.21270](https://github.com/intel/compute-runtime/releases/tag/21.42.21270).  The command `sudo clinfo` finds the GPU but `clinfo` results in:

``Number of platforms                               0``

How can I make `clinfo` finds GPU without using root? I have to fix this issue, cause currently my openvino program does not find GPU and does not run.

awsomecod, 2021-11-16:
I resolved the issue. I don't know which commands finally resolved the problem, but I guess the following ones: 

```
sudo usermod -a -G video $LOGNAME
sudo usermod -a -G render $LOGNAME

```

JacekDanecki, 2021-11-16:
See here: https://github.com/intel/compute-runtime#installation-options

closed 2021-11-16 by JacekDanecki

anton-vlasenko, 2023-02-22:
> sudo usermod -a -G render $LOGNAME

This fixed the issue for me (Debian).