intel/compute-runtime #479: Has the latest OpenCL release (21.46.21636) supported Adler Lake? Thanks!

heyu1, 2021-11-24:


JablonskiMateusz, 2021-12-03:
Hi @heyu1. Could you specify which Alder Lake platform?

heyu1, 2021-12-06:
ADL-S and ADL-P

JablonskiMateusz, 2021-12-07:
I hope they should be supported. You can check list of supported device ids here:
https://github.com/intel/compute-runtime/blob/21.46.21636/shared/source/dll/devices/devices_additional.inl#L10 and here https://github.com/intel/compute-runtime/blob/21.46.21636/shared/source/dll/devices/devices_base.inl#L35

closed 2021-12-20 by JablonskiMateusz

ngeorgis, 2021-12-26:
Thank you, but could you clarify please on ADL-P GT2 ? 

Device ID 0x46A0 is in https://github.com/intel/compute-runtime/blob/8c930049375806136d758fd87cfdbb9cac1ec7e3/opencl/test/unit_test/gen12lp/adlp/linux/dll/device_id_tests_adlp.cpp but not in https://github.com/intel/compute-runtime/blob/21.46.21636/shared/source/dll/devices/devices_base.inl#L35
If ADL-P GT2 supported or not yet? 

Or may I should rebuild with SUPPORT_ADLP ?