intel/compute-runtime #480: clarification about device UUID

bgoglin, 2021-11-30:
Hello
I am looking at using the UUID to uniquely identify devices in the L0 API. The UUID in ze_device_properties_t seems to be the same across multiple runs, and the code in ./level_zero/core/source/device/device_imp.cpp seems to confirm that, good.
However there's also some UUID code  ./level_zero/core/source/driver/driver_handle_imp.cpp that makes the UUID value depend on a timestamp, so it wouldn't be reproducible across runs. And I seem to remember having seen the UUID vary across runs in the past, but I don't remember when/how. Can you clarify the difference between these two uuid codes? Can you specify in the doc whether the UUID (at least in ze_device_properties_t) is guaranteed to remain the same between runs as long as we don't modify the hardware, etc?

Also, the documentation says that "subdevices will have their own UUID" but I currently see the same UUID for the root device and both subdevices.
Thanks

bgoglin, 2021-11-30:
> this is not defined in the specification, correct. So right now, it is up to the implementation to determine whether the UUID stays the same between runs. We will be updating the specification accordingly.

I am confused by you're "right now" and "will be updating". Are you saying you'll change the spec to say theat the UUID stays the same between runs? Or are you going to say "it's implementation-dependent, it may change" ?

jandres742, 2021-11-30:
@bgoglin 

> Are you saying you'll change the spec to say theat the UUID stays the same between runs? Or are you going to say "it's implementation-dependent, it may change" ?

that's what we are discussing, whether to say the former or the latter or something else.

jandres742, 2022-01-21:
7fc9b2c3dc54fbc7299548b8cba7b452792eb77f adds support for UUID based on BDF address, assigning a different value for root and subdevices. This should fulfill this issue.

bgoglin, 2022-01-23:
Yes, thanks.

closed 2022-01-23 by bgoglin