intel/compute-runtime #481: set ze_device_memory_properties_t.flags to 0 in zeDeviceGetMemoryProperties()

bgoglin, 2021-11-30:
Hello
The "flags" field of ze_device_memory_properties_t seems to be completely ignored in zeDeviceGetMemoryProperties(). It's not even set to 0 as documented.
I understand you may have nothing to put in there for now, but it may be annoying in the future: Once you start filling that field in the future, people will look at it in their code and consider it valid. However, if they ever link that modern code against an old oneAPI implementation that didn't set the field yet, they will see garbage flags and interpret them as if they were valid. You should make it 0 explicitly.

jandres742, 2021-11-30:
@bgoglin that's a good point. We will add it.

bgoglin, 2021-12-03:
By the way, would it make sense to add a flag to indicate whether that memory is local (HBM/GDDR on discrete GPU) or not (host memory shared with integrated GPU)? Sysman reports no memory module on my laptop while this API reports 80% of my RAM. That's not "real" GPU memory from the hwloc point of view.

jandres742, 2021-12-03:
@bgoglin  we already have name in structze_device_memory_properties_t. or you want something like "bool isLocalMemory"?

>  Sysman reports no memory module on my laptop while this API reports 80% of my RAM. That's not "real" GPU memory from the hwloc point of view.

didnt understand this part. Could you elaborate?

bgoglin, 2021-12-03:
ze_device_memory_properties_t.name is "DDR" both on my laptop (integrated HD Graphics 620) and on Iris Xe Max (discrete model 4905 on devcloud). Shouldn't they be different if Xe Max has its own local memory while my laptop doesn't?

On my laptop, zesDeviceEnumMemoryModules() returns nothing, which is normal because my integrated GPU  uses the main. However zeDeviceGetMemoryProperties() returns one memory, DDR 12.9GB which is likely my host CPU DDR, but limited to 80% of its capacity. I'd like to way to know when zeDeviceGetMemoryProperties() returns real (discrete?) GPU memory, or host shared memory.

bool isLocalMemory or a flag in the currently unused flags would work fine.

jandres742, 2021-12-04:
> However zeDeviceGetMemoryProperties() returns one memory, DDR 12.9GB which is likely my host CPU DDR, but limited to 80% of its capacity. 

we return 80% of memory since we need to save some space for internal allocations, but let's confirm we are returning the correct values.

what do you get for:

maxMemAllocSize from zeDeviceGetProperties.

totalSize from zeDeviceGetMemoryProperties.

bgoglin, 2021-12-04:
laptop:
```
zeDeviceGetProperties.maxMemAllocSize 4294959104
zeDeviceGetMemoryProperties.totalSize 12900782080 name DDR
zesDeviceEnumMemoryModules found 0 memory modules
```
Xe 4905:
```
zeDeviceGetProperties.maxMemAllocSize 4294959104
zeDeviceGetMemoryProperties.totalSize 8088715264 name DDR
zesDeviceEnumMemoryModules found 1 memory modules
zesMemoryGetProperties.physicalSize 0 type 1 /* ZES_MEM_TYPE_DDR */
```

jandres742, 2021-12-05:
thanks @bgoglin 
So going back to your question:

> On my laptop, zesDeviceEnumMemoryModules() returns nothing, which is normal because my integrated GPU uses the main. However zeDeviceGetMemoryProperties() returns one memory, DDR 12.9GB which is likely my host CPU DDR, but limited to 80% of its capacity.

that is correct. We dont report full memory since we use some memory for internal allocations, which is about 20% of the memory, hence the 80% reported. 

So zeDeviceGetMemoryProperties is not expected to return total physical memory, it returns only the total memory that the application can use. For the actual total, SYSMAN will report it.

Now, the other limit we have is that we report max allocatable memory up to 4GB because our GPUs only accept up to that amount when trying to allocate memory while providing stateful optimization. See here, https://dl.acm.org/doi/pdf/10.5555/3314872.3314902, "Addressing Mode Selection" section.

Does this clarify it?

bgoglin, 2021-12-08:
Thanks, I understand now. To come back to the original issue, it'd be nice to have a way to know whether memory is local or not when the core API seems to return similar things for discrete vs integrated GPUs as in https://github.com/intel/compute-runtime/issues/481#issuecomment-986008565

jandres742, 2021-12-08:
@bgoglin  we already return ZE_DEVICE_PROPERTY_FLAG_INTEGRATED in device properties. could that be used to infer that?

bgoglin, 2021-12-08:
Actually, I was thinking of using a workaround based on this flag + zeDeviceGetMemoryProperties.name="DDR". This strategy might fail if we ever see an integrated GPU with its own GDDR (except if zeDeviceGetMemoryProperties.name is different from "DDR") but I could assume this is not going to happen soon.

jandres742, 2021-12-09:
> The "flags" field of ze_device_memory_properties_t seems to be completely ignored in zeDeviceGetMemoryProperties(). It's not even set to 0 as documented.
I understand you may have nothing to put in there for now, but it may be annoying in the future: Once you start filling that field in the future, people will look at it in their code and consider it valid. However, if they ever link that modern code against an old oneAPI implementation that didn't set the field yet, they will see garbage flags and interpret them as if they were valid. You should make it 0 explicitly.

Fixed in ac2b1d403b89d56510677029153e0accff4960d5 

> Actually, I was thinking of using a workaround based on this flag + zeDeviceGetMemoryProperties.name="DDR". This strategy might fail if we ever see an integrated GPU with its own GDDR (except if [...]

Right, we might need that in the future if such HW becomes available.

AdamCetnerowski, 2022-01-13:
Closing as this is resolved.

closed 2022-01-13 by AdamCetnerowski