intel/compute-runtime #484: Create the documentation for environment variables

capatober, 2021-12-17:
Hi,
It was found that if I want to find some definition for environment variables from [the list](https://github.com/intel/compute-runtime/blob/master/shared/source/debug_settings/debug_variables_base.inl), I need to go over the list and try to not break eyes looking at the list and try to guess what is the real definition is.

Could you please create any human-friendly documentation (.md, github pages or something else) that can contain clear definitions for variables?
Example:
instead of `DECLARE_DEBUG_VARIABLE(int32_t, RenderCompressedBuffersEnabled, -1, "-1: default, 0: disabled, 1: enabled")`
let's have 
| RenderCompressedBuffersEnabled | what it does, how the behavior is changed | -1 (default value) | possible values with meaning |

jandres742, 2022-01-21:
@capatober : Thanks. We are currently thinking on how to add some more documentation, which will include further explanation on the most important debug keys. We will provide an update here once we have something well defined.

AdamCetnerowski, 2022-11-24:
We are slowly updating the descriptions to be clearer. We do not intend to publish standalone documentation for debug keys.