intel/compute-runtime #486: Build fail due to ocloc "Option 'XXX' registered more than once" errors

eero-t, 2022-01-05:
When trying to build latest compute-runtime "21.52.22081" release, after building:
* GMMlib: intel-gmmlib-22.0.1
* IGC: igc-1.0.9933, with:
  - SPIRV-Headers: sdk-1.2.198.0
  - SPIRV-LLVM-Translator: llvm_release_110
  - SPIRV-Tools: v2021.4
  - vc-intrinsics: HEAD
* Level-Zero: v1.7.4

I get following errors:
```
...
[338/2454] Building CXX object shared/source/CMakeFiles/neo_shared.dir/program/sync_buffer_handler.cpp.o
[339/2454] Generating ../../../bin/Gen12LPlp/0/test_files/x64/CopyBuffer_simd16_Gen12LPlp.spv-cl-kernel-arg-info, ../../../bin/Gen12LPlp/0/test_files/x64/CopyBuffer_simd16_Gen12LPlp.bin-cl-kernel-arg-info, ../../../bin/Gen12LPlp/0/test_files/x64/CopyBuffer_simd16_Gen12LPlp.gen-cl-kernel-arg-info
FAILED: bin/Gen12LPlp/0/test_files/x64/CopyBuffer_simd16_Gen12LPlp.spv-cl-kernel-arg-info bin/Gen12LPlp/0/test_files/x64/CopyBuffer_simd16_Gen12LPlp.bin-cl-kernel-arg-info bin/Gen12LPlp/0/test_files/x64/CopyBuffer_simd16_Gen12LPlp.gen-cl-kernel-arg-info 
cd /home/nobody/source/compute-runtime/opencl/test/unit_test/test_files && LD_LIBRARY_PATH=/home/nobody/source/compute-runtime/build/bin /home/nobody/source/compute-runtime/build/bin/ocloc -file /home/nobody/source/compute-runtime/opencl/test/unit_test/test_files/CopyBuffer_simd16.cl -device tgllp -64 -out_dir /home/nobody/source/compute-runtime/build/bin/Gen12LPlp/0/test_files/x64/ -revision_id 0 -options -cl-kernel-arg-info -options_name
: CommandLine Error: Option 'spirv-expand-step' registered more than once!
LLVM ERROR: inconsistency in registered CommandLine options
Aborted (core dumped)
[340/2454] Building CXX object shared/source/CMakeFiles/neo_shared.dir/os_interface/windows/os_environment_win.cpp.o
[341/2454] Generating ../../../bin/Gen12LPlp/0/test_files/x64/CopyBuffer_simd16_Gen12LPlp.spv-cl-finite-math-only, ../../../bin/Gen12LPlp/0/test_files/x64/CopyBuffer_simd16_Gen12LPlp.bin-cl-finite-math-only, ../../../bin/Gen12LPlp/0/test_files/x64/CopyBuffer_simd16_Gen12LPlp.gen-cl-finite-math-only
FAILED: bin/Gen12LPlp/0/test_files/x64/CopyBuffer_simd16_Gen12LPlp.spv-cl-finite-math-only bin/Gen12LPlp/0/test_files/x64/CopyBuffer_simd16_Gen12LPlp.bin-cl-finite-math-only bin/Gen12LPlp/0/test_files/x64/CopyBuffer_simd16_Gen12LPlp.gen-cl-finite-math-only 
cd /home/nobody/source/compute-runtime/opencl/test/unit_test/test_files && LD_LIBRARY_PATH=/home/nobody/source/compute-runtime/build/bin /home/nobody/source/compute-runtime/build/bin/ocloc -file /home/nobody/source/compute-runtime/opencl/test/unit_test/test_files/CopyBuffer_simd16.cl -device tgllp -64 -out_dir /home/nobody/source/compute-runtime/build/bin/Gen12LPlp/0/test_files/x64/ -revision_id 0 -options -cl-finite-math-only -options_name
: CommandLine Error: Option 'spirv-expand-step' registered more than once!
LLVM ERROR: inconsistency in registered CommandLine options
Aborted (core dumped)
[342/2454] Generating ../../../bin/Gen12LPlp/0/test_files/x64/CopyBuffer_simd16_Gen12LPlp.spv-cl-fast-relaxed-math, ../../../bin/Gen12LPlp/0/test_files/x64/CopyBuffer_simd16_Gen12LPlp.bin-cl-fast-relaxed-math, ../../../bin/Gen12LPlp/0/test_files/x64/CopyBuffer_simd16_Gen12LPlp.gen-cl-fast-relaxed-math
FAILED: bin/Gen12LPlp/0/test_files/x64/CopyBuffer_simd16_Gen12LPlp.spv-cl-fast-relaxed-math bin/Gen12LPlp/0/test_files/x64/CopyBuffer_simd16_Gen12LPlp.bin-cl-fast-relaxed-math bin/Gen12LPlp/0/test_files/x64/CopyBuffer_simd16_Gen12LPlp.gen-cl-fast-relaxed-math 
cd /home/nobody/source/compute-runtime/opencl/test/unit_test/test_files && LD_LIBRARY_PATH=/home/nobody/source/compute-runtime/build/bin /home/nobody/source/compute-runtime/build/bin/ocloc -file /home/nobody/source/compute-runtime/opencl/test/unit_test/test_files/CopyBuffer_simd16.cl -device tgllp -64 -out_dir /home/nobody/source/compute-runtime/build/bin/Gen12LPlp/0/test_files/x64/ -revision_id 0 -options -cl-fast-relaxed-math -options_name
: CommandLine Error: Option 'spirv-expand-step' registered more than once!
LLVM ERROR: inconsistency in registered CommandLine options
Aborted (core dumped)
```

Googling shows this to happen when components have been built against different LLVM versions.  However, above is built in a Ubuntu 21.04 container which has only (Ubuntu) LLVM v11 installed.

eero-t, 2022-01-05:
CMake options and resulting config it shows:
```
cmake -LH -Wno-dev -G Ninja -DCMAKE_INSTALL_PREFIX=${INSTALL_DIR} -DCMAKE_BUILD_TYPE=Release -DSUPPORT_GEN8=0 -DSUPPORT_GEN9=1 -DSUPPORT_GEN11=1 -DSUPPORT_GEN12LP=1 -DSUPPORT_DG1=1 -DSUPPORT_XE_HP_SDV=0 -DSUPPORT_XE_HP_CORE=0  -DNEO_DISABLE_LD_GOLD=1       -DDO_NOT_RUN_AUB_TESTS=1 -DDONT_CARE_OF_VIRTUALS=1 ../
[0m-- The C compiler identification is GNU 10.3.0
-- The CXX compiler identification is GNU 10.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- branch dir list: /
-- WDK include paths: /home/nobody/source/compute-runtime/third_party/DirectX-Headers/include/wsl;/home/nobody/source/compute-runtime/third_party/DirectX-Headers/include/wsl/stubs;/home/nobody/source/compute-runtime/third_party/DirectX-Headers/include/directx;/home/nobody/source/compute-runtime/third_party/DirectX-Headers/include/dxguids;/home/nobody/source/compute-runtime/third_party/libdxg/include/
-- Driver model : drm_or_wddm
-- Host processor: x86_64
-- Target processor: x86_64
-- Neo target processor: x86_64
-- Performing Test COMPILER_SUPPORTS_LLD
-- Performing Test COMPILER_SUPPORTS_LLD - Success
-- Performing Test COMPILER_SUPPORTS_GOLD
-- Performing Test COMPILER_SUPPORTS_GOLD - Success
-- Using linker: default
-- Release build configuration
-- Computed OpenCL version major is: 21
-- Computed OpenCL version minor is: 52
-- GTest repeat count set to 1
-- GTest shuffle set to --gtest_shuffle;--gtest_random_seed=0
-- Source Level Debugger headers dir: /home/nobody/source/compute-runtime/third_party/source_level_debugger
-- Aub Stream Headers dir: /home/nobody/source/compute-runtime/third_party/aub_stream/headers
-- Metrics Library dir: /home/nobody/source/compute-runtime/third_party/metrics_library
-- Metrics Discovery dir: /home/nobody/source/compute-runtime/third_party/metrics_discovery
-- Khronos OpenCL headers dir: /home/nobody/source/compute-runtime/third_party/opencl_headers
-- Khronos OpenGL headers dir: /home/nobody/source/compute-runtime/third_party/opengl_headers
-- Third party dir: /home/nobody/source/third_party
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'igc-opencl'
--   Found igc-opencl, version 1.0.1
-- IGC include dirs: /usr/local/include/igc;/usr/local/include/igc/cif;/usr/local/include/igc/ocl_igc_shared/executable_format;/usr/local/include/igc/ocl_igc_shared/device_enqueue
-- VISA Dir: /usr/local/include/visa
-- IGA Includes dir: /usr/local/include/iga
-- Checking for module 'igdgmm'
--   Found igdgmm, version 12.0.0
-- GmmLib include dirs: /usr/local/include/igdgmm;/usr/local/include/igdgmm/GmmLib;/usr/local/include/igdgmm/GmmLib/inc;/usr/local/include/igdgmm/inc;/usr/local/include/igdgmm/inc/common;/usr/local/include/igdgmm/util
-- Checking for module 'libva>=1.0.0'
--   No package 'libva' found
-- AUB_STREAM_DIR = 
-- Engine node dir: /home/nobody/source/compute-runtime/third_party/aub_stream/headers
-- Auto-Enabling GEN12LP tests for DG1
-- All supported platforms:  TGLLP DG1 RKL ADLS ADLP ICLLP LKF EHL SKL KBL GLK CFL BXT
-- All tested platforms:  TGLLP DG1 RKL ADLS ADLP ICLLP LKF EHL SKL KBL GLK CFL BXT
-- Default supported platform: SKL
-- Default tested platform: SKL
-- All supported core families: GEN9;GEN11;GEN12LP
-- All tested core families: GEN9;GEN11;GEN12LP
-- Default tested family name: SKLFamily
-- i915 includes dir: /home/nobody/source/compute-runtime/third_party/uapi

-- Performing Test COMPILER_SUPPORTS_INDIRECT_BRANCH_THUNK
-- Performing Test COMPILER_SUPPORTS_INDIRECT_BRANCH_THUNK - Failed
[91mCMake Warning at CMakeLists.txt:595 (message):
  Spectre mitigation -mindirect-branch=thunk flag is not supported by the
  compiler

[0m-- Performing Test COMPILER_SUPPORTS_FUNCTION_RETURN_THUNK
-- Performing Test COMPILER_SUPPORTS_FUNCTION_RETURN_THUNK - Failed
[91mCMake Warning at CMakeLists.txt:601 (message):
[... 213 lines cut ...]
TESTS_TGLLP:BOOL=TRUE

// Build ULTs for XE_HP_CORE devices
TESTS_XE_HP_CORE:BOOL=0

// Link with address sanitization support
USE_ASAN:BOOL=OFF

// Use OpenCL program binary cache
USE_CL_CACHE:BOOL=ON

// Path to a program.
__LD_GOLD_FOUND:FILEPATH=/usr/bin/ld.gold

// Path to a program.
__LD_LLD_FOUND:FILEPATH=__LD_LLD_FOUND-NOTFOUND

// The directory containing a CMake configuration file for igsc.
igsc_DIR:PATH=igsc_DIR-NOTFOUND

// Build unit tests.
level-zero-gpu_BUILD_TESTS:BOOL=ON
```

(XE_HP_* stuff is disabled as its unit tests have been failing in my Xeon build machine.)

eero-t, 2022-01-07:
Exactly same container build works fine if I downgrade IGC & compute-runtime to previous release tags:
* IGC: igc-1.0.9636
* compute-runtime: 21.51.21973

So this is regression in either of those, not issue with updates in other repositories.

eero-t, 2022-01-07:
Note: I'm using Ubuntu opencl-clang, because building that myself would have required pulling & building also LLVM, which would be taking way too much time for building just one of the Intel GPU drivers. Ubuntu "opencl-clang" depends on the same LLVM v11 against which IGC is built, so I do not see why that would be a cause of this issue though: https://packages.ubuntu.com/hirsute/libopencl-clang11

(Especially as previous IGC release worked fine with it.)

JablonskiMateusz, 2022-01-10:
Hi @eero-t ,

> (Especially as previous IGC release worked fine with it.)

Could you try to build 21.52.22081 with the previous IGC release?

eero-t, 2022-01-10:
> Could you try to build 21.52.22081 with the previous IGC release?

@JablonskiMateusz As I said, 21.52.22081 + igc-1.0.*9636* builds fine.

Whereas 21.52.22081 + igc-1.0.*9933* build breaks to ocloc failures (with everything except the release tag from which intel-graphics-compiler project is cloned, being the same).

JablonskiMateusz, 2022-01-10:
If 21.52.22081 works fine with igc-1.0.9636 while fails with igc-1.0.9933, then please transfer the issue to IGC team https://github.com/intel/intel-graphics-compiler

eero-t, 2022-01-10:
I thought it was clear from my above comment:

> Can you move this to IGC, or do I need to file a new bug against it?

That I do not any see functionality in this ticket to transfer it to IGC. So, I'm asking again, can you move this to IGC, or do I need to file a new bug against it?