intel/compute-runtime #487: After update compute-runtime, DPC++ cannot find device

troy818, 2022-01-13:
I'm using oneAPI Explicit SIMD Extension.
I'm using dpcpp_compiler to compiler and run the offical example [vadd_usm.cpp](https://github.com/intel/llvm-test-suite/blob/intel/SYCL/ESIMD/vadd_usm.cpp).

compiling:
`clang++ -fsycl vadd_usm.cpp`
running:
`SYCL_DEVICE_FILTER=opencl:gpu ./a.out`

And I get the following runtime error message:
`Device : Intel(R) Gen9 HD Graphics NEO
LLVM ERROR: SPIRV internal error.`

I use `dpkg -l | grep intel`, and get this:
[image]

I think the runtime is old, so I installed the release [21.38.21026](https://github.com/intel/compute-runtime/releases?page=2) by download *deb package. 
`dpkg -l | grep intel`
[image]
Now the  runtime error message is:
`terminate called after throwing an instance of 'cl::sycl::runtime_error'
  what():  No device of requested type available. Please check https://software.intel.com/content/www/us/en/develop/articles/intel-oneapi-dpcpp-system-requirements.html -1 (CL_DEVICE_NOT_FOUND)
Aborted (core dumped)`

My machine is i7-6700k with Intel® HD Graphics 530. The OS is Ubuntu 18.04. Before I update the compute-runtime, with `clinfo -l` I can see the GPU, but now cannot list available GPU device.

I refered [here](https://dgpu-docs.intel.com/installation-guides/ubuntu/ubuntu-bionic.html) for check permissions to files /dev/dri/renderD*. The permissions to files /dev/dri/renderD* is video and `group 'render' does not exist in /etc/group` shows. And My username is in the video group.

With `strace -s 255 -f -o log.txt clinfo`, I get this txt file [log.txt](https://github.com/intel/compute-runtime/files/7859554/log.txt)

jandres742, 2022-01-13:
@troy818 

> Before I update the compute-runtime, with clinfo -l I can see the GPU, but now cannot list available GPU device.

yes, it seems that something broke. could you do the following:

run with

```sh
./a.out
```
so it uses Level-Zero. This to confirm whether the problem is specific to L0 or OCL.

The change in version was a bit big. From 14441 to 21026. Would it be possible for you to try other releases in between to know when this started to happen?

JablonskiMateusz, 2022-01-14:
In the attached file I see 
> 11183 openat(AT_FDCWD, "/opt/intel/mediasdk/lib64/libigdgmm.so.11", O_RDONLY|O_CLOEXEC) = 5

This file probably conflicts with gmmlib library installed from intel-gmmlib package which installs libigdgmm.so in `/usr/local/lib/`.
There were similar issues reported in the past. Please try to uninstall `/opt/intel/mediasdk/lib64/libigdgmm.so.11` and try again

troy818, 2022-01-19:
@jandres742 @JablonskiMateusz Thanks very much for your help! I just run with *./a.out*, the error is still there. So I take the solution in [After installed compute-runtime, DPC++ cannot find device #450](https://github.com/intel/compute-runtime/issues/450), and it works now!

JablonskiMateusz, 2022-01-25:
Nice to hear that it is working now! I'm closing the issue.

closed 2022-01-25 by JablonskiMateusz