intel/compute-runtime #491: Sysman frequency request value not being reported (wrong file is read)

eero-t, 2022-01-19:
Setup:
* latest (5.16) drm-tip kernel + GuC v69.0.3
* compute-runtime 21.52.22081 [1]

Use-case:
* Query frequency info using Sysman APIs, or using compute-runtime zello_sysman tester:
`# docker run -it --env ZES_ENABLE_SYSMAN=1 --rm --user root --cap-drop ALL --cap-add SYS_ADMIN --cap-add SYS_PTRACE --device /dev/dri:/dev/dri:rw --network none gpu-drivers:latest strace -f -e openat zello_sysman --frequency`

Result:
* While actual frequency reporting works, frequency request value is checked from wrong file and incorrectly reported as `-1` (according to spec, -1 means not known)

TGL:
```
...
Device Name = Intel(R) UHD Graphics [0x9a60]

 ----  Frequency tests ---- 
freqProperties.type = 0
freqProperties.canControl = 
freqProperties.isThrottleEventSupported = 
freqProperties.min = 350
freqProperties.max = 1450
[pid    10] openat(AT_FDCWD, "/sys/class/drm/card0/punit_req_freq_mhz", O_RDONLY) = -1 ENOENT (No such file or directory)
[pid    10] openat(AT_FDCWD, "/sys/class/drm/card0/rapl_PL1_freq_mhz", O_RDONLY) = -1 ENOENT (No such file or directory)
[pid    10] openat(AT_FDCWD, "/sys/class/drm/card0/gt_RP1_freq_mhz", O_RDONLY) = 11
[pid    10] openat(AT_FDCWD, "/sys/class/drm/card0/gt_act_freq_mhz", O_RDONLY) = 11
freqState.currentVoltage = -1
freqState.request = -1   <====================
freqState.tdp = -1
freqState.efficient = 350
freqState.actual = 350
freqState.throttleReasons = 0
...
```

DG1:
```
...
Device Name = Intel(R) Iris(R) Xe MAX Graphics [0x4905]

 ----  Frequency tests ---- 
freqProperties.type = 0
freqProperties.canControl = 
freqProperties.isThrottleEventSupported = 
freqProperties.min = 300
freqProperties.max = 1650
[pid     9] openat(AT_FDCWD, "/sys/class/drm/card0/punit_req_freq_mhz", O_RDONLY) = -1 ENOENT (No such file or directory)
[pid     9] openat(AT_FDCWD, "/sys/class/drm/card0/rapl_PL1_freq_mhz", O_RDONLY) = -1 ENOENT (No such file or directory)
[pid     9] openat(AT_FDCWD, "/sys/class/drm/card0/gt_RP1_freq_mhz", O_RDONLY) = 19
[pid     9] openat(AT_FDCWD, "/sys/class/drm/card0/gt_act_freq_mhz", O_RDONLY) = 19
freqState.currentVoltage = -1
freqState.request = -1   <====================
freqState.tdp = -1
freqState.efficient = 1200
freqState.actual = 1650
freqState.throttleReasons = 0
...
```

As can be seen from above, Sysman is querying the wrong file (`punit_req_freq_mhz`).  Frequency request information is in `gt_cur_freq_mhz` file, as it has always been (at least since GEN7):
```
# head /sys/class/drm/card0/*mhz
==> /sys/class/drm/card0/gt_act_freq_mhz <==
350

==> /sys/class/drm/card0/gt_boost_freq_mhz <==
1450

==> /sys/class/drm/card0/gt_cur_freq_mhz <==
500

==> /sys/class/drm/card0/gt_max_freq_mhz <==
1450

==> /sys/class/drm/card0/gt_min_freq_mhz <==
350

==> /sys/class/drm/card0/gt_RP0_freq_mhz <==
1450

==> /sys/class/drm/card0/gt_RP1_freq_mhz <==
350

==> /sys/class/drm/card0/gt_RPn_freq_mhz <==
350
```
(`gt_act_freq_mhz` file which is correctly used by Sysman for actual frequency, was added only later, in 2015.)

[1] Later compute-runtime releases fail to build.  I do not have a bug on this as I'm waiting for latest IGC release build issue with compute-runtime to be solved first: https://github.com/intel/intel-graphics-compiler/issues/224

saik-intel, 2022-02-02:
thanks for the issue, we are looking internally and update you back

eero-t, 2022-02-21:
Same issue also with the (last week) drm-tip 5.17-rc4 kernel, GuC 69.0.3 and latest compute-runtime "22.05.22297" release.

eero-t, 2022-04-25:
Seems to be fixed with the (last week) drm-tip 5.18-rc3 kernel, GuC 70.1.1 and latest compute-runtime "22.16.22992" release.

closed 2022-04-25 by eero-t