intel/compute-runtime #493: [DG1] Power counter is always reported as zero

eero-t, 2022-01-19:
Setup:
* latest public (5.16) drm-tip kernel with DG1 force-probed + GuC v69.0.3
* compute-runtime 21.52.22081 [1]

Use-case:
* Query power info using Sysman APIs, or using compute-runtime zello_sysman tester:
`# docker run -it --env ZES_ENABLE_SYSMAN=1 --rm --user root --cap-drop ALL --cap-add SYS_ADMIN --cap-add SYS_PTRACE --device /dev/dri:/dev/dri:rw --network none gpu-drivers:latest strace -f -e openat zello_sysman --power`

Result:
* While power reporting works on TGL using same public kernel + GuC version and user-space stack, on DG1 power counter is always reported as zero, only timestamp is updated

TGL:
```
Device Name = Intel(R) UHD Graphics [0x9a60]

 ----  Power tests ---- 
properties.canControl = 
properties.isEnergyThresholdSupported= 
properties.defaultLimit= -1
properties.maxLimit =-1
properties.minLimit =-1
[pid    10] openat(AT_FDCWD, "/sys/class/drm/card0//energy1_input", O_RDONLY) = -1 ENOENT (No such file or directory)
[pid    10] openat(AT_FDCWD, "/sys/class/intel_pmt/telem0/telem", O_RDONLY) = 11
energyCounter.energy = 103741000000
energyCounter.timestamp = 2921663617
```

DG1:
```
Device Name = Intel(R) Iris(R) Xe MAX Graphics [0x4905]

 ----  Power tests ---- 
properties.canControl = 
properties.isEnergyThresholdSupported= 
properties.defaultLimit= -1
properties.maxLimit =-1
properties.minLimit =-1
[pid     9] openat(AT_FDCWD, "/sys/class/drm/card0//energy1_input", O_RDONLY) = -1 ENOENT (No such file or directory)
[pid     9] openat(AT_FDCWD, "/sys/class/intel_pmt/telem0/telem", O_RDONLY) = 19
energyCounter.energy = 0
energyCounter.timestamp = 107908796
```

[1] Later compute-runtime releases fail to build.  I do not have a bug on this as I'm waiting for latest IGC release build issue with compute-runtime to be solved first: https://github.com/intel/intel-graphics-compiler/issues/224

PS. I've tried both drm-tip Git head from yesterday which wants GuC 69.0.3, and week older version which was still at GuC 62.0.

jandres742, 2022-01-22:
thanks for the report. We are looking at it internally and will update back.

eero-t, 2022-02-21:
This has regressed further with (last week) drm-tip 5.17-rc4 kernel, GuC 69.0.3 and latest compute-runtime "22.05.22297" release. 

Now Sysman states that it "Could not retrieve Power domains".

eero-t, 2022-04-25:
Still happens with the (last week) drm-tip 5.18-rc3 kernel, GuC 70.1.1 and latest compute-runtime "22.16.22992" release.

eero-t, 2022-06-14:
Still happens with drm-tip 5.18 kernel, GuC 70.1.1 and latest compute-runtime "22.23.23405" release.

eero-t, 2022-11-25:
Power is reported with "drm-tip" v6.1.0-rc5, GuC v70.5.1 and compute-runtime "22.39.24347" (or newer) =>  Closing.

Power limits are not reported (at least yet).

closed 2022-11-25 by eero-t