intel/compute-runtime #494: Incorrect SPIR kernel launches just fine

fwinter, 2022-01-19:
I hope this is the right place to report. If not please point in the correct direction.

I realized that I can launch SPIR kernels which contain bogus function calls on Intel GPU. Background is that I'm having problems getting the barrier or work_group_barrier function to work properly on my device. I have tested a few possible ways to call that barrier and while doing so realized that I can call whatever I want and the tool chain (LLVM-SPIR, Level Zero, Intel GPU driver) never reports any error back. For instance, if I add a call to an external function that obviously doesn't exist into my SPIR LLVM IR kernel, like so

call void @_Z5bogus() 

declare spir_func void @_Z5bogus()

Then, the SPIR-LLVM translator produces the according SPIR:

12 Name 20 "_Z5bogus"
14 Decorate 20 LinkageAttributes "_Z5bogus" Import 
6 FunctionCall 18 82 20

So far so good as the translator only translates and can't possibly check for the existence of external functions. But this kernel can then be successfully created and launched with the Level Zero API using zeModuleCreate, zeKernelCreate, and zeKernelSetArgumentValue, zeKernelSetGroupSize and zeCommandListAppendLaunchKernel. This is oddly disturbing.

So, I guess my question or issue with this is the following:
1) Is this the desired behavior that there seems to be no SPIR kernel linkage check whatsoever?
2) How can I check that all external functions being called actually are available in the driver/system?

JablonskiMateusz, 2022-01-21:
Hi @fwinter 
Level Zero API allows to create modules that import external symbols but you should get ZE_RESULT_ERROR_INVALID_MODULE_UNLINKED error when you create a kernel from module with unresolved symbols. 

Could you share spv file with kernel that reproduces the issue?

fwinter, 2022-01-21:
Hi @JablonskiMateusz 

Sure! The attached file contains calls to an non-existent external function (_Z5bogus) and the most recent level zero/compute-runtime can load/create kernel and launch (!) it just fine.

[reduction.zip](https://github.com/intel/compute-runtime/files/7915841/reduction.zip)

fwinter, 2022-01-21:
I also attach the kernel in LLVM IR from which the translator generated the SPIR from

[reduction.ll.zip](https://github.com/intel/compute-runtime/files/7915865/reduction.ll.zip)

JablonskiMateusz, 2022-01-24:
I looked at the llvm code, the function returns no value and has no side-effects. Therefore I hope IGC remove the function as it won't change anything.

```
; Function Attrs: readnone
declare spir_func void @_Z5bogus(i32) #0
attributes #0 = { readnone }
```

> readnone
On a function, this attribute indicates that the function computes its result (or decides to unwind an exception) based strictly on its arguments, without dereferencing any pointer arguments or otherwise accessing any mutable state (e.g. memory, control registers, etc) visible to caller functions. It does not write through any pointer arguments (including byval arguments) and never changes any state visible to callers. 

reference: https://llvm.org/docs/LangRef.html

Could you try with a function that has other attributes or with non-void return type?

fwinter, 2022-01-24:
@JablonskiMateusz Good catch! The 'readnone' attribute was the culprit. Before I had looked at the SPIR text, saw the call being there and simply assumed there would be a call. Apparently I hadn't paid attention to the attributes still being in SPIR (which makes total sense) and then being interpreted and acted on. With the attribute 'readnone' removed the kernel creation function throws an error just as it should. Great!

closed 2022-01-24 by fwinter