intel/compute-runtime #496: Can bundled CL and GL headers be updated?

frantisekz, 2022-01-22:
CL and GL headers in compute-runtime/third_party/ are few years out of date compared to upstream. Would it be possible / is it planned to update them?

I am working on packaging compute-runtime into Fedora ( https://bugzilla.redhat.com/show_bug.cgi?id=2036135 ) and it'd be nice if I could use system headers with:
```
    -DKHRONOS_HEADERS_DIR="/usr/include/CL/" \
    -DKHRONOS_GL_HEADERS_DIR="/usr/include/GL/" \
```

For CL, I've used: https://github.com/KhronosGroup/OpenCL-Headers/commit/1aa1139
For GL, I've used: https://github.com/NVIDIA/libglvnd/releases/tag/v1.4.0

The compilation fails however as it seems the headers changed a bit upstream. There are bunch of errors when I try to build this with system headers:

- bunch of  errors like : 'CL_EXT_PREFIX__VERSION_1_1_DEPRECATED' does not name a type; did you mean 'CL_API_PREFIX__VERSION_1_1_DEPRECATED'? (I was abe to fix this)
- multiple "does not name a type" for "KHRpfn_clCreateImage2D, KHRpfn_clCreateImage3D, ..." (I've fixed that too)
- redefinitions of CL_QUEUE_DEFAULT_CAPABILITIES_INTEL, struct _cl_queue_family_properties_intel and struct _cl_device_pci_bus_info_khr: commenting these out of compute-runtime's cl_ext_private.h might've caused the error I don't have many ideas how to fix:

```
/builddir/build/BUILD/compute-runtime-22.02.22151/opencl/source/cl_device/cl_device_info.cpp: In member function 'cl_int NEO::ClDevice::getDeviceInfo(cl_device_info, size_t, void*, size_t*)':
/builddir/build/BUILD/compute-runtime-22.02.22151/opencl/source/cl_device/cl_device_info.cpp:200:48: error: request for member 'data' in '((NEO::ClDevice*)this)->NEO::ClDevice::deviceInfo.NEO::ClDeviceInfo::queueFamilyProperties', which is of non-class type 'int'
  200 |         src = deviceInfo.queueFamilyProperties.data();
      |                                                ^~~~
/builddir/build/BUILD/compute-runtime-22.02.22151/opencl/source/cl_device/cl_device_info.cpp:201:62: error: request for member 'size' in '((NEO::ClDevice*)this)->NEO::ClDevice::deviceInfo.NEO::ClDeviceInfo::queueFamilyProperties', which is of non-class type 'int'
  201 |         retSize = srcSize = deviceInfo.queueFamilyProperties.size() * sizeof(cl_queue_family_properties_intel);
      |                                                              ^~~~
/builddir/build/BUILD/compute-runtime-22.02.22151/opencl/source/cl_device/cl_device_info.cpp:291:31: error: 'struct NEO::ClDeviceInfo' has no member named 'pciBusInfo'
  291 |             src = &deviceInfo.pciBusInfo;
      |                               ^~~~~~~~~~
/builddir/build/BUILD/compute-runtime-22.02.22151/opencl/source/cl_device/cl_device_info.cpp:292:51: error: 'struct NEO::ClDeviceInfo' has no member named 'pciBusInfo'
  292 |             retSize = srcSize = sizeof(deviceInfo.pciBusInfo);
```

Fixing of some of the mentioned errors was this draft patch:

```
---
 opencl/source/api/dispatch.h | 36 ++++++++++++++++++------------------
 1 file changed, 18 insertions(+), 18 deletions(-)

diff --git a/opencl/source/api/dispatch.h b/opencl/source/api/dispatch.h
index 322e403..a891144 100644
--- a/opencl/source/api/dispatch.h
+++ b/opencl/source/api/dispatch.h
@@ -149,7 +149,7 @@ typedef CL_API_ENTRY cl_mem(CL_API_CALL *KHRpfn_clCreateBuffer)(
     void *hostPtr,
     cl_int *errcodeRet) CL_API_SUFFIX__VERSION_1_0;
 
-typedef CL_API_ENTRY CL_EXT_PREFIX__VERSION_1_1_DEPRECATED
+typedef CL_API_ENTRY CL_API_PREFIX__VERSION_1_1_DEPRECATED
 cl_mem(CL_API_CALL *KHRpfn_clCreateImage2D)(
     cl_context context,
     cl_mem_flags flags,
@@ -158,9 +158,9 @@ cl_mem(CL_API_CALL *KHRpfn_clCreateImage2D)(
     size_t imageHeight,
     size_t imageRowPitch,
     void *hostPtr,
-    cl_int *errcodeRet) CL_EXT_SUFFIX__VERSION_1_1_DEPRECATED;
+    cl_int *errcodeRet) CL_API_SUFFIX__VERSION_1_1_DEPRECATED;
 
-typedef CL_API_ENTRY CL_EXT_PREFIX__VERSION_1_1_DEPRECATED
+typedef CL_API_ENTRY CL_API_PREFIX__VERSION_1_1_DEPRECATED
 cl_mem(CL_API_CALL *KHRpfn_clCreateImage3D)(
     cl_context context,
     cl_mem_flags flags,
@@ -171,7 +171,7 @@ cl_mem(CL_API_CALL *KHRpfn_clCreateImage3D)(
     size_t imageRowPitch,
     size_t imageSlicePitch,
     void *hostPtr,
-    cl_int *errcodeRet) CL_EXT_SUFFIX__VERSION_1_1_DEPRECATED;
+    cl_int *errcodeRet) CL_API_SUFFIX__VERSION_1_1_DEPRECATED;
[... 122 lines cut ...]
 } cl_device_pci_bus_info_khr;
+*/
 
 /************************************************
 *   cl_intel_mem_compression_hints extension    *
diff --git a/opencl/source/api/api.h b/opencl/source/api/api.h
index 6fa18f6..223bfe2 100644
--- a/opencl/source/api/api.h
+++ b/opencl/source/api/api.h
@@ -919,7 +919,7 @@ extern CL_API_ENTRY cl_event CL_API_CALL
 clCreateEventFromGLsyncKHR(
     cl_context context,
     cl_GLsync sync,
-    cl_int *errcodeRet) CL_EXT_SUFFIX__VERSION_1_2;
+    cl_int *errcodeRet) CL_API_SUFFIX__VERSION_1_2;
 
 extern CL_API_ENTRY cl_program CL_API_CALL clCreateProgramWithILKHR(
     cl_context context,
-- 
2.34.1

```

Apart from these, there is bunch of "The Intel extensions have been moved into cl_ext.h.  Please include cl_ext.h directly.'
   19 | #pragma message("The Intel extensions have been moved into cl_ext.h.  Please include cl_ext.h directly." for opencl/source/api/dispatch.h , but I don't think that's too much of a problem.

luluu9, 2022-06-29:
OpenGL and OpenCL headers are updated starting from [this commit](https://github.com/intel/compute-runtime/commit/5f7073cdfee231b9a85f40eb39bfc08347376479).

frantisekz, 2022-07-10:
@luluu9 thanks for heads-up, I am able to compile the .27 release with system CL/GL, there is just one issue with -DCL_TARGET_OPENCL_VERSION=300 which doesn't seem to get passed to the system CL headers resulting in bunch of: 

```
/usr/include/CL/cl_version.h:22:104: note: '#pragma message: cl_version.h: CL_TARGET_OPENCL_VERSION is not defined. Defaulting to 300 (OpenCL 3.0)'
   22 | #pragma message("cl_version.h: CL_TARGET_OPENCL_VERSION is not defined. Defaulting to 300 (OpenCL 3.0)")
      |  
```

JablonskiMateusz, 2022-08-23:
@frantisekz  could you check if https://github.com/intel/compute-runtime/commit/670295e697ef0e0bfef535b29eb5173eecff64ba fixes your issues?

frantisekz, 2022-08-23:
> @frantisekz could you check if [670295e](https://github.com/intel/compute-runtime/commit/670295e697ef0e0bfef535b29eb5173eecff64ba) fixes your issues?

Yes, that solves the issue. Thanks!, this can be closed then :)

closed 2022-08-25 by JablonskiMateusz