intel/compute-runtime #499: NULL not allowed as pointer argument for zeKernelSetArgumentValue?

fwinter, 2022-01-27:
When I try setting a pointer argument to zero using `zeKernelSetArgumentValue` the call fails with error code `0x78000004` (ZE_RESULT_ERROR_INVALID_ARGUMENT).

One could argue why would anyone set a pointer argument to a value that's definitely illegal to use? Well, consider a simplified kernel that appears in a stencil kernel in a multi MPI rank job:

```
__kernel void multi_node_stencil(__global int* v, __global float* out , __global float* a , __global float* b)
{
  int i = get_global_id(0);

  if (v[i] < 0)
    out[i] = b[ -v[i] - 1 ];
  else
    out[i] = a[ v[i] ];
}

```

Based on the sign of the index `a` the source value comes either from the local grid (a) or an MPI receive buffer (b). In circumstances where there's no off-node communication (a runtime choice) this kernel must still work. In this case I'd pass `NULL` as value for the last parameter, and I know the kernel executes still correctly since it would never read from `b`. However, Level Zero API seem to not like this and throws the above error.

This is how I use it to set the last parameter:

```
void* ptr = nullptr;
zeKernelSetArgumentValue( function , 3 , sizeof(ptr) , &ptr );

```

The API call `zeKernelSetArgumentValue` should accept a pointer to NULL, right?

jandres742, 2022-01-28:
@fwinter from looking at our code, we do accept NULL. However, we might return INVALID_ARGUMENT from zeKernelSetArgumentValue in some occasions. Either you are getting that as a bug in the driver or application. could you share full reproducer?

fwinter, 2022-01-28:
Hi @jandres742 , here's the reproducer. 
The kernel which gets loaded is a bit more complicated than the example I've given earlier since it's the real world kernel but the idea is still visible. When I try to set the 4th argument (counting from 0) with the pointer to NULL then the call exits with:

Error at zeKernelSetArgumentValue(kernel, 4, sizeof(dummy), &dummy): main: 184
Exit with Error Code: 0x78000004

define spir_kernel void @eval_scalar0(i32 %arg0, i32 addrspace(1)* %arg1, float addrspace(1)* %arg2, i32 addrspace(1)* %arg3, float addrspace(1)* %arg4, float addrspace(1)* %arg5)

[bug_level_zero_invalid_argument.zip](https://github.com/intel/compute-runtime/files/7960295/bug_level_zero_invalid_argument.zip)

JablonskiMateusz, 2022-02-28:
Hi @fwinter  I think you should explicitly pass nullptr instead of pointer to nullptr, e.g.
```
zeKernelSetArgumentValue( function , 3 , sizeof(void *) , nullptr);
```
As per spec,
> pArgValue: [in][optional] argument value represented as matching arg type. If null then argument value is considered null.

fwinter, 2022-02-28:
That makes sense. Thanks!

closed 2022-02-28 by fwinter