intel/compute-runtime #502: 404 on intel-igc-core_1.0.9933_amd64.deb & intel-igc-opencl_1.0.9933_amd64.deb

Pookalton, 2022-02-01:
Missing files
> --2022-02-01 18:26:45--  https://github.com/intel/intel-graphics-compiler/releases/download/igc-64.deb/intel-igc-core_1.0.9933_amd64.deb
> Resolving github.com (github.com)... 13.236.229.21
> Connecting to github.com (github.com)|13.236.229.21|:443... connected.
> HTTP request sent, awaiting response... 404 Not Found
> 2022-02-01 18:26:46 ERROR 404: Not Found.

--2022-02-01 18:25:46--  https://github.com/intel/intel-graphics-compiler/releases/download/igc-64.deb/intel-igc-opencl_1.0.9933_amd64.deb
Resolving github.com (github.com)... 13.236.229.21
Connecting to github.com (github.com)|13.236.229.21|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2022-02-01 18:25:47 ERROR 404: Not Found.

wzmudat, 2022-02-01:
Issue is fixed: https://github.com/intel/compute-runtime/releases/tag/22.04.22286

wzmudat, 2022-02-01:
Similar issue that was fixed and closed: https://github.com/intel/compute-runtime/issues/500

closed 2022-02-01 by wzmudat