intel/compute-runtime #504: SIGSEGV on: DeviceFactoryTest.givenDebugFlagWithZeroXWhenPrepareDeviceEnvironmentsForProductFamilyOverrideIsCalledThenOverrideDeviceIdToHexValue

anbe42, 2022-02-03:
While building 22.04.22286 on Debian using igc 1.0.9933 built against llvm-12, some tests fail reproducibly.
When using igc 1.0.9933 built against llvm-11, the build succeeds and all tests pass.

This is a regression compared to 22.03.22192, and I bisected it to commit e39a09e6db7a06f4d75f733fe8b58a68b6624118 (the first commit after the 22.03.22192 tag).

The build was configured with `cmake -DTESTS_XE_HP_SDV=0 ..` due to issue #498
```
....
make  -f target_unit_tests/xe_hpg_core/dg2/CMakeFiles/run_dg2_0_unit_tests.dir/build.make target_unit_tests/xe_hpg_core/dg2/CMakeFiles/run_dg2_0_unit_tests.dir/build
make[2]: Entering directory '/tmp/compute-runtime/build'
cd /tmp/compute-runtime/build/bin && echo Running neo_shared_tests 2x4x5 in /tmp/compute-runtime/build/bin/dg2
Running neo_shared_tests 2x4x5 in /tmp/compute-runtime/build/bin/dg2
cd /tmp/compute-runtime/build/bin && /tmp/compute-runtime/build/bin/neo_shared_tests --product dg2 --slices 2 --subslices 4 --eu_per_ss 5 --gtest_repeat=1 --gtest_shuffle --gtest_random_seed=0 --disable_default_listener --rev_id 0
product family: dg2 (1270)
set timeout to: 45
Iteration: 1. random_seed: 49075

========================
==  DG2 ULTs PASSED   ==
========================
Tests run:      4278
Tests passed:   4255
Tests skipped:  23
Tests failed:   0
Tests disabled: 0
 Time elapsed:  510 ms
========================
cd /tmp/compute-runtime/build/bin && echo Running igdrcl_tests 2x4x5 in /tmp/compute-runtime/build/bin/dg2
Running igdrcl_tests 2x4x5 in /tmp/compute-runtime/build/bin/dg2
cd /tmp/compute-runtime/build/bin && /tmp/compute-runtime/build/bin/igdrcl_tests --product dg2 --slices 2 --subslices 4 --eu_per_ss 5 --gtest_repeat=1 --gtest_shuffle --gtest_random_seed=0 --disable_default_listener --rev_id 0
product family: dg2 (1270)
enable SIGALRM handler: 1
enable SIGSEGV handler: 1
enable SIGABRT handler: 1
set timeout to: 45
Iteration: 1. random_seed: 50156
SIGSEGV on: DeviceFactoryTest.givenDebugFlagWithZeroXWhenPrepareDeviceEnvironmentsForProductFamilyOverrideIsCalledThenOverrideDeviceIdToHexValue
SIGABRT on: DeviceFactoryTest.givenDebugFlagWithZeroXWhenPrepareDeviceEnvironmentsForProductFamilyOverrideIsCalledThenOverrideDeviceIdToHexValue
Aborted
make[2]: *** [target_unit_tests/xe_hpg_core/dg2/CMakeFiles/run_dg2_0_unit_tests.dir/build.make:76: run_dg2_0_unit_tests] Error 134
make[2]: Leaving directory '/tmp/compute-runtime/build'
...
```

JablonskiMateusz, 2022-02-04:
Hi @anbe42
I think the main reason you hit the issue is `-DTESTS_XE_HP_SDV=0`. Some unit tests wrongly rely on XeHpSdv as a default supported platform while if there are no XeHpSdv tests then we don't initialize related structures. I reproduced the issue locally and I'll provide a fix soon. 
As a workaround, you can disable XeHpSdv support using `-DSUPPORT_XE_HP_SDV=0 -DSUPPORT_XE_HP_CORE=0`

JablonskiMateusz, 2022-02-07:
@anbe42  could you try with https://github.com/intel/compute-runtime/commit/10e4e626001248938e3e767cf7ceed148669b135 ?

anbe42, 2022-02-08:
@JablonskiMateusz: Seems to work fine. Thanks!

JablonskiMateusz, 2022-02-08:
Great! I'm closing the issue.

closed 2022-02-08 by JablonskiMateusz