intel/compute-runtime #507: Integrate Beignet drivers for 3rd-4th Gen Core OCL Support

hilga007, 2022-02-09:
[Beignet](https://github.com/intel/beignet)  would probably only have OpenCL 2.0 support, but it would nonetheless be great to see that driver fully integrated here to better utilize OpenCL on say, NUCs with the newer icx compiler on Gentoo (for example)

antermin, 2022-04-18:
Apparently Beignet does not support Gen6 (Sandy Bridge).
Also, from ArchWiki and Mesa source code, apparently Clover only works for AMD (and limited support for iris+clover), while crocus+clover (Gen7 and Gen8LP use crocus) is not supported.

For Gen8LP, there was a feature request, but they [decided](https://github.com/intel/compute-runtime/issues/317#issuecomment-688236589) not to support Gen8LP in NEO Compute Runtime.

Maybe you will also want to check out the [WIP rusticl in Mesa](https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/15439) and [this comment](https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/15439#note_1304834).

hilga007, 2022-04-21:
Who to contact to request beignet simply be merged into mesa for
mesa-opencl which could include its AMD backends + a bringup to modern llvm
for beignet. It is for "outdated" tech (Ivy Bridge, Haswell) but still has
uses.

For example: budget-strapped healthcare organizations could benefit from
some ultrasound imaging that has a huge OpenCL perf boost vs. CPU software
computations. These organizations are stuck on Sandy Bridge / Ivy Bridge /
Haswell, etc.

Thoughts?

eero-t, 2022-04-26:
Old drivers (in this case beignet) are for supporting old HW.  It does not make sense to have single driver supporting all HW due to differences in how the HW is programmed and what features that HW supports.  At some point there just needs to be clean break to keep driver maintainable.

Btw. It's not just compute where the driver changed between GEN8 <-> GEN9 (beignet -> compute-runtime), also Mesa (i965/crocus -> Iris OpenGL driver) and Media (i965 -> iHD VA-API driver) did that.

And questions about Beignet and Mesa belong to those projects, not here.  Please close this ticket.

JablonskiMateusz, 2022-05-17:
Hi @hilga007, Gen6 and Gen7 GPUs are supported by beignet driver, please create an issue in their repo https://github.com/intel/beignet.

I'm closing this issue as we are not planning to support these GPUs.

closed 2022-05-17 by JablonskiMateusz

hilga007, 2022-05-17:
Apparently they no longer are, either, hence asking to have them included
in the supported driver.