intel/compute-runtime #508: Build issue

mukeshgarg81, 2022-02-14:
Followed build steps as mentioned on "https://github.com/intel/compute-runtime/blob/master/BUILD.md"

Error logs:-
[  6%] Linking CXX shared library ../../bin/libocloc.so
[  6%] Built target ocloc_lib
Scanning dependencies of target ocloc
[  6%] Building CXX object offline_compiler/source/CMakeFiles/ocloc.dir/main.cpp.o
[  6%] Linking CXX executable ../../bin/ocloc
[  6%] Built target ocloc
[  6%] Generating ../../../../bin/built_ins/x64/spirv/copy_image3d_to_buffer_stateless.spv
Command was: workspace/build/bin/ocloc -q -spv_only -file workspace/neo/shared/source/built_ins/kernels/copy_image3d_to_buffer_stateless.builtin_kernel -out_dir workspace/build/bin/built_ins/x64/spirv/ -output_no_suffix -options "-cl-kernel-arg-info"
make[2]: *** [shared/source/built_ins/kernels/CMakeFiles/builtins_spirv.dir/build.make:238: bin/built_ins/x64/spirv/copy_image3d_to_buffer_stateless.spv] Error 250
make[1]: *** [CMakeFiles/Makefile2:3396: shared/source/built_ins/kernels/CMakeFiles/builtins_spirv.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  7%] Building CXX object shared/source/CMakeFiles/neo_shared.dir/gen9/state_base_address_gen9.cpp.o
[  7%] Building CXX object shared/source/CMakeFiles/neo_shared.dir/gen9/tbx_command_stream_receiver_gen9.cpp.o
[  7%] Building CXX object igdrcl_lib_release/CMakeFiles/igdrcl_lib_release.dir/event/event.cpp.o
[  7%] Building CXX object shared/source/CMakeFiles/neo_shared.dir/gen9/image_core_gen9.cpp.o
[  7%] Building CXX object shared/source/CMakeFiles/neo_shared.dir/gen9/hw_info_skl.cpp.o
[  7%] Building CXX object shared/source/CMakeFiles/neo_shared.dir/gen9/hw_info_kbl.cpp.o
[  7%] Building CXX object shared/source/CMakeFiles/neo_shared.dir/gen9/hw_info_glk.cpp.o

eero-t, 2022-02-17:
Which version of libgmm, opencl-clang, SPRV headers + tools + LLVM-translator and IGC you are using, and which compute-runtime version you are trying to build, and on which OS version?

JablonskiMateusz, 2022-08-22:
Hi @mukeshgarg81 , is it still a valid issue?

closed 2022-10-13 by AdamCetnerowski