intel/compute-runtime #510: zesDeviceEnumPsus() returns ZE_RESULT_ERROR_UNSUPPORTED_FEATURE

eero-t, 2022-02-21:
I'm filing this because:
* This return value is not documented for this function in the spec: https://spec.oneapi.io/level-zero/latest/sysman/api.html#zesdeviceenumpsus
* It differs from what all the other Sysman enumeration calls return

I would expect zesDeviceEnumPsus() to return successfully (unless there's some exceptional failure like device loss or alloc failure), and just return zero power supplies if there are none it supports, like all the other enumeration functions.

PS: internal Sysman backend version works the same (with internal kernel / FW e.g. on DG2) in this respect as the latest public "22.07.22465" compute-runtime release (with latest public drm-tip Git kernel on DG1).  I.e. this is not specific to the public version.

eero-t, 2022-04-25:
Still happens with the (last week) drm-tip 5.18-rc3 kernel, GuC 70.1.1 and latest compute-runtime "22.16.22992" release.

jandres742, 2022-06-14:
thanks @eero-t 

> I would expect zesDeviceEnumPsus() to return successfully (unless there's some exceptional failure like device loss or alloc > failure), and just return zero power supplies if there are none it supports, like all the other enumeration functions

I guess this is a matter of opinion on how to implement unsupported features :)

If we return SUCCESS and 0 domains as you suggest, it might imply that the L0 driver actually tried to read the PSU domains, but didnt find any, so we return 0. 

However, by returning UNSUPPORTED, the L0 driver is being more honest, indicating that it didnt even try to search for domains, since the feature is not currently supported, so it is not that it found 0 domains, it is that we dont even know where or what to look for :)

> This return value is not documented for this function in the spec:

This is a generic error we return, as indicated in the spec. See here:

https://spec.oneapi.io/level-zero/latest/core/api.html?highlight=unsupported#_CPPv4N11ze_result_t35ZE_RESULT_ERROR_UNSUPPORTED_FEATUREE

enumerator ZE_RESULT_ERROR_UNSUPPORTED_FEATURE
[Validation] generic error code for unsupported features

Maybe we could have it in the spec on every function, but I guess spec has defined generically in the list of errors.

eero-t, 2022-06-15:
Spec "Returns" part should either:
* list *all* valid values,
* refer to part of spec listing them, or
* state that it lists only subset of values

Please file bug about this for the spec.

I'm closing this one as based on you description backend works as expected.

closed 2022-06-15 by eero-t