intel/compute-runtime #518: Looking for MediaSDK driver components..

ketxxx, 2022-03-10:
This might be a lost cause but in short I'm looking for updated MediaSDK Win10 64bit driver components for the intel HD3000 IGP, its far from being an amazing IGP I know but the Windows Update Catalog has HD3000 drivers dated 2016, version 9.17.10.4459 which were supposedly provided by intel but are missing the MediaSDK components for hardware encoding support which naturally makes the drivers by themselves pretty useless. The latest driver intel themselves offer is 9.17.10.4229 which is only a Win8.1 driver so pulling the MediaSDK components from that package isn't ideal. My goal is simply to put together a true Win10 64bit driver for the IGP and the MediaSDK component is the last piece I'm in search of so literally anything newer than the components in the 4229 driver will do here.

I'm assuming updated MediaSDK components must exist as a driver does, but I'll be damned if I can find them.

dmitryermilov, 2022-03-11:
Hi @ketxxx ,

Not sure how it happened that 9.17.10.4459 in the Windows Update Catalog doesn't have MediaSDK libs. It was ~6 years ago, hard to find details. I don't see 4459 on public Intel websites so can't compare Intel distributed package with what is available at MSFT side. 
However I see Intel internal 9.17.10.4459 package has all the libs. All MediaSDK libs in 4459 are b2b the same as libs from 4229.

ketxxx, 2022-03-11:
Hi @dmitryermilov,

Thanks for the verification thats at least one less thing I have to concern myself with. One last long shot, it came to my attention yesterday that OpenGL support for the HD3000 in Win10 is completely broken, certainly a compatibility issue and one that seems to be more on Win10s side than it does the OGL DLLs as some people report having "full" OGL 3.1 support (despite the IGP being cable of OGL 3.3... lazy intel) via use of the MS Application Compatibility Tools but that workaround doesn't seem to work for everyone, or on later builds of W10. So my question is; would you happen to have come across OGL DLLs for HD3000 that are fully compatible with W10? I'm not even looking for something that brings the OGL spec back up to 3.1 just something thats far better than the MS generic driver that hobbles everyone with OGL 1.1. Ironically with no HD3000 driver installed at all just running of the Basic Display Adapter driver W10 detects and defaults HD3000 to OGL 3.3 compatibility profile using Meas 22.0 :)

dmitryermilov, 2022-03-11:
Sorry @ketxxx , I can't answer this part. I'm from the media side (MSDK, media driver), hope compute folks can help. Please rename the issue or create another one to draw their attention.

ketxxx, 2022-03-11:
Ok, thanks for your help @dmitryermilov :o)

closed 2022-03-11 by ketxxx