intel/compute-runtime #519: Build failure: CommandLine Error: Option 'spirv-text' registered more than once!

tgurr, 2022-03-11:
I'm running into the following build failure:
```
[...]
make  -f shared/source/built_ins/kernels/CMakeFiles/builtins_spirv.dir/build.make shared/source/built_ins/kernels/CMakeFiles/builtins_spirv.dir/depend
make[2]: Entering directory '/var/tmp/paludis/build/dev-libs-intel-compute-runtime-22.09.22577/work/build'
[ 29%] Generating ../../../../bin/built_ins/x64/spirv/fill_image3d.spv
cd /var/tmp/paludis/build/dev-libs-intel-compute-runtime-22.09.22577/work/compute-runtime-22.09.22577/shared/source/built_ins/kernels && LD_LIBRARY_PATH=/var/tmp/paludis/build/dev-libs-intel-compute-runtime-22.09.22577/work/build/bin /var/tmp/paludis/build/dev-libs-intel-compute-runtime-22.09.22577/work/build/bin/ocloc -q -spv_only -file /var/tmp/paludis/build/dev-libs-intel-compute-runtime-22.09.22577/work/compute-runtime-22.09.22577/shared/source/built_ins/kernels/fill_image3d.builtin_kernel -out_dir /var/tmp/paludis/build/dev-libs-intel-compute-runtime-22.09.22577/work/build/bin/built_ins/x64/spirv/ -output_no_suffix -options -cl-kernel-arg-info
: CommandLine Error: Option 'spirv-text' registered more than once!
LLVM ERROR: inconsistency in registered CommandLine options
make[2]: Leaving directory '/var/tmp/paludis/build/dev-libs-intel-compute-runtime-22.09.22577/work/build'
make[1]: Leaving directory '/var/tmp/paludis/build/dev-libs-intel-compute-runtime-22.09.22577/work/build'
make[2]: *** [shared/source/built_ins/kernels/CMakeFiles/builtins_spirv.dir/build.make:223: bin/built_ins/x64/spirv/fill_image3d.spv] Aborted (core dumped)
make[1]: *** [CMakeFiles/Makefile2:2579: shared/source/built_ins/kernels/CMakeFiles/builtins_spirv.dir/all] Error 2
make: *** [Makefile:159: all] Error 2
```

compute-runtime version I'm trying to build:
* 22.09.22577

Installed dependencies: 
* gmmlib: 22.0.3
* igc: 1.0.10409 (compiled using the "bundled" vc-intrinsics/spirv-headers/spirv-tools)
* llvm: 13.0.1
* opencl-clang: 13.0.0
* os: linux 64bit
* gcc 11.2.0

Complete build log:
[intel-compute-runtime-22.09.22577.log](https://github.com/intel/compute-runtime/files/8230815/intel-compute-runtime-22.09.22577.log)

tgurr, 2022-03-11:
Passing `-DNEO_DISABLE_BUILTINS_COMPILATION:BOOL=TRUE` as e.g. Fedora does https://src.fedoraproject.org/rpms/intel-compute-runtime/blob/rawhide/f/intel-compute-runtime.spec appears to work around the build issue however I'm then still getting the error on runtime:

```
user@localhost ~ $ clinfo 
: CommandLine Error: Option 'spirv-text' registered more than once!
LLVM ERROR: inconsistency in registered CommandLine options
Aborted (core dumped)
user@localhost ~ $ clpeak 
: CommandLine Error: Option 'spirv-text' registered more than once!
LLVM ERROR: inconsistency in registered CommandLine options
Aborted (core dumped)
```

eero-t, 2022-05-24:
@tgurr everything in your compute stack needs to use the same LLVM and SPIRV versions, otherwise you get that kind of mismatch issues. See last comments in: https://github.com/intel/intel-graphics-compiler/issues/224

You need to build both opencl-clang and llvm-spirv yourself, or take them both from the system, not mix them.

tgurr, 2022-07-28:
@tgurr sorry for the late response, it took me a while to get a new release to actually build and work, so what I found out is that as soon as I built igc with vc-intrinsics support by either checking out and building vc-intrinsics in-place with igc or having it packaged separately and passing `-DIGC_OPTION__VC_INTRINSICS_MODE=Prebuilds` it results in the reported error. igc itself builds fine but compute-runtime fails with the above mentioned error. If I pass `-DIGC_BUILD__VC_ENABLED:BOOL=FALSE` to igc then compute-runtime builds and works fine (though I guess I'm missing some kind of feature/optimization then?)

I do use prebuilds from system for the mentioned packages (which are all built against the same LLVM version), see:
```
-- [Clang] Will use Clang prebuilds
-- IGC_LLVM_SPIRV_PATHS: /var/tmp/paludis/build/dev-util-intel-graphics-compiler-1.0.11485/work/intel-graphics-compiler-igc-1.0.11485/external/llvm/../../../SPIRV-LLVM-Translator;/var/tmp/paludis/build/dev-util-intel-graphics-compiler-1.0.11485/work/intel-graphics-compiler-igc-1.0.11485/external/llvm/../../../../SPIRV-LLVM-Translator;/var/tmp/paludis/build/dev-util-intel-graphics-compiler-1.0.11485/work/intel-graphics-compiler-igc-1.0.11485/external/llvm/../../../llvm-spirv;/var/tmp/paludis/build/dev-util-intel-graphics-compiler-1.0.11485/work/intel-graphics-compiler-igc-1.0.11485/external/llvm/../../../../llvm-spirv
-- [SPIRV] Will use SPIRV translator prebuilds
-- [LLVM] Current value of CMAKE_CURRENT_SOURCE_DIR : /var/tmp/paludis/build/dev-util-intel-graphics-compiler-1.0.11485/work/intel-graphics-compiler-igc-1.0.11485/IGC
-- [lld] Will use lld prebuilds
-- [LLVM] Current value of CMAKE_CURRENT_SOURCE_DIR : /var/tmp/paludis/build/dev-util-intel-graphics-compiler-1.0.11485/work/intel-graphics-compiler-igc-1.0.11485/IGC
-- [IGC] IGC will take prebuilt LLVM
-- [IGC] Searching for prebuilt LLVM in /llvm_prebuilt_windows;/prebuild-llvm;/opt/intel-llvm-static-13.0.1 and system directories
-- Found ZLIB: /usr/x86_64-pc-linux-gnu/lib/libz.so (found version "1.2.12") 
-- Found LibXml2: /usr/x86_64-pc-linux-gnu/lib/libxml2.so (found version "2.9.14") 
-- [IGC] Found LLVM: /usr/x86_64-pc-linux-gnu/lib/llvm/13/lib/cmake/llvm
-- Linker detection: GNU ld
-- [IGC] Trying to find prebuilt SPIRV library
-- Found SPIRVLLVMTranslator: /usr/x86_64-pc-linux-gnu/lib/llvm/13/lib/libLLVMSPIRVLib.a (found suitable version "13.0.0.0", minimum required is "13")  
-- ============================ SPIRV-Tools project ============================
-- [SPIRV-Tools] : IGC_OPTION__SPIRV_TOOLS_MODE set to Prebuilds
-- [SPIRV-Tools] : IGC__OPTION_USE_PREINSTALLED_SPRIV_HEADERS set to ON, using preinstalled SPIRV-Headers
-- [SPIRV-Tools] : Using preinstalled SPIRV-Headers
-- [SPIRV-Tools] : Searching prebuilt SPIRV-Tools package
-- [SPIRV-Tools] : Found prebuilt SPIRV-Tools package in: /usr/x86_64-pc-linux-gnu/lib64/cmake/SPIRV-Tools/SPIRV-ToolsConfig.cmake
-- =============================================================================
CMake Warning at IGC/cmake/igc_find_protobuf.cmake:25 (message):
  Cannot find Protoc program or library for Protobuf, please visit
  https://github.com/protocolbuffers/protobuf/releases and install - disable
  of IGC Metrics
Call Stack (most recent call first):
  IGC/CMakeLists.txt:1397 (include)

-- [IGC] : opencl-clang will be taken from system
-- [IGC] Found clang-13 executable: /usr/bin/x86_64-pc-linux-gnu-clang-13
-- [IGC] Found opencl-c.h: /usr/bin/../lib/llvm/13/lib/clang/13/include/opencl-c.h
-- [CMFE] CM Frontend headers are taken from: /var/tmp/paludis/build/dev-util-intel-graphics-compiler-1.0.11485/work/intel-graphics-compiler-igc-1.0.11485/IGC/CMFE/AdaptorCM
-- [CMFE] Using CMFE library name: libclangFEWrapper.so
-- +++ Source/IGC/VectorCompiler +++
-- [VC] Build proj: VectorCompiler
-- [VC] : Searching prebuilt vc-intrinsics package
-- [VC] : Found prebuilt vc-intrinsics package in: /usr/x86_64-pc-linux-gnu/lib64/cmake/VCIntrinsics13/VCIntrinsics13Config.cmake
```

Looks like Gentoo is also running into a similar or the same issue as they disable VC as well:
https://gitweb.gentoo.org/repo/gentoo.git/tree/dev-util/intel-graphics-compiler/intel-graphics-compiler-1.0.11485-r1.ebuild#n84

NixOS was also running into it according to a comment by @calbrecht at https://github.com/NixOS/nixpkgs/issues/169729#issuecomment-1114339505 however I can't find any reference on how they solved the issue.

eero-t, 2022-07-28:
So far, I've seen "registered more than once" errors only when even one component in the whole stack used mismatching LLVM version at build time.  Although you've tried to make sure that build uses only single LLVM version, do you other LLVM version(s) installed on your system?

(To make sure your builds are really avoiding wrong LLVM versions, it may be best to do a test build in a container which has nothing else than components built by the same LLVM installed.  If that does not work, then we can be sure it's not LLVM mismatch.)

Btw. Based on this: https://github.com/intel/intel-graphics-compiler/projects?type=classic

IGC production testing is still done against LLVM 11, with LLVM 14 being next one to be production supported, and LLVM 12 + 13 having just beta support.  While I can build compute stack (with VC) fine with LLVM 12, compute-runtime tests fail with LLVM 13 on Ubuntu 22.04: https://github.com/intel/compute-runtime/issues/553

So LLVM 13 might be an issue too.

tgurr, 2022-07-28:
I do not have any other LLVM versions on the system, everything is handled by the package manager and nothing is/was installed manually. Exherbo is a source-based distribution like Gentoo so I can be quite sure about this.

I'll migrate our packages to LLVM 14 as soon as we've packaged that, currently LLVM 13.0.1 is the latest we've packaged to date. I'll report back if the situation improved with LLVM 14 and also monitor the Gentoo progress on https://github.com/intel/intel-graphics-compiler/issues/236.

calbrecht, 2022-07-28:
@tgurr that particular issue was related to a `libLLVMSPIRVLib.a` beeing present in the linker dependencies (iirc and with rather limited to zero understanding of the issue at hand on my side). Also iirc, this was fixed by changing `SPIRV-LLVM-Translator` to not output static libs, through`-DBUILD_SHARED_LIBS=YES` and potentially some other lines from https://github.com/NixOS/nixpkgs/pull/171656/files

tgurr, 2022-07-29:
@calbrecht huge thanks for that hint! We actually still built `SPIRV-LLVM-Translator` as static library because of running into a runtime error of its `llvm-spirv` command not being able to find its shared library when built with `-DBUILD_SHARED_LIBS:BOOL=TRUE`. After applying a (dirty) workaround and having `SPIRV-LLVM-Translator` now built as shared library I was also able to successfully build igc against it and compute-runtime now also built and runs without segfaulting. Closing this issue and agian **huge** thanks to anyone involved!

closed 2022-07-29 by tgurr

eero-t, 2022-07-29:
Maybe this is related to IGC docs already stating that static version LLVM-SPIRV-Tools is not supported: https://github.com/intel/intel-graphics-compiler/commit/205e9c3def7f57858f4a4f70665c7bd991a12e35 ?

(See also the linked ticket in the Khronos SPIRV project.)

tgurr, 2022-07-29:
We previously already built `SPIRV-Tools` as shared but not `SPIRV-LLVM-Translator`, maybe the note in the docs could be expanded to also cover that statement for `SPIRV-LLVM-Translator` additionally to just `SPIRV-Tools`.

bvbfan, 2022-11-17:
I compile graphics compiler https://github.com/intel/intel-graphics-compiler/issues/263#issuecomment-1315224158 but I face the 
```
[ 20%] Generating ../../../../bin/built_ins/x64/spirv/fill_image3d.spv
[0]: /home/toni/intel-compute-runtime/src/build/bin/libocloc.so(_ZN16SafetyGuardLinux9sigActionEiP9siginfo_tPv+0x30) [0x7f73e77d9850]
```
probably compiler is in broken state, but out of the scope, entire project architecture is madness, broken binary interface is unstable...