intel/compute-runtime #521: Any compute folks about to help with this OGL issue on intel HD3000?

ketxxx, 2022-03-11:
One last long shot I need to take in making a true 64bit Win10 driver for the intel HD3000 IGP (if anyone wants a bit more insight into what I'm doing you can see the case I closed here; https://github.com/intel/compute-runtime/issues/518), it came to my attention yesterday that OpenGL support for the HD3000 in Win10 is completely broken, certainly a compatibility issue and one that seems to be more on Win10s side than it does the OGL DLLs as some people report having "full" OGL 3.1 support (despite the IGP being capable of OGL 3.3) via use of the MS Application Compatibility Tools but that workaround doesn't seem to work for everyone, or on later builds of W10 even with the ADK for W10 builds for 2004 and newer. So my question is; does anyone have OGL DLLs for HD3000 that are fully compatible with W10? I'm not even looking for something that brings the OGL spec back up to 3.1 just something thats far better than the MS generic driver that hobbles everyone with OGL 1.1 when the HD3000 driver is installed. Ironically with no HD3000 driver installed at all just running off the Basic Display Adapter driver W10 detects and defaults HD3000 to OGL 3.3 compatibility profile using Mesa 22.0 :)

Any assist would be greatly appreciated by anyone and everyone with intel HD3000 IGP.

eero-t, 2022-03-17:
Please close the ticket, this project OpenCL / OneAPI L0 driver, **not** OpenGL driver.

ketxxx, 2022-03-18:
Well, that is unfortunate. I guess 70%~ functionality/compatibility is going to be as good as it gets for people with HD3000 IGPs which to be frank rankles us more than just a bit especially as HD4000 gets fully compatible W10 drivers despite it's announcement date only being 50 days after HD3000. Thanks anyway.

closed 2022-03-18 by ketxxx