intel/compute-runtime #522: Miscompilation on Tiger Lake Xe GPU with early loop exit

maleadt, 2022-03-15:
I'm experiencing a miscompilation with our Julia-based oneAPI stack. The reproducer is as follows (I've documented the code, but it should be very straightforward even without knowing Julia):

```julia
using oneAPI

# simple kernel that copies `len` values from `src` to `dst`.
# each thread processes `elem_per_thread` values
function kernel(dest, src, elem_per_thread, len)
    for j in 1:elem_per_thread
        # compute the (1-based) index to copy
        i = get_local_id(0) + (get_group_id(0) - 1) * get_local_size(0)
        i += (j - 1) * get_local_size(0) * get_num_groups(0)

        # this early-exit causes the miscompilation
        i > len && return

        val = unsafe_load(src, i)
        unsafe_store!(dest, val, i)
    end
    return
end

function main(T=Int32, n=10)
    # allocate two vectors of 100 elements
    A = oneAPI.ones(T, n*n)
    B = oneAPI.zeros(T, n*n)
    ptrA = reinterpret(Core.LLVMPtr{T,AS.Global}, pointer(A))
    ptrB = reinterpret(Core.LLVMPtr{T,AS.Global}, pointer(B))

    # invoke the kernel with 10 groups of 10 threads
    @oneapi items=n groups=n kernel(ptrB, ptrA, 1, n*n)

    # display the result as an 10x10 matrix
    reshape(Array(B), (n,n))
end
```

The LLVM IR we generate for this kernel looks as follows:

```llvm
source_filename = "text"
target datalayout = "e-i64:64-v16:16-v24:32-v32:32-v48:64-v96:128-v192:256-v256:256-v512:512-v1024:1024"
target triple = "spir64-unknown-unknown"

declare i64 @_Z12get_local_idj(i32)

declare i64 @_Z12get_group_idj(i32)

declare i64 @_Z14get_local_sizej(i32)

declare i64 @_Z14get_num_groupsj(i32)

define spir_kernel void @_Z17julia_kernel_78787LLVMPtrI5Int32Li1EES_IS0_Li1EE5Int64S1_(i8 addrspace(1)* %0, i8 addrspace(1)* %1, i64 signext %2, i64 signext %3) local_unnamed_addr #0 {
entry:
  %.inv = icmp sgt i64 %2, 0
  %4 = select i1 %.inv, i64 %2, i64 0
  br i1 %.inv, label %L13.preheader, label %L56

L13.preheader:                                    ; preds = %entry
  %5 = bitcast i8 addrspace(1)* %1 to i32 addrspace(1)*
  %6 = bitcast i8 addrspace(1)* %0 to i32 addrspace(1)*
  br label %L13

L13:                                              ; preds = %L57, %L13.preheader
  %value_phi4 = phi i64 [ %23, %L57 ], [ 1, %L13.preheader ]
  %7 = call i64 @_Z12get_local_idj(i32 0)
  %8 = add i64 %7, 1
  %9 = call i64 @_Z12get_group_idj(i32 0)
  %10 = call i64 @_Z14get_local_sizej(i32 0)
  %11 = mul i64 %10, %9
[... 138 lines cut ...]
               OpStore %56 %55 Aligned 1
      %_not1 = OpIEqual %bool %value_phi4 %26
         %31 = OpIAdd %ulong %value_phi4 %ulong_1
               OpBranchConditional %_not1 %L56 %L13
               OpFunctionEnd
```

Executing this simple kernel should evidently yield an output consisting of all ones, but instead I'm getting zeros at certain places:

``` 
 1  1  1  1  1  1  1  1  1  1
 1  1  1  1  1  1  1  1  1  1
 1  1  1  1  1  1  1  1  1  1
 1  1  1  1  1  1  1  1  1  1
 1  1  1  1  1  1  1  1  1  1
 1  1  1  1  1  1  1  1  1  1
 1  1  1  1  1  1  1  1  1  1
 1  1  1  1  1  1  1  1  1  1
 1  0  0  0  0  0  0  0  0  0
 1  1  1  1  1  1  1  1  1  1
```

The exact pattern is consistent, but depends on the amount of threads/blocks. It seems caused by the early exit from the loop, even though that code isn't hit (in this example I'm launching exactly 10*10=100 threads for arrays of length 100). It also doesn't reproduce on an Intel HD Graphics P630. I'm using compute-runtime 22.8.22549, which depends on IGC 1.0.10395, on a 64-bit Linux running kernel 5.10.

I hope this is enough information to help isolate the issue. Feel free to reach out to me for more information, or if needed for help to set-up a Julia environment (which is really easy).

jandres742, 2022-03-15:
thanks @maleadt . this is Julia over oneAPI. would https://github.com/intel/llvm be more appropriate?

maleadt, 2022-03-16:
Why? We don't use any code from that repository. Does the SPIR-V look invalid maybe? Since we generate the very same SPIR-V for other Intel hardware, where it executes correctly, I'd say this is a bug in the back-end compiler (i.e. this repo or IGC).

jandres742, 2022-03-16:
@maleadt 
Sorry, I got confused by the some parts of the description.
So you have julia code, that generates IR, which then generates SPIRV, and then you call L0 underneath to offload to the GPU, right? Could you provide the ze_tracer -c logs to see what the code does?

maleadt, 2022-03-16:
Sure:

```
>>>> [1199929503] zeInit: flags = 0
<<<< [1199969483] zeInit [39980 ns] -> ZE_RESULT_SUCCESS(0x0)
>>>> [3604389460] zeDriverGet: pCount = 0x7fff9d08f558 (Count = 0) phDrivers = 0
<<<< [3604412814] zeDriverGet [23354 ns] Count = 1 -> ZE_RESULT_SUCCESS(0x0)
>>>> [3604419091] zeDriverGet: pCount = 0x7fff9d08f558 (Count = 1) phDrivers = 0x7f48fe979ac0 (hDrivers = 0x7f49b58dfbc0)
<<<< [3604420664] zeDriverGet [1573 ns] Count = 1 hDrivers = 0x19c2cf0 -> ZE_RESULT_SUCCESS(0x0)
>>>> [3604425537] zeContextCreate: hDriver = 0x19c2cf0 desc = 0x7fff9d08f5a0 {ZE_STRUCTURE_TYPE_CONTEXT_DESC(0xd) 0 0} phContext = 0x7fff9d08f590 (hContext = 0x4)
<<<< [3604436429] zeContextCreate [10892 ns] hContext = 0x2fed800 -> ZE_RESULT_SUCCESS(0x0)
>>>> [3612130292] zeDeviceGet: hDriver = 0x19c2cf0 pCount = 0x7fff9d08f438 (Count = 0) phDevices = 0
<<<< [3612147264] zeDeviceGet [16972 ns] Count = 1 -> ZE_RESULT_SUCCESS(0x0)
>>>> [3612154814] zeDeviceGet: hDriver = 0x19c2cf0 pCount = 0x7fff9d08f438 (Count = 1) phDevices = 0x7f48fe9f2600 (hDevices = 0)
<<<< [3612156824] zeDeviceGet [2010 ns] Count = 1 hDevices[0] = 0x19c32c0 -> ZE_RESULT_SUCCESS(0x0)
>>>> [3618860141] zeMemAllocDevice: hContext = 0x2fed800 device_desc = 0x7fff9d08f5b0 {ZE_STRUCTURE_TYPE_DEVICE_MEM_ALLOC_DESC(0x15) 0 0 0} size = 400 alignment = 4 hDevice = 0x19c32c0 pptr = 0x7fff9d08f590 (ptr = 0)
<<<< [3618915341] zeMemAllocDevice [55200 ns] ptr = 0xffffd556aa7e0000 -> ZE_RESULT_SUCCESS(0x0)
>>>> [3618927142] zeContextMakeMemoryResident: hContext = 0x2fed800 hDevice = 0x19c32c0 ptr = 0xffffd556aa7e0000 size = 400
<<<< [3618930800] zeContextMakeMemoryResident [3658 ns] -> ZE_RESULT_SUCCESS(0x0)
>>>> [3651573981] zeCommandQueueCreate: hContext = 0x2fed800 hDevice = 0x19c32c0 desc = 0x7fff9d08f600 {ZE_STRUCTURE_TYPE_COMMAND_QUEUE_DESC(0xe) 0 0 0 0 0 0} phCommandQueue = 0x7fff9d08f5e0 (hCommandQueue = 0x7f49b3b18070)
<<<< [3651648628] zeCommandQueueCreate [74647 ns] hCommandQueue = 0x3031380 -> ZE_RESULT_SUCCESS(0x0)
>>>> [3672639333] zeCommandListCreate: hContext = 0x2fed800 hDevice = 0x19c32c0 desc = 0x7fff9d08f180 {ZE_STRUCTURE_TYPE_COMMAND_LIST_DESC(0xf) 0 0 0} phCommandList = 0x7fff9d08f160 (hCommandList = 0x5)
<<<< [3672702042] zeCommandListCreate [62709 ns] hCommandList = 0x22b5870 -> ZE_RESULT_SUCCESS(0x0)
>>>> [3672710160] zeCommandListAppendMemoryFill: hCommandList = 0x22b5870 ptr = 0xffffd556aa7e0000 pattern = 0x7f48fed14380 pattern_size = 4 size = 400 hSignalEvent = 0 numWaitEvents = 0 phWaitEvents = 0x7f48fed15580 (hWaitEvents = 0)
<<<< [3672821676] zeCommandListAppendMemoryFill [111516 ns] hWaitEvents = 0 -> ZE_RESULT_SUCCESS(0x0)
>>>> [3672829227] zeCommandListClose: hCommandList = 0x22b5870
<<<< [3672831328] zeCommandListClose [2101 ns] -> ZE_RESULT_SUCCESS(0x0)
>>>> [3672835843] zeCommandQueueExecuteCommandLists: hCommandQueue = 0x3031380 numCommandLists = 1 phCommandLists = 0x7f49b64f8810 (hCommandLists = 0x22b5870) hFence = 0
<<<< [3673166676] zeCommandQueueExecuteCommandLists [330833 ns] hCommandLists = 0x22b5870 -> ZE_RESULT_SUCCESS(0x0)
>>>> [3673197278] zeMemAllocDevice: hContext = 0x2fed800 device_desc = 0x7fff9d08f5b0 {ZE_STRUCTURE_TYPE_DEVICE_MEM_ALLOC_DESC(0x15) 0 0 0} size = 400 alignment = 4 hDevice = 0x19c32c0 pptr = 0x7fff9d08f590 (ptr = 0x7f49cbde41a0)
<<<< [3673243611] zeMemAllocDevice [46333 ns] ptr = 0xffffd556aa7d0000 -> ZE_RESULT_SUCCESS(0x0)
>>>> [3673250226] zeContextMakeMemoryResident: hContext = 0x2fed800 hDevice = 0x19c32c0 ptr = 0xffffd556aa7d0000 size = 400
<<<< [3673255156] zeContextMakeMemoryResident [4930 ns] -> ZE_RESULT_SUCCESS(0x0)
>>>> [3673275581] zeCommandListCreate: hContext = 0x2fed800 hDevice = 0x19c32c0 desc = 0x7fff9d08f180 {ZE_STRUCTURE_TYPE_COMMAND_LIST_DESC(0xf) 0 0 0} phCommandList = 0x7fff9d08f160 (hCommandList = 0x800)
<<<< [3673310467] zeCommandListCreate [34886 ns] hCommandList = 0x29b3190 -> ZE_RESULT_SUCCESS(0x0)
>>>> [3673315718] zeCommandListAppendMemoryFill: hCommandList = 0x29b3190 ptr = 0xffffd556aa7d0000 pattern = 0x7f48fed15780 pattern_size = 4 size = 400 hSignalEvent = 0 numWaitEvents = 0 phWaitEvents = 0x7f48fed157c0 (hWaitEvents = 0)
<<<< [3673335497] zeCommandListAppendMemoryFill [19779 ns] hWaitEvents = 0 -> ZE_RESULT_SUCCESS(0x0)
>>>> [3673341001] zeCommandListClose: hCommandList = 0x29b3190
<<<< [3673345002] zeCommandListClose [4001 ns] -> ZE_RESULT_SUCCESS(0x0)
>>>> [3673350728] zeCommandQueueExecuteCommandLists: hCommandQueue = 0x3031380 numCommandLists = 1 phCommandLists = 0x7f49b64f8b80 (hCommandLists = 0x29b3190) hFence = 0
<<<< [3673460380] zeCommandQueueExecuteCommandLists [109652 ns] hCommandLists = 0x29b3190 -> ZE_RESULT_SUCCESS(0x0)
>>>> [3776394795] zeCommandListDestroy: hCommandList = 0x29b3190
<<<< [3776429469] zeCommandListDestroy [34674 ns] -> ZE_RESULT_SUCCESS(0x0)
>>>> [3776436199] zeCommandListDestroy: hCommandList = 0x22b5870
<<<< [3776440182] zeCommandListDestroy [3983 ns] -> ZE_RESULT_SUCCESS(0x0)

>>>> [11059390906] zeModuleCreate: hContext = 0x2fed800 hDevice = 0x19c32c0 desc = 0x7fff9d08f360 {ZE_STRUCTURE_TYPE_MODULE_DESC(0x1b) 0 0 1944 0x7f48fcf5cff8  0x7fff9d08f340} phModule = 0x7f48fc6da320 (hModule = 0x1020100) phBuildLog = 0x7f48fc6da310 (hBuildLog = 0x7f0001020100)
<<<< [11103366053] zeModuleCreate [43975147 ns] hModule = 0x3ceb6f0 hBuildLog = 0x3ddd240 -> ZE_RESULT_SUCCESS(0x0)
>>>> [11103393488] zeModuleBuildLogGetString: hModuleBuildLog = 0x3ddd240 pSize = 0x7fff9d08f360 pBuildLog = 0
[... cut ...]
<<<< [11513037093] zeCommandQueueDestroy [16856 ns] -> ZE_RESULT_SUCCESS(0x0)
>>>> [11513044594] zeContextEvictMemory: hContext = 0x2fed800 hDevice = 0x19c32c0 ptr = 0xffffd556aa7e0000 size = 400
<<<< [11513048580] zeContextEvictMemory [3986 ns] -> ZE_RESULT_SUCCESS(0x0)
>>>> [11513051303] zeMemFree: hContext = 0x2fed800 ptr = 0xffffd556aa7e0000
<<<< [11513066794] zeMemFree [15491 ns] -> ZE_RESULT_SUCCESS(0x0)
>>>> [11515967334] zeContextDestroy: hContext = 0x2fed800
<<<< [11515979890] zeContextDestroy [12556 ns] -> ZE_RESULT_SUCCESS(0x0)
```

maleadt, 2022-03-23:
I was asked the output of:
```
export IGC_DumpToCurrentDir=1
export IGC_ShaderDumpEnable=1
```

[igc.zip](https://github.com/intel/compute-runtime/files/8336787/igc.zip)

This is with the example from my initial post, with the only difference that I'm passing the pointers in a struct (representing an array). But the behavior is the same.

jgu222, 2022-04-06:
@maleadt How to reproduce it on Linux Ubuntu 20.04 TGLLP ?  Please provide the detailed steps.  We do have TGLLP Linux machine in our lab.

maleadt, 2022-04-06:
Here's the instructions I had mailed:

```
# download and extract Julia 1.8
wget https://julialangnightlies-s3.julialang.org/bin/linux/x64/1.8/julia-latest-linux64.tar.gz
tar -xvf julia-latest-linux64.tar.gz
export JULIA=$(echo $(pwd)/julia-*/bin/julia)

# clone oneAPI.jl and install dependencies (this will take a minute)
git clone https://github.com/JuliaGPU/oneAPI.jl
cd oneAPI.jl
$JULIA --project -e 'using Pkg; Pkg.instantiate()'

# that's it. you can now run julia with the `--project` flag in the oneAPI dir
$JULIA --project -e 'using oneAPI; oneAPI.versioninfo()'

# or for something more useful: run the oneAPI.jl test suite
$JULIA --project --threads=auto -e 'using Pkg; Pkg.test()'
# on tiger lake hardware, this will result in test failures (the bug we're talking about)

# optional: have oneAPI.jl use system dependencies of the oneAPI toolchain
$JULIA res/local.jl
```

As a reminder, this is the reproduces that should print all ones, but contains some zeros on Tiger Lake hardware:

```julia
using oneAPI

# simple kernel that copies `len` values from `src` to `dst`.
# each thread processes `elem_per_thread` values
function kernel(dest, src, elem_per_thread, len)
    for j in 1:elem_per_thread
        # compute the (1-based) index to copy
        i = get_local_id(0) + (get_group_id(0) - 1) * get_local_size(0)
        i += (j - 1) * get_local_size(0) * get_num_groups(0)

        # this early-exit causes the miscompilation
        i > len && return

        @inbounds dest[i] = src[i]
    end
    return
end

function main(T=Int32, n=10)
    # allocate two vectors of 100 elements
    A = oneAPI.ones(T, n*n)
    B = oneAPI.zeros(T, n*n)

    # invoke the kernel with 10 groups of 10 threads
    @oneapi items=n groups=n kernel(B, A, 1, n*n)

    # display the result as an 10x10 matrix
    reshape(Array(B), (n,n))
end

display(main())
```

See the README for more details on using a local toolchain: https://github.com/JuliaGPU/oneAPI.jl#using-a-local-toolchain=

maleadt, 2022-04-06:
> Thanks. What about OneAPI setup ? I have not done any oneapi before. Any links ?

oneAPI.jl automatically downloads a oneAPI toolchain. For a local toolchain you can install the debs listed on the compute-runtime release page, https://github.com/intel/compute-runtime/releases/tag/22.13.22789 and have it picked up by oneAPI.jl by running the local.jl script I mentioned, but you only need to do so if you have specific reasons for using a different toolchain (e.g. because it's a debug build).

> Does your group id starts from 1 ? If it were OCL, 'i' would be out of bound.

Correct, Julia is 1-indexed

> Also, maybe I need your get aub by setting "export SetCommandStreamReceiver=3" when running your TGLLP.

What do you mean? Setting that environment variable doesn't change the output here.

jgu222, 2022-04-06:
@maleadt  'aub' file is the input for our debugger.  Do get aub file,  do "export SetCommandStreamReceiver=3" before running the test.  After the test finishes,  the current dir will have .aub file.

But for now,  I don't need it.

maleadt, 2022-04-06:
Hmm, it didn't generate such a file here (while as previously asked running with e.g. IGC_DumpToCurrentDir=1 did produce a bunch of files).

jgu222, 2022-04-06:
I am getting error installing julia:
 $wget https://julialangnightlies-s3.julialang.org/bin/linux/x64/1.8/julia-latest-linux64.tar.gz
--2022-04-06 14:25:41--  https://julialangnightlies-s3.julialang.org/bin/linux/x64/1.8/julia-latest-linux64.tar.gz
Resolving proxy-chain.intel.com (proxy-chain.intel.com)... 10.1.201.150
Connecting to proxy-chain.intel.com (proxy-chain.intel.com)|10.1.201.150|:912... connected.
Proxy request sent, awaiting response... 200 OK
Length: 133405581 (127M) [application/x-tar]
julia-latest-linux64.tar.gz: Permission denied

Cannot write to ‘julia-latest-linux64.tar.gz’ (Success)

maleadt, 2022-04-07:
You don't have the permissions to write the downloaded file to the current folder. `cd` to a writable location.

jgu222, 2022-04-07:
thanks.  I have a fix pending.  The bug causes the visa local copyprog to propagate acc over a madw inst. The fix is to make sure madw has implAccDst.

jgu222, 2022-04-13:
IGC fix is probably in open source now.  Could you verify ?

maleadt, 2022-04-14:
Thanks. What exactly is the fix? Is it part of a release already?

jgu222, 2022-04-15:
yes, it is https://github.com/intel/intel-graphics-compiler/commit/756595a9fef899a644c4d06a0153fb7706f7dc99

maleadt, 2022-04-25:
Same with IGC v1.0.10988 and compute-runtime v22.16.22992 (with the cherry-picked fix, of course).

jgu222, 2022-04-25:
It works on my side.  Maybe the driver isn't used properly ?  Maybe wait until normal merge happens.  If it still has an issue. Let me know.

pengtu, 2022-04-25:
@jgu222 : After applying your change set, can we compare the generated code sequences on your test system and @maleadt's system to do a spot check?  What are the flags to generate such a trace?  Thanks.

jgu222, 2022-04-25:
@pengtu the change is in master-11085 now.
[image]

If your shader dump has that mov (circled)  it should have the fix.  Unfortunately,  I deleted the test from my testing machine.

maleadt, 2022-04-26:
The engineering build mailed to me did work, so I guess there other changes on IGC#master that matter for this test (assuming that build was just using the latest versions of everything). I'll just wait until everything is released first before debugging again.

maleadt, 2022-04-29:
Turns out I still had an override active with an older engineering build, and the change does really fix the issue as reported in https://github.com/JuliaGPU/oneAPI.jl/issues/154#issuecomment-1111459064=. Sorry for the confusion!

closed 2022-04-29 by maleadt