intel/compute-runtime #524: Miscompilation due to conditional store moving across work group barrier

maleadt, 2022-03-18:
We're experiencing a miscompilation due to a store to local memory within a conditional branch moving across a barrier. MWE in Julia using oneAPI.jl (which compiles to LLVM IR -> SPIRV using the SPIRV to LLVM translator, and interfaces with oneAPI using the Level Zero APIs):

```julia
function kernel(ptr)
    tx = get_local_id()

    # mark array accesses as inbounds to avoid runtime bounds checking code
    @inbounds begin
        # initialize some local memory
        y = oneLocalArray(Int32, 2)
        y[tx] = unsafe_load(ptr, tx)
        barrier()

        # show its contents on thread 1
        if tx == 1
            for i in 1:2
                oneAPI.@println("y[$i] = $(y[i])")
            end
        end

        # write to it, but _after_ having print the array elements
        barrier()
        if tx > 1
            y[tx] = 0
        end
    end

    return
end

x = oneArray(Int32[10, 10])

# pass a plain pointer to make the generated code easier to debug
ptr = reinterpret(Core.LLVMPtr{Int32,AS.Global}, pointer(x))
GC.@preserve x begin
    @oneapi items=2 kernel(ptr)
    synchronize()
end
```

This yields the following output:

```
y[1] = 10
y[2] = 0
```

The store `y[tx] = 0`, which happens after a work group barrier, is executed too early. Changing that store to be unconditional (i.e. by removing the `if tx > 1`) results in the expected output, with both `y[1]` and `y[2]` being 10.

Generated LLVM IR for this MWE:

```llvm
; ModuleID = 'text'
source_filename = "text"
target datalayout = "e-i64:64-v16:16-v24:32-v32:32-v48:64-v96:128-v192:256-v256:256-v512:512-v1024:1024"
target triple = "spir64-unknown-unknown"

@local_memory = internal unnamed_addr addrspace(3) global [2 x i32] zeroinitializer, align 4
@0 = private unnamed_addr constant [13 x i8] c"y[%ld] = %d\0A\00", align 1

declare i32 @printf(i8*, ...)

declare i64 @_Z12get_local_idj(i32)

declare void @_Z7barrierj(i32)

define spir_kernel void @_Z18julia_kernel_118217LLVMPtrI5Int32Li1EE(i8 addrspace(1)* %0) local_unnamed_addr #0 {
entry:
  %1 = call i64 @_Z12get_local_idj(i32 0)
  %2 = bitcast i8 addrspace(1)* %0 to i32 addrspace(1)*
[... 269 lines cut ...]
         %36 = OpLoad %v3ulong %__spirv_BuiltInLocalInvocationId Aligned 32
         %37 = OpCompositeExtract %ulong %36 0
         %39 = OpBitcast %_ptr_CrossWorkgroup_uint %32
         %40 = OpInBoundsPtrAccessChain %_ptr_CrossWorkgroup_uint %39 %37
         %41 = OpLoad %uint %40 Aligned 1
         %44 = OpInBoundsPtrAccessChain %_ptr_Workgroup_uint %local_memory %ulong_0 %37
               OpStore %44 %41 Aligned 4
               OpControlBarrier %uint_2 %uint_2 %uint_0
       %_not = OpIEqual %bool %37 %ulong_0
         %49 = OpInBoundsPtrAccessChain %_ptr_Workgroup_uint %local_memory %ulong_0 %ulong_0
         %51 = OpInBoundsPtrAccessChain %_ptr_Function_uchar %24 %ulong_0 %ulong_0
         %53 = OpInBoundsPtrAccessChain %_ptr_Workgroup_uint %local_memory %ulong_0 %ulong_1
               OpBranchConditional %_not %L56_preheader %L104
%L56_preheader = OpLabel
         %54 = OpLoad %uint %49 Aligned 4
         %55 = OpExtInst %uint %1 printf %51 %ulong_1 %54
         %56 = OpLoad %uint %53 Aligned 4
         %57 = OpExtInst %uint %1 printf %51 %ulong_2 %56
               OpBranch %L104
       %L104 = OpLabel
               OpControlBarrier %uint_2 %uint_2 %uint_0
               OpStore %44 %uint_0 Aligned 4
               OpReturn
               OpFunctionEnd
```

cmc-rep, 2022-03-23:
The code you generated for the bad case already  has issues. llvm optimization makes the barriers inside control-flow.

`br i1 %.not, label %L104.thread, label %L104

L104.thread:                                      ; preds = %entry
  %6 = load i32, i32 addrspace(3)* getelementptr inbounds ([2 x i32], [2 x i32] addrspace(3)* @local_memory, i64 0, i64 0), align 4
  %7 = call i32 (i8*, ...) @printf(i8* getelementptr inbounds ([13 x i8], [13 x i8]* @0, i64 0, i64 0), i64 1, i32 %6)
  %8 = load i32, i32 addrspace(3)* getelementptr inbounds ([2 x i32], [2 x i32] addrspace(3)* @local_memory, i64 0, i64 1), align 4
  %9 = call i32 (i8*, ...) @printf(i8* getelementptr inbounds ([13 x i8], [13 x i8]* @0, i64 0, i64 0), i64 2, i32 %8)
  call void @_Z7barrierj(i32 0)
  br label %L133

L104:                                             ; preds = %entry
  %10 = add i64 %1, 1
  call void @_Z7barrierj(i32 0)
  %11 = icmp slt i64 %10, 2
  br i1 %11, label %L133, label %L125

L125:                                             ; preds = %L104
  store i32 0, i32 addrspace(3)* %5, align 4
  br label %L133

L133:                                             ; preds = %L125, %L104, %L104.thread
  ret void`

cmc-rep, 2022-03-23:
Once barrier is inside control-flow, then its behavior is undefined.

For the good case, the input looks fine:
`L104:                                             ; preds = %L56.preheader, %entry
  call void @_Z7barrierj(i32 0)
  store i32 0, i32 addrspace(3)* %5, align 4
  ret void`

pengtu, 2022-03-23:
@cmc-rep : Is there a LLVM function attribute to set so we can mark "barrier" as not movable/splitable?

maleadt, 2022-03-23:
Yes, `convergent`, and this fixes the issue.

I'm not sure whether this is still a bug then (i.e., what the exact semantics of a SPIR-V barrier are, in CUDA this would require an unaligned barrier), so feel free to close this if it isn't.

closed 2022-03-24 by maleadt