intel/compute-runtime #525: Checksum issue

leoheck, 2022-03-18:
Hello 

This file has some files with wrong names (extensions):
https://github.com/intel/compute-runtime/releases/download/22.10.22597/ww10.sum

See the `.ddeb` on lines 4 and 6.
[image]

Then the check fails
```bash
➜  neo sha256sum -c ww10.sum
intel-gmmlib_22.0.2_amd64.deb: OK
intel-igc-core_1.0.10409_amd64.deb: OK
intel-igc-opencl_1.0.10409_amd64.deb: OK
sha256sum: intel-level-zero-gpu-dbgsym_1.3.22597_amd64.ddeb: No such file or directory
intel-level-zero-gpu-dbgsym_1.3.22597_amd64.ddeb: FAILED open or read <========================
intel-level-zero-gpu_1.3.22597_amd64.deb: OK
sha256sum: intel-opencl-icd-dbgsym_22.10.22597_amd64.ddeb: No such file or directory
intel-opencl-icd-dbgsym_22.10.22597_amd64.ddeb: FAILED open or read <========================
intel-opencl-icd_22.10.22597_amd64.deb: OK
sha256sum: WARNING: 2 listed files could not be read
```

Pookalton, 2022-03-20:
Maybe no one actually uses the checksums?...
That's a bit scary
I noticed this a few days ago too any decided not to install.

leoheck, 2022-03-20:
Did you rename files to use the wrong extension `.ddeb` or did you update the `.sum` file with the right extension `.deb`?

JablonskiMateusz, 2022-03-21:
Hi @leoheck could you just download ddeb files? They are included to release:
```
wget https://github.com/intel/compute-runtime/releases/download/22.10.22597/intel-level-zero-gpu-dbgsym_1.3.22597_amd64.ddeb
wget https://github.com/intel/compute-runtime/releases/download/22.10.22597/intel-opencl-icd-dbgsym_22.10.22597_amd64.ddeb
```

wzmudat, 2022-03-21:
Issue fixed: https://github.com/intel/compute-runtime/releases/tag/22.10.22597

closed 2022-03-21 by wzmudat