intel/compute-runtime #526: Compilation fails due ocloc

ConiKost, 2022-03-24:
If I do try to compile compute-runtime, I am getting a compile failure and have no idea, how to debug this.

Compilation always fails here:

```
[565/2208] /usr/bin/x86_64-pc-linux-gnu-g++  @shared/source/built_ins/kernels/CMakeFiles/builtins_spirv.dir/__/__/__/__/bin/built_ins/x64/spirv/copy_image3d_to_buffer_stateless.cpp.o.rsp -MD -MT shared/source/built_ins/kernels/CMakeFiles/builtins_spirv.dir/__/__/__/__/bin/built_ins/x64/spirv/copy_image3d_to_buffer_stateless.cpp.o -MF shared/source/built_ins/kernels/CMakeFiles/builtins_spirv.dir/__/__/__/__/bin/built_ins/x64/spirv/copy_image3d_to_buffer_stateless.cpp.o.d -o shared/source/built_ins/kernels/CMakeFiles/builtins_spirv.dir/__/__/__/__/bin/built_ins/x64/spirv/copy_image3d_to_buffer_stateless.cpp.o -c /var/tmp/portage/dev-libs/intel-compute-runtime-22.11.22682/work/compute-runtime-22.11.22682_build/bin/built_ins/x64/spirv/copy_image3d_to_buffer_stateless.cpp
[566/2208] cd /var/tmp/portage/dev-libs/intel-compute-runtime-22.11.22682/work/compute-runtime-22.11.22682/shared/source/built_ins/kernels && LD_LIBRARY_PATH=/var/tmp/portage/dev-libs/intel-compute-runtime-22.11.22682/work/compute-runtime-22.11.22682_build/bin /var/tmp/portage/dev-libs/intel-compute-runtime-22.11.22682/work/compute-runtime-22.11.22682_build/bin/ocloc -q -file /var/tmp/portage/dev-libs/intel-compute-runtime-22.11.22682/work/compute-runtime-22.11.22682/shared/source/built_ins/kernels/copy_buffer_rect.builtin_kernel -spv_only -device bdw -force_stos_opt -64 -output bindful_copy_buffer_rect -out_dir /var/tmp/portage/dev-libs/intel-compute-runtime-22.11.22682/work/compute-runtime-22.11.22682_build/bin/built_ins/x64/gen8 -options -cl-kernel-arg-info
[567/2208] cd /var/tmp/portage/dev-libs/intel-compute-runtime-22.11.22682/work/compute-runtime-22.11.22682/shared/source/built_ins/kernels && LD_LIBRARY_PATH=/var/tmp/portage/dev-libs/intel-compute-runtime-22.11.22682/work/compute-runtime-22.11.22682_build/bin /var/tmp/portage/dev-libs/intel-compute-runtime-22.11.22682/work/compute-runtime-22.11.22682_build/bin/ocloc -q -file /var/tmp/portage/dev-libs/intel-compute-runtime-22.11.22682/work/compute-runtime-22.11.22682_build/bin/built_ins/x64/gen8/bindful_copy_buffer_rect_Gen8core.spv -spirv_input -device bdw -force_stos_opt -64 -output bindful_copy_buffer_rect_0 -out_dir /var/tmp/portage/dev-libs/intel-compute-runtime-22.11.22682/work/compute-runtime-22.11.22682_build/bin/built_ins/x64/gen8 -revision_id 0 -options -cl-kernel-arg-info
FAILED: bin/built_ins/x64/gen8/bindful_copy_buffer_rect_0_Gen8core.gen bin/built_ins/x64/gen8/bindful_copy_buffer_rect_0_Gen8core.bin /var/tmp/portage/dev-libs/intel-compute-runtime-22.11.22682/work/compute-runtime-22.11.22682_build/bin/built_ins/x64/gen8/bindful_copy_buffer_rect_0_Gen8core.gen /var/tmp/portage/dev-libs/intel-compute-runtime-22.11.22682/work/compute-runtime-22.11.22682_build/bin/built_ins/x64/gen8/bindful_copy_buffer_rect_0_Gen8core.bin 
cd /var/tmp/portage/dev-libs/intel-compute-runtime-22.11.22682/work/compute-runtime-22.11.22682/shared/source/built_ins/kernels && LD_LIBRARY_PATH=/var/tmp/portage/dev-libs/intel-compute-runtime-22.11.22682/work/compute-runtime-22.11.22682_build/bin /var/tmp/portage/dev-libs/intel-compute-runtime-22.11.22682/work/compute-runtime-22.11.22682_build/bin/ocloc -q -file /var/tmp/portage/dev-libs/intel-compute-runtime-22.11.22682/work/compute-runtime-22.11.22682_build/bin/built_ins/x64/gen8/bindful_copy_buffer_rect_Gen8core.spv -spirv_input -device bdw -force_stos_opt -64 -output bindful_copy_buffer_rect_0 -out_dir /var/tmp/portage/dev-libs/intel-compute-runtime-22.11.22682/work/compute-runtime-22.11.22682_build/bin/built_ins/x64/gen8 -revision_id 0 -options -cl-kernel-arg-info
ninja: build stopped: subcommand failed.
```

If I do manually run the command, I get the follow output:
```
cd /var/tmp/portage/dev-libs/intel-compute-runtime-22.11.22682/work/compute-runtime-22.11.22682/shared/source/built_ins/kernels && LD_LIBRARY_PATH=/var/tmp/portage/dev-libs/intel-compute-runtime-22.11.22682/work/compute-runtime-22.11.22682_build/bin /var/tmp/portage/dev-libs/intel-compute-runtime-22.11.22682/work/compute-runtime-22.11.22682_build/bin/ocloc -q -file /var/tmp/portage/dev-libs/intel-compute-runtime-22.11.22682/work/compute-runtime-22.11.22682_build/bin/built_ins/x64/gen8/bindful_copy_buffer_rect_Gen8core.spv -spirv_input -device bdw -force_stos_opt -64 -output bindful_copy_buffer_rect_0 -out_dir /var/tmp/portage/dev-libs/intel-compute-runtime-22.11.22682/work/compute-runtime-22.11.22682_build/bin/built_ins/x64/gen8 -revision_id 0 -options -cl-kernel-arg-info
[0]: /var/tmp/portage/dev-libs/intel-compute-runtime-22.11.22682/work/compute-runtime-22.11.22682_build/bin/libocloc.so(+0x9eea1) [0x7f396db4cea1]
[1]: /lib64/libc.so.6(+0x3dc40) [0x7f396d8c6c40]
[2]: /usr/lib/llvm/13/lib64/libLLVM-13.so(_ZN4llvm11PointerType3getEPNS_4TypeEj+0x15) [0x7f3965501ff5]
[3]: /usr/lib64/libigc.so.1(+0x26fd26) [0x7f396b90ad26]
[4]: /usr/lib64/libigc.so.1(+0x28e032) [0x7f396b929032]
[5]: /usr/lib64/libigc.so.1(+0x2928a9) [0x7f396b92d8a9]
[6]: /usr/lib64/libigc.so.1(+0x2989c0) [0x7f396b9339c0]
[7]: /usr/lib64/libigc.so.1(+0x29e19d) [0x7f396b93919d]
[8]: /usr/lib64/libigc.so.1(+0x29eb00) [0x7f396b939b00]
[9]: /usr/lib64/libigc.so.1(+0x1cd2f9) [0x7f396b8682f9]
[10]: /usr/lib64/libigc.so.1(+0x1ce9cf) [0x7f396b8699cf]
[11]: /usr/lib64/libigc.so.1(+0x1cfc32) [0x7f396b86ac32]
[12]: /usr/lib64/libigc.so.1(+0x2be2c1) [0x7f396b9592c1]
[13]: /usr/lib64/libigc.so.1(+0x1d3132) [0x7f396b86e132]
[14]: /usr/lib64/libigc.so.1(+0x2ac8ec) [0x7f396b9478ec]
[15]: /usr/lib64/libigc.so.1(+0x2adb90) [0x7f396b948b90]
[16]: /var/tmp/portage/dev-libs/intel-compute-runtime-22.11.22682/work/compute-runtime-22.11.22682_build/bin/libocloc.so(_ZN3NEO15OfflineCompiler15buildSourceCodeEv+0x207) [0x7f396db18297]
[17]: /var/tmp/portage/dev-libs/intel-compute-runtime-22.11.22682/work/compute-runtime-22.11.22682_build/bin/libocloc.so(_ZN3NEO15OfflineCompiler5buildEv+0x45) [0x7f396db1cd65]
[18]: /var/tmp/portage/dev-libs/intel-compute-runtime-22.11.22682/work/compute-runtime-22.11.22682_build/bin/libocloc.so(+0x9ef9a) [0x7f396db4cf9a]
[19]: /var/tmp/portage/dev-libs/intel-compute-runtime-22.11.22682/work/compute-runtime-22.11.22682_build/bin/libocloc.so(_Z20buildWithSafetyGuardPN3NEO15OfflineCompilerE+0xbe) [0x7f396db4ccce]
[20]: /var/tmp/portage/dev-libs/intel-compute-runtime-22.11.22682/work/compute-runtime-22.11.22682_build/bin/libocloc.so(oclocInvoke+0xd38) [0x7f396db00c88]
[21]: /var/tmp/portage/dev-libs/intel-compute-runtime-22.11.22682/work/compute-runtime-22.11.22682_build/bin/ocloc(main+0x5a) [0x617b29e668ea]
[22]: /lib64/libc.so.6(+0x293eb) [0x7f396d8b23eb]
[23]: /lib64/libc.so.6(__libc_start_main+0x79) [0x7f396d8b2499]
[24]: /var/tmp/portage/dev-libs/intel-compute-runtime-22.11.22682/work/compute-runtime-22.11.22682_build/bin/ocloc(_start+0x21) [0x617b29e667c1]
[1]    19896 IOT instruction  LD_LIBRARY_PATH=  -q -file  -spirv_input -device bdw -force_stos_opt -64     
```

Any ideas, what could go wrong?

ArchangeGabriel, 2022-03-26:
This is https://github.com/intel/intel-graphics-compiler/issues/204.

ConiKost, 2022-04-27:
Ack, thanks!

closed 2022-04-27 by ConiKost