intel/compute-runtime #527: Failed to reset GuC, ret = -110 leads to hard LOCKUP on cpu

JHarding86, 2022-03-28:
Hello,
I am running an Alder Lake with Proxmox (debian). I have been running into this issue for the past few weeks which leads to a complete system crash.

```
[i915] *ERROR* Failed to reset GuC, ret = -110
NMI watchdog: Watchdog detected hard LOCKUP on cpu 8
NMI watchdog: Watchdog detected hard LOCKUP on cpu 4
```

Check the attached picture for more errors. I can provide answers to any questions, and I wish I could provide more information, I'm just not sure where to look
[image]
.

jandres742, 2022-03-28:
this seems to indicate your program is crashing or hanging, and sending the GPU into a bad state. is this a new test you are running, or was it working before? could you share a bit of details of what you are executing?

JHarding86, 2022-03-28:
I have a Plex Linux Container that is attempting to do some hardware transcoding.

Admittedly i915 is new to me, and this is a new setup. I initially battled getting hardware transcoding working at all on Adler Lake. I have a thread on the Proxmox forums (here is the thread https://forum.proxmox.com/threads/alder-lake-gvt-d-integrated-graphics-passthrough.105983/) that details my journey up until this point, but I transitioned here because I think this is more pertinent to i915. Please let me know if I'm wrong.

JHarding86, 2022-03-30:
This happened again, here is a picture of the output. [image]

JablonskiMateusz, 2022-08-22:
As @eero-t mentioned:
> Make sure you're using latest kernel + matching GuC version, and if that does not help, please file the GPU reset failure issue against kernel: https://gitlab.freedesktop.org/drm/intel/-/issues
You could file another ticket against media-driver about it triggering the reset, and link the kernel ticket there.

I'm closing this issue

closed 2022-08-22 by JablonskiMateusz