intel/compute-runtime #528: Linking failure (due to -Werror) with LTO

ArchangeGabriel, 2022-03-28:
When compiling with LTO, the build ultimately fails like this:
```
/build/intel-compute-runtime/src/compute-runtime-22.12.22749/third_party/uapi/drm/i915_drm.h:2951: error: type ‘struct drm_i915_gem_create_ext’ violates the C++ One Definition Rule [-Werror=odr]
 2951 | struct drm_i915_gem_create_ext {
      | 
/build/intel-compute-runtime/src/compute-runtime-22.12.22749/third_party/uapi/dg1/drm/i915_drm.h:726: note: a different type is defined in another translation unit
  726 | struct drm_i915_gem_create_ext {
      | 
/build/intel-compute-runtime/src/compute-runtime-22.12.22749/third_party/uapi/drm/i915_drm.h:2971: note: the first difference of corresponding definitions is field ‘flags’
 2971 |         __u32 flags;
      | 
/build/intel-compute-runtime/src/compute-runtime-22.12.22749/third_party/uapi/dg1/drm/i915_drm.h:740: note: a field with different name is defined in another translation unit
  740 |         __u32 pad;
      | 
/build/intel-compute-runtime/src/compute-runtime-22.12.22749/third_party/uapi/drm/i915_drm.h:2951: error: type ‘struct drm_i915_gem_create_ext’ violates the C++ One Definition Rule [-Werror=odr]
 2951 | struct drm_i915_gem_create_ext {
      | 
/build/intel-compute-runtime/src/compute-runtime-22.12.22749/third_party/uapi/dg1/drm/i915_drm.h:726: note: a different type is defined in another translation unit
  726 | struct drm_i915_gem_create_ext {
      | 
/build/intel-compute-runtime/src/compute-runtime-22.12.22749/third_party/uapi/drm/i915_drm.h:2971: note: the first difference of corresponding definitions is field ‘flags’
 2971 |         __u32 flags;
      | 
/build/intel-compute-runtime/src/compute-runtime-22.12.22749/third_party/uapi/dg1/drm/i915_drm.h:740: note: a field with different name is defined in another translation unit
  740 |         __u32 pad;
      | 
lto1: all warnings being treated as errors
```

JablonskiMateusz, 2022-04-26:
Hi @ArchangeGabriel. Thanks for reporting the issue. I've added this to our queue.

JablonskiMateusz, 2022-07-18:
@ArchangeGabriel could you retry with https://github.com/intel/compute-runtime/commit/78c96db2faacdc3406a24a91f017aeae40b7ca54 ?

ArchangeGabriel, 2022-07-21:
With that commit on top of latest release, I get this:
```
/build/intel-compute-runtime/src/compute-runtime-22.28.23726/level_zero/tools/test/unit_tests/sources/sysman/firmware_util/mock_fw_util_fixture.h:43: error: type ‘struct MockOsLibrary’ violates the C++ One Definition Rule [-Werror=odr]
   43 | struct MockOsLibrary : public OsLibraryUtil {
      | 
/build/intel-compute-runtime/src/compute-runtime-22.28.23726/level_zero/tools/test/unit_tests/sources/metrics/test_metric_oa_initialization.cpp:24:7: note: a type with the same name but different base type is defined in another translation unit
   24 | class MockOsLibrary : public NEO::OsLibrary {
      |       ^
/build/intel-compute-runtime/src/compute-runtime-22.28.23726/level_zero/tools/test/unit_tests/sources/sysman/firmware_util/mock_fw_util_fixture.h:42: note: type name ‘L0::ult::OsLibraryUtil’ should match type name ‘NEO::OsLibrary’
   42 | class OsLibraryUtil : public OsLibrary {};
      | 
/build/intel-compute-runtime/src/compute-runtime-22.28.23726/shared/source/os_interface/os_library.h:27:7: note: the incompatible type is defined here
   27 | class OsLibrary {
      |       ^
```
```
In member function ‘load’,
    inlined from ‘__conv_op ’ at /usr/include/c++/12.1.0/bits/atomic_base.h:348:0,
    inlined from ‘getTaskLevel’ at /build/intel-compute-runtime/src/compute-runtime-22.28.23726/opencl/source/event/event.cpp:729:0,
    inlined from ‘getTaskLevelFromWaitList’ at /build/intel-compute-runtime/src/compute-runtime-22.28.23726/opencl/source/command_queue/command_queue.cpp:434:56,
    inlined from ‘isTaskLevelUpdateRequired’ at /build/intel-compute-runtime/src/compute-runtime-22.28.23726/opencl/source/command_queue/enqueue_common.h:648:64:
/usr/include/c++/12.1.0/bits/atomic_base.h:488: error: ‘__atomic_load_4’ writing 4 bytes into a region of size 0 overflows the destination [-Werror=stringop-overflow=]
  488 |         return __atomic_load_n(&_M_i, int(__m));
      | 
In member function ‘load’,
    inlined from ‘__conv_op ’ at /usr/include/c++/12.1.0/bits/atomic_base.h:348:0,
    inlined from ‘getTaskLevel’ at /build/intel-compute-runtime/src/compute-runtime-22.28.23726/opencl/source/event/event.cpp:729:0,
    inlined from ‘getTaskLevelFromWaitList’ at /build/intel-compute-runtime/src/compute-runtime-22.28.23726/opencl/source/command_queue/command_queue.cpp:434:56,
    inlined from ‘obtainTaskLevelAndBlockedStatus’ at /build/intel-compute-runtime/src/compute-runtime-22.28.23726/opencl/source/command_queue/enqueue_common.h:616:41:
/usr/include/c++/12.1.0/bits/atomic_base.h:488: error: ‘__atomic_load_4’ writing 4 bytes into a region of size 0 overflows the destination [-Werror=stringop-overflow=]
  488 |         return __atomic_load_n(&_M_i, int(__m));
      | 
In member function ‘load’,
    inlined from ‘__conv_op ’ at /usr/include/c++/12.1.0/bits/atomic_base.h:348:0,
    inlined from ‘getTaskLevel’ at /build/intel-compute-runtime/src/compute-runtime-22.28.23726/opencl/source/event/event.cpp:729:0,
    inlined from ‘getTaskLevelFromWaitList’ at /build/intel-compute-runtime/src/compute-runtime-22.28.23726/opencl/source/command_queue/command_queue.cpp:434:56,
    inlined from ‘isTaskLevelUpdateRequired’ at /build/intel-compute-runtime/src/compute-runtime-22.28.23726/opencl/source/command_queue/enqueue_common.h:648:64,
    inlined from ‘obtainTaskLevelAndBlockedStatus’ at /build/intel-compute-runtime/src/compute-runtime-22.28.23726/opencl/source/command_queue/enqueue_common.h:619:61:
/usr/include/c++/12.1.0/bits/atomic_base.h:488: error: ‘__atomic_load_4’ writing 4 bytes into a region of size 0 overflows the destination [-Werror=stringop-overflow=]
  488 |         return __atomic_load_n(&_M_i, int(__m));
      | 
In member function ‘load’,
    inlined from ‘__conv_op ’ at /usr/include/c++/12.1.0/bits/atomic_base.h:348:0,
    inlined from ‘getTaskLevel’ at /build/intel-compute-runtime/src/compute-runtime-22.28.23726/opencl/source/event/event.cpp:729:0,
    inlined from ‘getTaskLevelFromWaitList’ at /build/intel-compute-runtime/src/compute-runtime-22.28.23726/opencl/source/command_queue/command_queue.cpp:434:56,
    inlined from ‘isTaskLevelUpdateRequired’ at /build/intel-compute-runtime/src/compute-runtime-22.28.23726/opencl/source/command_queue/enqueue_common.h:648:64:
/usr/include/c++/12.1.0/bits/atomic_base.h:488: error: ‘__atomic_load_4’ writing 4 bytes into a region of size 0 overflows the destination [-Werror=stringop-overflow=]
  488 |         return __atomic_load_n(&_M_i, int(__m));
      | 
In member function ‘load’,
    inlined from ‘__conv_op ’ at /usr/include/c++/12.1.0/bits/atomic_base.h:348:0,
    inlined from ‘getTaskLevel’ at /build/intel-compute-runtime/src/compute-runtime-22.28.23726/opencl/source/event/event.cpp:729:0,
    inlined from ‘getTaskLevelFromWaitList’ at /build/intel-compute-runtime/src/compute-runtime-22.28.23726/opencl/source/command_queue/command_queue.cpp:434:56,
    inlined from ‘obtainTaskLevelAndBlockedStatus’ at /build/intel-compute-runtime/src/compute-runtime-22.28.23726/opencl/source/command_queue/enqueue_common.h:616:41:
/usr/include/c++/12.1.0/bits/atomic_base.h:488: error: ‘__atomic_load_4’ writing 4 bytes into a region of size 0 overflows the destination [-Werror=stringop-overflow=]
  488 |         return __atomic_load_n(&_M_i, int(__m));
      | 
In member function ‘load’,
    inlined from ‘__conv_op ’ at /usr/include/c++/12.1.0/bits/atomic_base.h:348:0,
[... cut ...]
/build/intel-compute-runtime/src/compute-runtime-22.28.23726/level_zero/core/test/unit_tests/mocks/mock_kernel.h:102:5: error: ‘getProperties’ violates the C++ One Definition Rule [-Werror=odr]
  102 |     ADDMETHOD_NOBASE(getProperties, ze_result_t, ZE_RESULT_SUCCESS, (ze_kernel_properties_t * pKernelProperties))
      |     ^
/build/intel-compute-runtime/src/compute-runtime-22.28.23726/level_zero/core/test/unit_tests/mocks/mock_kernel.h:102:5: note: ‘getProperties’ was previously declared here
  102 |     ADDMETHOD_NOBASE(getProperties, ze_result_t, ZE_RESULT_SUCCESS, (ze_kernel_properties_t * pKernelProperties))
      |     ^
```

JablonskiMateusz, 2022-07-22:
could you share more details of your build environment? e.g. compiler version, OS version etc

ArchangeGabriel, 2022-07-22:
Sure, you can find most information from the package page: https://archlinux.org/packages/community/x86_64/intel-compute-runtime/

So OS is Arch (no version), compiler is gcc 12.1.0, cmake command is at https://github.com/archlinux/svntogit-community/blob/packages/intel-compute-runtime/trunk/PKGBUILD#L29. Please tell me if you need specific information other than that.

ArchangeGabriel, 2023-01-07:
FTR, with current release it reduces to:
```
In member function ‘load’,
    inlined from ‘__conv_op ’ at /usr/include/c++/12.2.0/bits/atomic_base.h:348:0,
    inlined from ‘getTaskLevel’ at /build/intel-compute-runtime/src/compute-runtime-22.43.24595.30/opencl/source/event/event.cpp:732:0,
    inlined from ‘getTaskLevelFromWaitList’ at /build/intel-compute-runtime/src/compute-runtime-22.43.24595.30/opencl/source/command_queue/command_queue.cpp:492:56,
    inlined from ‘obtainTaskLevelAndBlockedStatus’ at /build/intel-compute-runtime/src/compute-runtime-22.43.24595.30/opencl/source/command_queue/enqueue_common.h:628:41:
/usr/include/c++/12.2.0/bits/atomic_base.h:488: error: ‘__atomic_load_4’ writing 4 bytes into a region of size 0 overflows the destination [-Werror=stringop-overflow=]
  488 |         return __atomic_load_n(&_M_i, int(__m));
      | 
In member function ‘load’,
    inlined from ‘__conv_op ’ at /usr/include/c++/12.2.0/bits/atomic_base.h:348:0,
    inlined from ‘getTaskLevel’ at /build/intel-compute-runtime/src/compute-runtime-22.43.24595.30/opencl/source/event/event.cpp:732:0,
    inlined from ‘getTaskLevelFromWaitList’ at /build/intel-compute-runtime/src/compute-runtime-22.43.24595.30/opencl/source/command_queue/command_queue.cpp:492:56,
    inlined from ‘isTaskLevelUpdateRequired’ at /build/intel-compute-runtime/src/compute-runtime-22.43.24595.30/opencl/source/command_queue/enqueue_common.h:660:64,
    inlined from ‘obtainTaskLevelAndBlockedStatus’ at /build/intel-compute-runtime/src/compute-runtime-22.43.24595.30/opencl/source/command_queue/enqueue_common.h:631:61:
/usr/include/c++/12.2.0/bits/atomic_base.h:488: error: ‘__atomic_load_4’ writing 4 bytes into a region of size 0 overflows the destination [-Werror=stringop-overflow=]
  488 |         return __atomic_load_n(&_M_i, int(__m));
      | 
In member function ‘load’,
    inlined from ‘__conv_op ’ at /usr/include/c++/12.2.0/bits/atomic_base.h:348:0,
    inlined from ‘getTaskLevel’ at /build/intel-compute-runtime/src/compute-runtime-22.43.24595.30/opencl/source/event/event.cpp:732:0,
    inlined from ‘getTaskLevelFromWaitList’ at /build/intel-compute-runtime/src/compute-runtime-22.43.24595.30/opencl/source/command_queue/command_queue.cpp:492:56,
    inlined from ‘obtainTaskLevelAndBlockedStatus’ at /build/intel-compute-runtime/src/compute-runtime-22.43.24595.30/opencl/source/command_queue/enqueue_common.h:628:41:
/usr/include/c++/12.2.0/bits/atomic_base.h:488: error: ‘__atomic_load_4’ writing 4 bytes into a region of size 0 overflows the destination [-Werror=stringop-overflow=]
  488 |         return __atomic_load_n(&_M_i, int(__m));
      | 
In member function ‘load’,
    inlined from ‘__conv_op ’ at /usr/include/c++/12.2.0/bits/atomic_base.h:348:0,
    inlined from ‘getTaskLevel’ at /build/intel-compute-runtime/src/compute-runtime-22.43.24595.30/opencl/source/event/event.cpp:732:0,
    inlined from ‘getTaskLevelFromWaitList’ at /build/intel-compute-runtime/src/compute-runtime-22.43.24595.30/opencl/source/command_queue/command_queue.cpp:492:56,
    inlined from ‘isTaskLevelUpdateRequired’ at /build/intel-compute-runtime/src/compute-runtime-22.43.24595.30/opencl/source/command_queue/enqueue_common.h:660:64,
    inlined from ‘obtainTaskLevelAndBlockedStatus’ at /build/intel-compute-runtime/src/compute-runtime-22.43.24595.30/opencl/source/command_queue/enqueue_common.h:631:61:
/usr/include/c++/12.2.0/bits/atomic_base.h:488: error: ‘__atomic_load_4’ writing 4 bytes into a region of size 0 overflows the destination [-Werror=stringop-overflow=]
  488 |         return __atomic_load_n(&_M_i, int(__m));
      | 
[100%] Built target ze_intel_gpu
In member function ‘load’,
    inlined from ‘__conv_op ’ at /usr/include/c++/12.2.0/bits/atomic_base.h:348:0,
    inlined from ‘getTaskLevel’ at /build/intel-compute-runtime/src/compute-runtime-22.43.24595.30/opencl/source/event/event.cpp:732:0,
    inlined from ‘getTaskLevelFromWaitList’ at /build/intel-compute-runtime/src/compute-runtime-22.43.24595.30/opencl/source/command_queue/command_queue.cpp:492:56,
    inlined from ‘obtainTaskLevelAndBlockedStatus’ at /build/intel-compute-runtime/src/compute-runtime-22.43.24595.30/opencl/source/command_queue/enqueue_common.h:628:41:
/usr/include/c++/12.2.0/bits/atomic_base.h:488: error: ‘__atomic_load_4’ writing 4 bytes into a region of size 0 overflows the destination [-Werror=stringop-overflow=]
  488 |         return __atomic_load_n(&_M_i, int(__m));
      | 
In member function ‘load’,
    inlined from ‘__conv_op ’ at /usr/include/c++/12.2.0/bits/atomic_base.h:348:0,
    inlined from ‘getTaskLevel’ at /build/intel-compute-runtime/src/compute-runtime-22.43.24595.30/opencl/source/event/event.cpp:732:0,
    inlined from ‘getTaskLevelFromWaitList’ at /build/intel-compute-runtime/src/compute-runtime-22.43.24595.30/opencl/source/command_queue/command_queue.cpp:492:56,
    inlined from ‘isTaskLevelUpdateRequired’ at /build/intel-compute-runtime/src/compute-runtime-22.43.24595.30/opencl/source/command_queue/enqueue_common.h:660:64,
    inlined from ‘obtainTaskLevelAndBlockedStatus’ at /build/intel-compute-runtime/src/compute-runtime-22.43.24595.30/opencl/source/command_queue/enqueue_common.h:631:61:
/usr/include/c++/12.2.0/bits/atomic_base.h:488: error: ‘__atomic_load_4’ writing 4 bytes into a region of size 0 overflows the destination [-Werror=stringop-overflow=]
  488 |         return __atomic_load_n(&_M_i, int(__m));
      | 
In member function ‘load’,
[... cut ...]
    inlined from ‘isTaskLevelUpdateRequired’ at /build/intel-compute-runtime/src/compute-runtime-22.43.24595.30/opencl/source/command_queue/enqueue_common.h:660:64,
    inlined from ‘obtainTaskLevelAndBlockedStatus’ at /build/intel-compute-runtime/src/compute-runtime-22.43.24595.30/opencl/source/command_queue/enqueue_common.h:631:61:
/usr/include/c++/12.2.0/bits/atomic_base.h:488: error: ‘__atomic_load_4’ writing 4 bytes into a region of size 0 overflows the destination [-Werror=stringop-overflow=]
  488 |         return __atomic_load_n(&_M_i, int(__m));
      | 
```

JablonskiMateusz, 2023-07-05:
@ArchangeGabriel  remaining ODR issues were fixed in https://github.com/intel/compute-runtime/commit/d2f1cf98d715ee2d00733bd8a7c0a9b18ae72e93  

Mentioned `__atomic_load` issues look like g++12 issue, it comes from `std::atomic<int>::load` function

kgibala, 2025-11-12:
Hi @ArchangeGabriel,

We’d like to know if this issue is still affecting you. If so, please provide an update or any additional information. Otherwise, we’ll close this issue after 30 days of inactivity. Your feedback is appreciated!

ArchangeGabriel, 2025-11-12:
Sorry, I’m not maintaining compute-runtime for Arch anymore (actually since my last message here basically), but according to https://gitlab.archlinux.org/archlinux/packaging/packages/intel-compute-runtime/-/blob/main/PKGBUILD?ref_type=heads#L25 we still disable LTO because of this. That being said, I’m not sure someone checked if something changed since then, so pinging @dbermond and @tpkessler who are the current maintainers so that they can have a look.

dbermond, 2025-11-13:
When enabling LTO, a lot of warnings happens during link time, giving a build failure due to `-Werror`. All of them are like this:

```
Linking CXX executable ../../../bin/neo_shared_tests
In function ‘make_unique_for_overwrite’,
    inlined from ‘populateFreeTags’ at /usr/src/debug/intel-compute-runtime/compute-runtime-25.40.35563.4/shared/source/utilities/tag_allocator.inl:122:10:
/usr/include/c++/15.2.1/bits/unique_ptr.h:1135:30: error: argument 1 value ‘18446744073709551615’ exceeds maximum object size 9223372036854775807 [-Werror=alloc-size-larger-than=]
 1135 |     { return unique_ptr<_Tp>(new remove_extent_t<_Tp>[__num]); }
      |                              ^
/usr/src/debug/intel-compute-runtime/compute-runtime-25.40.35563.4/shared/test/common/helpers/memory_management.cpp: In member function ‘populateFreeTags’:
/usr/src/debug/intel-compute-runtime/compute-runtime-25.40.35563.4/shared/test/common/helpers/memory_management.cpp:391:7: note: in a call to allocation function ‘operator new []’ declared here
lto1: all warnings being treated as errors
make[3]: *** [/tmp/cc6EWwDI.mk:62: /tmp/ccuG6ISc.ltrans30.ltrans.o] Error 1
make[3]: *** Waiting for unfinished jobs....
/usr/src/debug/intel-compute-runtime/compute-runtime-25.40.35563.4/shared/source/debug_settings/debug_settings_manager.cpp: In member function ‘getStringWithFlags’:
/usr/src/debug/intel-compute-runtime/compute-runtime-25.40.35563.4/shared/source/debug_settings/debug_settings_manager.cpp:96:6: note: variable tracking size limit exceeded with ‘-fvar-tracking-assignments’, retrying without
/usr/src/debug/intel-compute-runtime/compute-runtime-25.40.35563.4/shared/source/debug_settings/debug_settings_manager.cpp: In member function ‘getStringWithFlags’:
/usr/src/debug/intel-compute-runtime/compute-runtime-25.40.35563.4/shared/source/debug_settings/debug_settings_manager.cpp:96:6: note: variable tracking size limit exceeded with ‘-fvar-tracking-assignments’, retrying without
lto-wrapper: fatal error: make returned 2 exit status
compilation terminated.
/usr/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
make[2]: *** [shared/test/unit_test/CMakeFiles/neo_shared_tests.dir/build.make:10859: bin/neo_shared_tests] Error 1
make[1]: *** [CMakeFiles/Makefile2:28725: shared/test/unit_test/CMakeFiles/neo_shared_tests.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
```

Being such, this issue is still valid, but now with a different error.

When disabling LTO, no warnings happens during link time, and it builds fine.

If `-Werror` is removed by patching, it builds fine with LTO enabled. The unit tests passes fine with LTO, and I have no problems using the LTO build in my OpenCL usage (ffmpeg opencl filters). 

Compiler: GCC 15.2.1

@ArchangeGabriel happy to see you around!

thesamesam, 2025-11-13:
That means that inlining may lead to a too-large stack frame but it's not the same as the original issue (definite UB). Not sure if that warning has some false positives with LTO or not too (it falls into the general middle-end category of warnings where it's flow-sensitive). I wouldn't worry about that one.

JablonskiMateusz, 2025-11-17:
Hi @dbermond  
I was able to reproduce the issue and I prepared fix.
1. removal of disabling LTO based on compiler version https://github.com/intel/compute-runtime/commit/133421a5e9c8f26120413a8b770125c86e5e68d2 (without this one LTO was disabled and couldn't repro)
2. reduced type of variable that was used for allocating memory https://github.com/intel/compute-runtime/commit/7660b29bbbdcf05df14a1a4fe71ebea180a851cd 

Could you please retry with these two changes applied?

dbermond, 2025-11-21:
For the current legacy version, there is another warning unrelated to LTO that needs to be fixed before I can test the LTO build itself:

```
/build/intel-compute-runtime-legacy/src/compute-runtime-24.35.30872.36/level_zero/core/test/black_box_tests/common/zello_common.cpp: In function ‘LevelZeroBlackBoxTests::counterBasedEventsExtensionPresent(_ze_driver_handle_t*&)’:
/build/intel-compute-runtime-legacy/src/compute-runtime-24.35.30872.36/level_zero/core/test/black_box_tests/common/zello_common.cpp:393:12: error: ‘strncpy’ output truncated before terminating nul copying 40 bytes from a string of the same length [-Werror=stringop-truncation]
  393 |     strncpy(cbEventsExtension.name, cbEventsExtensionString.c_str(), cbEventsExtensionString.size());
      |            ^
```

This error happens regardless of LTO.

By enabling the unit tests in the legacy version, I also discovered more missing headers for GCC 15 that needs a fix. I have just sent a pull request for this.

dbermond, 2026-02-19:
> Could you please retest if legacy branch can be now built with LTO enabled?

When trying to compile legacy tag [24.35.30872.43](https://github.com/intel/compute-runtime/releases/tag/24.35.30872.43), the following error happens (see my previous comments [here](https://github.com/intel/compute-runtime/issues/528#issuecomment-3563053759) and [here](https://github.com/intel/compute-runtime/issues/528#issuecomment-3652222673)):

```
/build/intel-compute-runtime-legacy/src/compute-runtime-24.35.30872.43/level_zero/core/test/black_box_tests/common/zello_common.cpp: In function ‘LevelZeroBlackBoxTests::counterBasedEventsExtensionPresent(_ze_driver_handle_t*&)’:
/build/intel-compute-runtime-legacy/src/compute-runtime-24.35.30872.43/level_zero/core/test/black_box_tests/common/zello_common.cpp:393:12: error: ‘strncpy’ output truncated before terminating nul copying 40 bytes from a string of the same length [-Werror=stringop-truncation]
  393 |     strncpy(cbEventsExtension.name, cbEventsExtensionString.c_str(), cbEventsExtensionString.size());
      |
```

This is fixed in the non-legacy versions by commit https://github.com/intel/compute-runtime/commit/595fe107705e077052c8dbe8bfec23eaa673bd94, but it's not yet integrated into the legacy versions. Need to have this same fix in the legacy versions.

JablonskiMateusz, 2026-02-20:
@dbermond  please re-test on 24.35.30872.44, I applied the mentioned commit